Ylianst/ESP-IQ2020 #2: Protocol Decoding

wolfson292, 2024-05-29:
I've been working extensively on reverse engineering their IQ2020 firmware and have identified the rs485 address (0x1F) of their internet gateway and some of the functions. Followed you on Twitter, and happy to collaborate.

Ylianst, 2024-05-31:
Oh nice. Thanks. I spent a lot of the last 48 hours on this and made a lot of progress. You can see all of my [protocol decoding work here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/protocol.md). I got lights, temperature, jet pumps and more decoded and working.

I also made progress on the ESP32 integration and got lights and temperature working really well. I how have my Home Assistant automatically change the temperature based on electricity rates and recording the temperature graph.

[image]

[image]

I still have a few more days to go before it's something others can use. I have to work a bit more to get both Celsius and Fahrenheit modes working well. More importantly, the Spa Connection Kit polls for the data every minute and so, I just listen to that traffic, no need to add more traffic to the RS485 bus if I don't need to. I am coding it so the ESP32 will detect if a Spa Connection Kit device is already present and will not poll, but if it's not present it will do the polling itself.

Are you a Home Assistant user? I would go buy a M5Stack+RS485+Wires right away (it takes a while to deliver and it's cheap). If your not using Home Assistant, let me know what setup your looking to integrate with.

Also, if you have a music module on your Spa, let me know. I would absolutely love to get a traffic capture of the audio device. I would to impersonate it so I can control Home Assistant audio from the spa display,

Ylianst, 2024-05-31:
@wolfson292 - Just tried your commands and yes, they do work. Just added them to the protocol document.

Ylianst, 2024-06-01:
FYI for everyone, more progress this morning. I got a lot more features supported and exposed to Home Assistant.

[image]

wolfson292, 2024-06-01:
0x02, 0x55 and 0x02, 0x56 have the same handler, but 0x56 includes additional fields. Everything up to the length of 0x55 is the same in 0x56 response.

To tell if the device is in C or F mode, you can check byte 15, bit 6, of the response (1 = C, 0 = F).

Ylianst, 2024-06-01:
@wolfson292 What are C and F modes?

npinguin, 2024-06-02:
Excellent work, ordered the hardware and looking forward to try it out.

I am from Belgium and have a Hotspring Envoy 2017 with an ace saltwater system

Ylianst, 2024-06-02:
Thanks @npinguin. Hopefully you can help me test and give me feedback.

Today I am working on getting jets that have multiple speeds working nicely in Home Assistant. I also need to add data polling for this ESP-Home component to work for people that don't have the SPA connection kit, but I should get the done in the next few days.

[image]

fricker-ben, 2024-06-02:
Great work! Following closely with much interest! Just the thing I have been looking to implement for a while now. Parts are on order & hopefully arriving soon.

Ylianst, 2024-06-02:
Thanks @fricker-ben. Nice to hear this. Certainly motivating me to code. I can't wait to hear your feedback.

Ylianst, 2024-06-02:
Thanks to the information provided by @wolfson292. I added the version string as a sensor.

[image]

Ylianst, 2024-06-03:
I may have found the water heater wattage value. 3866W in the picture below. I am polling this every minute anyway and so, a good value for HA graphs.

[image]

Ylianst, 2024-06-03:
Ok. I added the polling support long with a bunch more things and updated all the documentation. I think this is pretty much a v1.0 at this point, it's probably something most geeks/nerds can use.

Ylianst, 2024-06-03:
Ok, one more feature. The integration will now show when the hot tub is heating in the climate user interface.

[image]

[image]

wolfson292, 2024-06-03:
> @wolfson292 What are C and F modes?
> 
> Oh... Celsius or Fahrenheit probably. I can see that when I read the temperature. However, good to know.

I was originally determining this like you were seeing if the temp string ended with `F`, but having the status bit seemed cleaner.

The third temperature is the heater outlet temperature. You'll see it spike when the heater turns on.

Your identification of these functions has allowed me to significantly improve my disassembly of the controller firmware, identifying many more global variables in the code for various statuses. Unsure on the legality of publicly sharing the current disassembly, but it's based off a version WR4.04 and DK4.00 `hex` file of the firmware for the controller and wireless display that was available on a public website. I was previously doing a MiTM setup on the RS422 interface with the wireless display, but this RS485 interface is far simpler and doesn't impact the ability to use the display itself.

Plan to upload my version of an IQ2020 component shortly to GitHub. Started on it before I found yours.

Ylianst, 2024-06-03:
Hi @wolfson292. To be clear, I do not want to ever see or be exposed to 3rd party code and please don't post any details about other code or binary from any sources here or on any forum. API's and interfaces are ok. I do appreciate the enthusiasm and that your even capable of doing this, that is absolutely insane! But yes, please keep it legal and respect other's work.

In any case, yes, thanks for the information on exhaust temp. I will add the outlet temperature to my component right away and the C/F bit to the documentation.

Ylianst, 2024-06-03:
I just added the heater outlet temperature to the integration. One more thing to graph!

[image]

wolfson292, 2024-06-04:
> Hi @wolfson292. To be clear, I do not want to ever see or be exposed to 3rd party code and please don't post any details about other code or binary from any sources here or on any forum. API's and interfaces are ok. I do appreciate the enthusiasm and that you are even capable of doing this, that is absolutely insane! But yes, please keep it legal and respect other's work.
> 
> In any case, yes, thanks for the information on exhaust temp. I will add the outlet temperature to my component right away and the C/F bit to the documentation.

Completely understand having worked on many commercial reverse engineering projects.

Identified a few more of the data fields from the 0x02, 0x55/0x56 command.

```
Data[35-38] Heater Total Runtime Seconds
Data[39-42] Jets 1 Total Runtime Seconds
Data[55-58] Jets 2 Total Runtime Seconds
Data[59-61] Jets 3 Total Runtime Seconds
Data[67-70] Lights Total Runtime Seconds
Data[73-76] Lifetime Runtime Seconds
Data[43-46] Lifetime Runtime Seconds (6 minutes higher than previous value, 1.5yrs, for my own Spa)
Data[47-50] Unknown Counter, 30 on my own Spa
Data[63-66] Unknown Seconds Timer, 0 on my own Spa
Data[63-66] Unknown Seconds Timer, 0 on my own Spa
Data[77-80] Unknown Seconds Timer Related to Jets 1, 0 on my own Spa
Data[81-84] Unknown Seconds Timer Related to Jets 2, 0 on my own Spa
```

Ylianst, 2024-06-04:
Wow @wolfson292 that is amazing data! I will add these runtime sensors in the next few hours. I also am reassured you know this business.

By the way, if you ever find a command to control light brightness and colors, that would be a fun one. I could add the brightness control on the existing light. I can read the lights brightness (0 to 5) using 0x0256 but can't control it.

Any other tricks we can do we the spa would be great!!

Ylianst, 2024-06-04:
I just added your counters to the C# DataViewer app. I index starting from the start of the packet.

```
int HeaterTotalRuntime = getIntFromByteArray(data, 40);
int Jets1TotalRuntime = getIntFromByteArray(data, 44);
int LifetimeRuntimeSeconds1 = getIntFromByteArray(data, 48);
int UnknownCounter1 = getIntFromByteArray(data, 52);
int Jets2TotalRuntime = getIntFromByteArray(data, 60);
int Jets3TotalRuntime = getIntFromByteArray(data, 64);
int UnknownCounter2 = getIntFromByteArray(data, 68);
int LightsTotalRuntime = getIntFromByteArray(data, 72);
int LifetimeRuntimeSeconds2 = getIntFromByteArray(data, 78);
int UnknownCounter3 = getIntFromByteArray(data, 82);
int UnknownCounter4 = getIntFromByteArray(data, 86);
```

These are my own values, very similar to yours.

[image]

Ylianst, 2024-06-04:
Done. Total runtime sensors are now included. I just added the ones we know what they indicate.

[image]

Ylianst, 2024-06-04:
I am starting to graph temperature with my electricity time-of-use rules since last night. I had a bad data point added during development, but in general, I like this.

[image]

Ylianst, 2024-06-05:
@wolfson292. I just added the protocol decoding to get/set the current time. This time is also at the end of the 0x0256 response.

Get Current Time
```
--> 01 1F 40 024C
<-- 1F 01 80 024C3431011400D40701
                 SSMMHHDDMMYYYY
Encoded as:
SS:MM:HH Seconds (0 to 59), Minutes (0 to 59), Hours (0 to 24).
DD:MM:YYYY Days (1 to 31), Month (0 to 11), Year (2 byte Big-Endian).
```

Set Current Time
```
--> 01 1F 40 024C2C2B140405E807
<-- 1F 01 80 024C2C2B140405E80701

Same encoding as Get Current Time.
```

Just for kicks, I added a new "Set Current Time" to the Data Viewer to set the spa time to your computer's clock.

[image]

npinguin, 2024-06-05:
Would it be possible to transform the timers in seconds in  yyy/mm/dd hh:mm:ss in home assistant?

Ylianst, 2024-06-05:
@npinguin I will look into that.

Ylianst, 2024-06-06:
@npinguin I just added a new document with [Home Assistant Sensor Templates](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/templates.md). This allows you to change the seconds into any format you like. I provide a full example to show the following:

[image]

Hope that helps.

Ylianst, 2024-06-06:
More data on my hot tub power usages over 2 days. Temperature is automatically adjusted. Where I live the prices are approximately:

 - Off-peak = 8c / KWh
 - Mid-peak = 14c / KWh
 - On-peak = 43c / KWh

So this makes a huge difference if I can heat only during off-peak. It's also good for the grid and the environment.

[image]

Ylianst, 2024-06-07:
I just fixed the problem with controlling jets that have a low/high settings. Works great now.

npinguin, 2024-06-07:
Just installed it, so far so good. Filling up the tub to see if the pumps work etc.

just noticed that power on counter is not visible for me. Tried a new install but did not see the sensor

Ylianst, 2024-06-07:
Good catch @npinguin . I just updated the configuration yaml file on the [main page](https://github.com/Ylianst/ESP-IQ2020) to add the `power_on_counter` value in the `sensor` section. That should work now.

[image]

Let let us know how it goes.

Ylianst, 2024-06-07:
I received a private email from someone trying to use the RS485MAX module. They can send commands but not receive any. I think the module looks like this:

[image]

I just updated the code to support `flow_control_pin` in the `iq2020` section of the configuration yaml.

```
iq2020:
   uart_id: SpaConnection
   flow_control_pin: GPIO0
   port: 1234
```

If you use this specific board, I think you need to set `flow_control_pin` to the `DE` GPIO pin. However, I will need confirmation that it work, I don't have this setup.

npinguin, 2024-06-07:
Filled up the tub and all is working perfectly!

Absolutely great work, very happy with this.

how did you get the software version from the spa?

Ylianst, 2024-06-07:
Thanks @npinguin for reporting back. If you can, please let me know what your software version and spa model is. I will keep a running list of tested versions and models. Also, do you happen to have any of the extra modules? Music/ACE/Freshwater?

For the version, I used the following command given by @wolfson292 above.

```
--> 01 1F 40 0100
<-- 1F 01 80 01005752342E30346465316345303032444B342E303006
```

npinguin, 2024-06-08:
> Thanks @npinguin for reporting back. If you can, please let me know what your software version and spa model are. I will keep a running list of tested versions and models. Also, do you happen to have any of the extra modules? Music/ACE/Freshwater?
> 
> For the version, I used the following command given by @wolfson292 above.
> 
> ```
> --> 01 1F 40 0100
> <-- 1F 01 80 01005752342E30346465316345303032444B342E303006
> ```

It is a Hotspring Envoy 2017 with an ace system.

Regarding the command to get the software version, how can I execute it and display it in home assistant? Is it possible to add it as a sensor ?

Ylianst, 2024-06-08:
Thanks. Yes, I have it as a sensor called 'version' already. Are you not seeing it?

[image]

npinguin, 2024-06-08:
No don’t see it, is also not listed in the yaml?

Ylianst, 2024-06-08:
I just added it at the end of the yaml on the [main page](https://github.com/Ylianst/ESP-IQ2020). Add it and let me know if it works.

[image]

npinguin, 2024-06-08:
Top, just added it and confirms that my firmware is old i guess

[image]

Ylianst, 2024-06-08:
Wow. That is super impressive.

npinguin, 2024-06-08:
anyone experience with firmware upgrades of the tub and what it could bring ? If it ain't broken do not upgrade is the typical recommendation. But i like keeping things up to date

Ylianst, 2024-06-08:
@npinguin  - Yes, getting the firmware change log for different versions would be great. I wonder if dealers have that.

Ylianst, 2024-06-09:
@wolfson292 - I sent all 64k possible commands to the IQ2020 from 0x1F looked for responses. I made the following table and if you have any guesses as to what some of the unknown commands are, please let me know. Thanks!

```
0100 - Get version string
0109 - Change temperature
0241 - Unknown (02413C001E0000)
024C - Unknown (024C1B0C110300E80701)
0255 - Get status short
0256 - Get status long
0B02 - Jets 1 & Lights on/off - (0B028F)
0B03 - Jets 2 on/off - (0B038E)
0B04 - Jets 3? on/off - (0B0400)
0B07 - Jets 4? on/off - (0B0700)
0B1C - Summer Timer
0B1D - Spa Lock
0B1E - Temperature Lock
0B1F - Clean Cycle
0B20 - Unknown (0B2001)
0B27 - Unknown (0B278C)
1702 - Lights on/off
1705 - Read lights status
1900 - Unknown (190015)
1901 - Unknown Periodic Polling (190100190000000B0004010000)
1D07 - Unknown Periodic Polling (1D07FFFF)
1E02 - Freshwater Salt System - Set Power / Start Test
1E03 - Get FreshWater Salt Module Data
```

Ylianst, 2024-06-10:
Just published YouTube videos on this integration:

Ylianst, 2024-06-16:
I just put in more updates. When changing settings, the ESP32 will auto-retry a command 3 times at 1/5 second interval. This is needed since there is a tiny chance that two commands would collide on the bus or there is an error and the IQ2020 controller did not get the command.

Also, I added a new sensor with the salt system power level (0 is off and 10 is max power). Only useful if you have this system. I could also remotely change the setting, I am not sure what Home Assistant control I could use for it.

Lastly, if anyone has this document... that would be super useful.

[image]

fricker-ben, 2024-06-16:
Finally home after a week away. All plugged in & working nicely on my Hot Spring Sovereign! 
Is there a way to set & view the current temperature in 0.5°C increments?

Ylianst, 2024-06-16:
@fricker-ben - Yes, Fahrenheit is set in 1.0F increments and Celsius in 0.5C increments. I have tested Fahrenheit a lot more than Celsius. You need to change some `_f_` to `_c_` in the ESPHome device config. In the view below, hit +/- and it should change by 0.5. If it does not, let me know, I will work to fix.

[image]

fricker-ben, 2024-06-16:
I have changed the _f_ to _c_, but when pushing the plus/minus button it's going in 1°C increments. Also the temp display only shows to 0 decimal places.
The heater outlet temp displays to 0.1°C resolution.

Ylianst, 2024-06-16:
Ok. I will do some testing and fixing on this later today. This is an essential feature of this integration.

Ylianst, 2024-06-16:
@fricker-ben I see the problem, working on it.

npinguin, 2024-06-16:
I have been testing the new version with the salt water, unfortunately it is not reusing the ace system interfaces which I was hoping it would. Would be great to have the ace system integrated, let me know what you need

fricker-ben, 2024-06-16:
> @fricker-ben I see the problem, working on it.

Thank you for your very quick response!

Ylianst, 2024-06-16:
@npinguin ACE uses completely different commands. I don't have a ACE system, however, I will open a new issue for you, just like I did for #4. If you feel up to it, you can send me traffic captures and I will add support for it. I will finish up with @fricker-ben  and get that done.

Ylianst, 2024-06-16:
@fricker-ben I just put in a fix for Celsius for the client control. I just updated the main page. In the `climate` section of the ESP home configuration, add the line `celsius: true` like this.

```
climate:
  - platform: iq2020
    name: Temperature
    celsius: true
```

Since I just did the change in GitHub, you can force the ESP Home add-in to check for updates by adding `refresh: 30s` to the `external_components` section like this:

```
external_components:
  - source: github://ylianst/esp-iq2020
    refresh: 30s
```

Put in the two lines, update your ESP32 device over WIFI and it should start working. It's not just the .5 step, the min/max temperatures in Celsius are also a bit different and needed fixing.

[image]

Also, can I have your hot tub model, version string and what is any extra modules you have? ACE/Freshwater/Music? I am trying to keep track of what is working and what is not. Let me know if this fix works for you. Thanks!

Ylianst, 2024-06-16:
@npinguin - Seems you like having access to a lot of sensors. I added instructions to add the [WIFI signal strength sensors](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/extras.md). Maybe you are interested in this.

fricker-ben, 2024-06-16:
@Ylianst - Thank you, the temperature set point now acts as expected.
It's a 2015 Hot Spring Sovereign, no extra modules.
Version: 1.08.D0172E002HL1.02

npinguin, 2024-06-16:
> @npinguin - Seems you like having access to a lot of sensors. I added instructions to add the [WIFI signal strength sensors](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/extras.md). Maybe you are interested in this.

Thanks :-)

wifi signal is indeed poor waiting for my outdoor Unifi access point to improve the signal ;-)

Ylianst, 2024-06-16:
Excellent. Thank you @fricker-ben. That was a really important fix and thank you for that version string, your the oldest version of the firmware I know about that it still works! This is excellent news. I was concerned that older firmwares would not work or need changes, but it seems not. Nice!!

@npinguin - I just added #6 for you. Let me know if you have any questions. I hope my instructions are clear and it should be simple.

fricker-ben, 2024-06-16:
@Ylianst - All seems to be working very well, although my life time runtime is displaying as approx 29.22 years (921453056 seconds), which I think it a little off! All other runtimes seem reasonable for the age of the Hot Tub.

Ylianst, 2024-06-16:
@fricker-ben That is interesting. I will take note of this. The lifetime encoding is very basic with not much room for error, basically, the value is directly obtained from the hot tub. If you even want to investigate this further, you can send me a packet capture like #4 to see what is going on, but not something that is crucial.

fricker-ben, 2024-06-16:
@Ylianst - Certainly not crucial! I will dig out a windows laptop at some point & do a packet capture for you. Always interesting to see what's going on, especially on the different (much older in my case) firmware versions.

Ylianst, 2024-06-17:
You would be the first one to provide a v1 firmware package capture... interesting to see what change have been made since.

[78 more replies]

closed 2024-07-18 by Ylianst