Ylianst/ESP-IQ2020 #3: ESP YAML - error

npinguin, 2024-06-04:
The yaml file does not seem to work

`esphome:
  name: jacuzzi
  friendly_name: Hotspring Envoy 2017
  comment: "Luxury Spa"
esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxx"

ota:
  password: "xxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Jacuzzi Fallback Hotspot"
    password: "1231231231"

captive_portal:

external_components:
- source: github://ylianst/esp-iq2020

uart:
  id: SpaConnection
  tx_pin: GPIO26
  rx_pin: GPIO32
  baud_rate: 38400

iq2020:
   uart_id: SpaConnection
   port: 1234

# If using celsius units on the hot tub remote, replace _f_ with _c_ in the three entries below.
# Feel free to remove any sensor that are not relevent for your hot tub.
sensor:
  - platform: iq2020
    current_f_temperature:
      name: Current Temperature
    target_f_temperature:
      name: Target Temperature
    outlet_f_temperature:
      name: Heater Outlet
    heater_wattage:
      name: Heater Power
    heater_total_runtime:
      name: Heater Runtime
    jets1_total_runtime:
      name: Jets 1 Runtime
    lifetime_runtime:
      name: Lifetime Runtime
    jets2_total_runtime:
      name: Jets 2 Runtime
#    jets3_total_runtime:
#      name: Jets 3 Runtime
    lights_total_runtime:
      name: Lights Runtime

switch:
  - platform: iq2020
    name: Lights
    id: lights_switch
    icon: "mdi:lightbulb"
    switch_datapoint: 0
  - platform: iq2020
    name: Spa Lock
    id: spa_lock_switch
    icon: "mdi:lock"
    switch_datapoint: 1
  - platform: iq2020
    name: Temperature Lock
    id: temp_lock_switch
    icon: "mdi:lock"
    switch_datapoint: 2
  - platform: iq2020
    name: Clean Cycle
    id: clean_cycle_switch
    icon: "mdi:vacuum"
    switch_datapoint: 3
  - platform: iq2020
    name: Summer Timer
    id: summer_timer_switch
    icon: "mdi:sun-clock"
    switch_datapoint: 4

fan:
  - platform: iq2020
    name: Jets 1
    id: jets1
    icon: "mdi:turbine"
    datapoint: 0
    speeds: 1
  - platform: iq2020
    name: Jets 2
    id: jets2
    icon: "mdi:turbine"
    datapoint: 1
    speeds: 2

climate:
  - platform: iq2020
    name: Temperature`

`INFO ESPHome 2024.5.4
INFO Reading configuration /config/esphome/jacuzzi.yaml...
Failed config

switch.iq2020: [source /config/esphome/jacuzzi.yaml:72]
  
  'datapoint' is a required option for [switch.iq2020].
  platform: iq2020
  name: Lights
  id: lights_switch
  icon: mdi:lightbulb
  
  [switch_datapoint] is an invalid option for [switch.iq2020]. Did you mean [datapoint]?
  switch_datapoint: 0
switch.iq2020: [source /config/esphome/jacuzzi.yaml:77]
  
  'datapoint' is a required option for [switch.iq2020].
  platform: iq2020
  name: Spa Lock
  id: spa_lock_switch
  icon: mdi:lock
  
  [switch_datapoint] is an invalid option for [switch.iq2020]. Did you mean [datapoint]?
  switch_datapoint: 1
switch.iq2020: [source /config/esphome/jacuzzi.yaml:82]
  
  'datapoint' is a required option for [switch.iq2020].
  platform: iq2020
  name: Temperature Lock
  id: temp_lock_switch
  icon: mdi:lock
  
  [switch_datapoint] is an invalid option for [switch.iq2020]. Did you mean [datapoint]?
  switch_datapoint: 2
switch.iq2020: [source /config/esphome/jacuzzi.yaml:87]
  
  'datapoint' is a required option for [switch.iq2020].
  platform: iq2020
  name: Clean Cycle
  id: clean_cycle_switch
  icon: mdi:vacuum
  
  [switch_datapoint] is an invalid option for [switch.iq2020]. Did you mean [datapoint]?
  switch_datapoint: 3
switch.iq2020: [source /config/esphome/jacuzzi.yaml:92]
  
  'datapoint' is a required option for [switch.iq2020].
  platform: iq2020
  name: Summer Timer
  id: summer_timer_switch
  icon: mdi:sun-clock
  
  [switch_datapoint] is an invalid option for [switch.iq2020]. Did you mean [datapoint]?
  switch_datapoint: 4`

Ylianst, 2024-06-04:
Your right. I updated the code from `switch_datapoint` to just `datapoint` and forgot to update the readme.md. I just updated it. Thanks! Let me know if you encounter any other issues. I may be the first one to try this integration.

closed 2024-06-04 by Ylianst

npinguin, 2024-06-04:
Thank you for the fast update(s) !

Compilation works, and no errors in the log file and used the celcius parameters

Unfortianetly I can only install the device next week when the jacuzzi is back up and running. I keep you posted.

Did you find any parameters to monitor the ace system ?

Ylianst, 2024-06-04:
I don't have the ACE system myself. However, once you get it installed, I can guide you to capture traffic and if we can decode that the data indicates, we can add ACE support.

Ylianst, 2024-06-04:
By the way, I use Fahrenheit myself and I know that works. I may switch to Celcius this evening and do more testing on that just to make sure it works.