Ylianst/ESP-IQ2020 #4: Fix support for Pump 1 with Off/Low/High settings

Ylianst, 2024-06-13:
Current integration supports Pump 1 with Off/High and Pump 2 with Off/Low/High. However, for people with Pump 1 Off/Low/High setting, the integration will always show "High" even if pump 1 is running at low. To fix this, I need a packet capture to know what bits are used to show Pump 1 "Low" state.

To do this, add the following line to your ESP-Home config.yaml.

```
# Relay UART messages over TCP
iq2020:
   uart_id: SpaConnection
   port: 1234                               <---------- Add this line
```

Re-flash the ESP32 device. Download the [Windows Data Viewer app](https://github.com/Ylianst/ESP-IQ2020/raw/main/DataViewer/IQ2020-DataViewer.exe) and connect it to the ESP32 using the IP address and port 1234. You may start seeing traffic like this:

[image]

In the "Commands" menu, select "Ask Status 0x256"

[image]

You will see a response like this. You can ignore any other traffic. It will start with `1F 01 80 0256`:

```
<-- 1F 01 80 0256000820040000060400020622F21100201C201C201C8403605400000000313033467667750092F10A007C8DED0547000000010000007BEF0C000000000000000000DAE00F0002005D54ED05000000004E920A0020393546313032467800F800F800000000000000000000000000530000000000013C001E0000726E003714150800E80701
```

Now, turn on pump 1 to low, capture this line. turn pump 1 to high and capture this again. Send back the off, low and high captures.

The expectation is that you should see a change near the start of the packet, around here:

```
<-- 1F 01 80 025600082004000006040002
                   ^^^^^^
```

This said, surprises can happen. Once we know the right bits, I will update the integration and Pump 1 should report correctly. Thanks!

bdtrocke, 2024-06-13:
When controlling from the control panel of the hot tub, I couldn't corelate any changes in the Data packets. However, when I change the pump from off to speed 1 in HA I got line "  <-- 1F 01 80 0B0201" then from speed 1 to speed 2 "<-- 1F 01 80 0B0202". Then off " <-- 1F 01 80 0B0200"

` --> 01 1F 40 0256
 <-- 1F 01 80 02560000000000000404000A0A2040160E201C201C201C840360540000000031303346185B82014731EB04E6A6DD134C000000000000000000000000000000000000003A4324010300000000005185BF04000000003130334631303446EA00EA00EA0000000000000000000000000000000000000000B400B400006F00000000000000DC0701
 <-- 1F 01 80 17050000000000000000000000000006010101000000
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 21 01 40 210100
 <-- 33 01 40 190100
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 21 01 40 210100
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 33 01 40 190100
 <-- 21 01 40 210100
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 33 01 40 190100
 <-- 1F 01 80 02560008040000000404000A0A2040160E201C201C201C840360540200000031303346185B82010532EB04BFA7DD134C000000000000000000000000000000000000003A432401030000000000D185BF04000000003130334631303446E900E900E90000000000000000000000000000000000000000B400B400006F00000000000000DC0701
 <-- 1F 01 80 17050000000000000000000000000006010101000000
 <-- 21 01 40 210100
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 33 01 40 190100
 <-- 21 01 40 210100
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 33 01 40 190100
 <-- 21 01 40 210100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 33 01 40 190100
 <-- 33 01 40 190100
 <-- 1F 01 80 02560000000000000404000A0A2040160E201C201C201C840360540000000031303346185B82017132EB0498A8DD134C000000000000000000000000000000000000003A432401030000000000D185BF04000000003130334631303446E800E800E80000000000000000000000000000000000000000B400B400006F00000000000000DC0701
 <-- 1F 01 80 17050000000000000000000000000006010101000000
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 21 01 40 210100
 <-- 33 01 40 190100
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 21 01 40 210100
 <-- 33 01 40 190100
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 21 01 40 210100
 <-- 33 01 40 190100
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 21 01 40 210100
 <-- 1F 01 80 02560000000000000404000A0A2040160E201C201C201C840360540000000031303346185B8201D732EB0471A9DD134C0000000000000000000000000000000000000053432401030000000000FC85BF04000000003130334631303446E900E900E90000000000000000000000000000000000000000B400B400006F00000000000000DC0701
 <-- 1F 01 80 17050000000000000000000000000006010101000000
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 33 01 40 190100
 <-- 21 01 40 210100
 <-- 33 01 40 190100
 <-- 1F 01 80 0B0201
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 33 01 40 190100
 <-- 21 01 40 210100
 <-- 1F 01 80 0B0202
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 1F 01 80 0B0200
 <-- 33 01 40 190100
 <-- 21 01 40 210100
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 1F 01 80 02560000000000000404000A0A2040160E201C201C201C840360540000000031303346185B82013433EB0449AADD134C00000000000000000000000000000000000000534324010300000000003286BF04000000003130334631303446E900E900E90000000000000000000000000000000000000000B400B400006F00000000000000DC0701
 <-- 1F 01 80 17050000000000000000000000000006010101000000`

Ylianst, 2024-06-13:
Thanks @bdtrocke, I edited you post to make everything easier to read. You almost go it! So close!

The speed change commands you are seeing are the responses from the hot tub to the ESP32 commands that you can't see.

All you have to do now is, while the pump1 is on low, send a "Ask Status 0x256" commands, then when on high, send another "Ask Status 0x256" command.

One alternative is that the ESP32 is asking for the state every minute... so start the capture and keep the pump at low for over a minute, than high for over a minute and then off.

I need the long `<-- 1F 01 80 02560000...` status in between the speed changes.

bdtrocke, 2024-06-13:
Gotcha! Here you go. Hopefully that helps and I was on the right track.
1st segment is when I connected to the hot tub
2nd segment is when I had pump 1 on speed 1
3rd segment is when I had pump 1 on speed 2
4th segment is when I had pump 1 off

 ` <-- 21 01 40 210100
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCF
 <-- 1F 01 80 02560000000000000404000A0A2040160E201C201C201C840360540000000031303346185B82013433EB0449AADD134C00000000000000000000000000000000000000534324010300000000003286BF04000000003130334631303446E900E900E90000000000000000000000000000000000000000B400B400006F00000000000000DC0701
 <-- 1F 01 80 17050000000000000000000000000006010101000000
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCD
 <-- 21 01 40 210100
 <-- 33 01 40 190100
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCD
 --> 01 1F 40 0256
 <-- 21 01 40 210100
 <-- 1F 01 80 02560000000100000404000A0A2040160E201C201C201C840360540100000031303246185B82015633EB044FCEDD134C00000000000000000000000000000000000000534324010300000000005486BF04000000003130334631303346E600E600E60000000000000000000000000000000000000000B400B400006E00000000000000DC0701
 <-- 1F 01 80 17050000000000000000000000000006010101000000
 <-- 33 01 40 190100
 <-- 33 01 40 190100
 <-- 24 01 40 1E01FF00FFFFFF00FF02FFFFFFFFCE
 --> 01 1F 40 0256
 <-- 1F 01 80 02560008040000000404000A0A2040160E201C201C201C840360540200000031303246185B82018633EB047FCEDD134C00000000000000000000000000000000000000534324010300000000007486BF04000000003130334631303346E700E700E70000000000000000000000000000000000000000B400B400006E00000000000000DC0701
 <-- 1F 01 80 17050000000000000000000000000006010101000000
 <-- 21 01 40 210100
 <-- 33 01 40 190100
 --> 01 1F 40 0256
 <-- 1F 01 80 02560000000000000404000A0A2040160E201C201C201C840360540000000031303246185B82019533EB049ECEDD134C00000000000000000000000000000000000000534324010300000000007486BF04000000003130334631303346E700E700E70000000000000000000000000000000000000000B400B400006F00000000000000DC0701
 <-- 1F 01 80 17050000000000000000000000000006010101000000
`

Ylianst, 2024-06-13:
Thanks. That looks good. Checking this out now.

```
 <-- 1F 01 80 02560000000000000404000A0A2040160E201C201C201C840360540000000031303346185B82013433EB0449AADD134C00000000000000000000000000000000000000534324010300000000003286BF04000000003130334631303446E900E900E90000000000000000000000000000000000000000B400B400006F00000000000000DC0701
 <-- 1F 01 80 02560000000100000404000A0A2040160E201C201C201C840360540100000031303246185B82015633EB044FCEDD134C00000000000000000000000000000000000000534324010300000000005486BF04000000003130334631303346E600E600E60000000000000000000000000000000000000000B400B400006E00000000000000DC0701
 <-- 1F 01 80 02560008040000000404000A0A2040160E201C201C201C840360540200000031303246185B82018633EB047FCEDD134C00000000000000000000000000000000000000534324010300000000007486BF04000000003130334631303346E700E700E70000000000000000000000000000000000000000B400B400006E00000000000000DC0701

```

Ylianst, 2024-06-13:
Patch done and on GitHub now. By default, ESP-Home checks GitHib for an update every day, so you can add this line to force it to update right away.

```
external_components:
  - source: github://ylianst/esp-iq2020
    refresh: 30s                                             <--- Add this to force refresh
```

Add this, reflash and it should work great. Huge thank you. Let me know if it works.

bdtrocke, 2024-06-13:
We are golden! When I turn pump 1 on speed 1 from the hot tub control panel, within a minute HA shows that pump 1 is on. When I select the entity it shows 50% and when I up pump 1 to speed 2 from the control panel it updates to 100%. Perfect!

Ylianst, 2024-06-13:
Nice!! Thank you so much for your help. Many others will benefit.

bdtrocke, 2024-06-13:
No worries! Glad to help. Super thankful for this integration! Do you have a “buy me a coffee” link?

Ylianst, 2024-06-13:
No need.  Your participation and help are plenty. Your probably one of the first to do this packet capture thing, I am very happy it worked.

One day, I need to find someone with the music device on their hot tub willing to do the same capture. I would love to be able to impersonate the music device and have Home Assistant show music tracks (or automations) on the spa remote.

closed 2024-06-13 by Ylianst