Ylianst/ESP-IQ2020 #5: Heater Power value not displaying anything but 0

bdtrocke, 2024-06-13:
Hey @Ylianst. I've noticed that the heater power doesn't work for my hot tub. I am not sure if that is a limit of my control board or not. HA shows 0 watts since deploying the ESP32. The heater outlet temp seems to be accurate, just not power. Here is a packet capture from the hot tub. Let me know if you need the full packet capture of just these snipits.

` 
#Heater Off
1F 01 80 02560008000100000404000A0A2040160E201C201C201C8403605401000000313033466B6C8201D10BEC04F2E3E0134C0000000000000000000000000000000000000059432401030000000000CE5DC004000000003130334631303446F100F100F10000000000000000000000000000000000000000B400B400047700000000000000DC0701
 
#Heater On
1F 01 80 02560008000105000404000A0A2040160E201C201C201C8403605401000000313033461F6D8201AA0CEC04CBE4E0134C0000000000000000000000000000000000000059432401030000000000A75EC004000000003130364631303446EF00EF00EF000000000000000F000000000000000000000000B400B400047700000000000000DC0701
 
#Heater On
1F 01 80 02560008000105000404000A0A2040160E201C201C201C840360540100000031303346426D8201CD0CEC04EEE4E0134C0000000000000000000000000000000000000059432401030000000000CA5EC004000000003130364631303446F000F000F0000000000000000F000000000000000000000000B400B400047700000000000000DC0701

#Heater Off
1F 01 80 02560008000100000404000A0A2040160E201C201C201C840360540100000031303346536D8201EF0CEC0410E5E0134C0000000000000000000000000000000000000059432401030000000000EC5EC004000000003130334631303446F100F100F10000000000000000000000000000000000000000B400B400047700000000000000DC0701
`

Ylianst, 2024-06-13:
I will take a look at this later today. Thanks for the capture. I have noticed a few things about the hot tub, it does not have a clock (not that it's useful anyway). It may be that you don't have a power meter, but will check.

bdtrocke, 2024-06-14:
Yeah no clock mentioned in the owners manual. Filter cycles are just based on minutes and when the hot tub was powered on.

bdtrocke, 2024-06-14:
Nope, hot spring

Ylianst, 2024-06-15:
HI @bdtrocke. I took the second packet you provided above and broke it down manually to see that it looks like. I have to say, after looking at data from my hot tub for a long time, it's so nice to see data from a different one... a bit alien in a way.

A few things to note:
 - Yes, the heater wattage is indeed 0 in your case. Nothing I can do, you can remove that sensor from the ESP32 config file.
 - Some of the lifetime counters are a different then mine. I have some values that you have zero at, and some values you have that I have zeros at. I will need to find the dealer settings for both our tubs (it's a fix set of settings for each tub model), I bet I could figure out that the values
 - I have a watch battery in my IQ2020 and I figured it was for a real clock and yes, I did find it was and you can ever use a command to se the correct date time. In playing around with the command, I did find that the start or Jan 1st 2012 was the minimal possible value of this clock. Well, your clock is at that value and not moving. I don't know what purpose it has, the time is not shown on the remote. I am guessing it's to clean cycle the time at the same times regardless of power state. I will have to check this.

There are still areas of data I don't know what they are for. I bet the tub model settings are some of them.

```
1F                - Destination Spa Connection Kit (0x1F).
01                - Source IQ2020 (0x01).
80                - Response (0x40 = Request, 0x80 = Response).
0256              - Main Status Data.
0008
00 = Flags1: None
01 = Flags2: Currently Jet1 Medium
05000404000A0A2040160E201C201C201C8403605401000000  -?
31303346 = ASCII string, the heater outlet temperature.
1F6D8201 - Heater total runtime in seconds (~293 days)
AA0CEC04 - Jets 1 total runtime in seconds (~39 days)
CBE4E013 - Lifetime in seconds (~10 years)
4C000000 - Power on / Boot counter (76 boots)
00000000 - ?
00000000 - Jet 2 runtime (none)
00000000 - Jet 3? runtime (none)
00000000 - Jet 4? runtime (none)
59432401 - Lights runtime (~221 days)
0300     - ? I have 0200 here, not idea what this is.
00000000 - ? This number is close to the lifetime of my hot tub, but zero here.
A75EC004 - Lifetime? I don't know that this is (~922 days) here, zero on mine.
00000000 - Lifetime? Zero here, 190 hour in mine
3130364631303446 - "106F104F", temperatures: 106F target, 104F currently.
EF00EF00EF00000000000000 - ? In the same ranges as my values.
0F       - Water heater relay. 0x00 = Off, 0x0F = On (Currently on)
0000000000 - ?
0000     - On mine, this is variable from 0x4D00 to 0xEA00, changes almost at every poll. Zero here.
0000     - On mine Water heater wattage, zero here.
000000B400B40004770000  - ?
000000    - 00:00:00 (Time: hours, minutes, seconds)
0000DC07  - Jan 1th, 2012 - This is the minimal clock value and not moving.
01  - ?
```

Ylianst, 2024-06-15:
If anyone has this document... that would be nice. Settings for various models.

[image]

closed 2024-06-16 by Ylianst