Ylianst/ESP-IQ2020 #6: Add support for the ACE system

Ylianst, 2024-06-16:
Current integration of IQ2020 does not supports the ACE system. It uses very different commands from the Freshwater Salt system and I will need data capture to figure out the ACE status and commands. In order to capture data start by adding the following line `port:1234` to your ESP-Home config.yaml in the `iq2020` section like this:

```
# Relay UART messages over TCP
iq2020:
   uart_id: SpaConnection
   port: 1234                               <---------- Add this line
```

Re-flash the ESP32 device then download the [Windows Data Viewer app](https://github.com/Ylianst/ESP-IQ2020/raw/main/DataViewer/IQ2020-DataViewer.exe) and connect it to the ESP32 using the IP address and port 1234. You may start seeing traffic like this:

[image]

I am guessing the ACE system will have commands that look like this:

```
<-- 24 01 40 1E01FF00FFFFFF00FF01FFFFFFFF00
<-- 01 24 80 ??????????
```

Note that commands that start with `24 01` are command from the IQ2020 controller to the ACE module. Commands that start with `24 01` are responses from the ACE module.

In the `File` menu, hit `Log to file...` and `OK`. Then, wait a bit to get traffic. I expect the IQ2020 will ask the ACE module for a status update every 10 to 20 seconds or so. Once you get some data, start changing the ACE settings on the hot tub remote. If you can, there is a `Annotate Log...` option in the `Actions` menu. Write something like "Changing power to level 4" and hit ok, then do that operation and wait for things to settle.

Basically, your creating a annotated log of what you are doing as the log is recorded. Once done, send the log over. I would be able to figure things out. Any status you know about the ACE system would also be appreciated. If you ever get an error with the ACE system, record the status and send it over with the error message, etc. That way, we can improve our knowledge of how ACE works.

Thanks.

npinguin, 2024-06-16:
Thanks will connect my pc tomorrow and do the capturing. Your work motivates me to start an ESP32 device for our ventilation system

Ylianst, 2024-06-16:
@npinguin nice. I built the integration with a WIFI packet capture system built-in and I am glad I did. Once enabled, it makes things so easy. I don't have to be anywhere near the hot tub and I can debug things.

npinguin, 2024-06-17:
[logfile.txt](https://github.com/user-attachments/files/15875726/logfile.txt)

Find attached to log file

Did all the commands, 
- testing the water (in the green)
- boost on / off
- output up to 10 and down to 0
Hope you can decode the traffice, let me know if you need additional info

Ylianst, 2024-06-17:
@npinguin Wow, that is a wonderful log. Very clear and yes, I can see the power level setting right away. In the next few days I will add support for it. I am still wondering what is the appropriate control in home assistant for 0 to 10 output level. I think I will use the fan control with 10 levels a bit like I did for the jets.

For the boost, do you know if it auto-turned off after 5 minutes? Not that it makes any difference, but just curious.

Ylianst, 2024-06-17:
For my reference, I found what the remote control UI looks like for ACE.

[image]

npinguin, 2024-06-17:
Wow that is exactly how my screen looks like now ;-)

As for the boost do not know if it turns off after a while, because I never used it so far.

thank you for the great integration and support!

Ylianst, 2024-06-19:
So, I got a lot more information on the protocol decoding on the other discussion thread. I am working a bit on that and will get back to ACE shortly.

Ylianst, 2024-06-19:
@npinguin - Just updated the integration, the `Salt System Power` sensor should now display the ACE power state. Controlling it will come later, but wanted to make sure this part works.

[image]

npinguin, 2024-06-19:
just tried it, but does not seem to work for me

[image]

Ylianst, 2024-06-19:
Ok, let me check this for a bit.

Ylianst, 2024-06-19:
Fixed it. You may need to use `refresh` in the config like this to force the update. Let me know if it works.

```
external_components:
  - source: github://ylianst/esp-iq2020
    refresh: 30s
```

npinguin, 2024-06-19:
I do not  manage to compile it anymore

[image]
[image]

Ylianst, 2024-06-19:
Oh dear! Working on that now.

Ylianst, 2024-06-19:
Made that fix... trying it now.

Ylianst, 2024-06-19:
Oh. I see why I did not catch that. It should work now.

npinguin, 2024-06-19:
It works perfectly ! 

Tested changing values to 10 and down to 0, all perfectly.

are you able to figure out the measurements of the salt system ?

Ylianst, 2024-06-19:
@npinguin Nice! Thanks for reporting back. I will add a way to control the 0 to 10 power value remotely at some point. I looked at the data and it's not clear if there are any measurements at all. The "boost" does not cause any immediate reaction in the data and the "test" button does not seem to do anything at all (at least nothing I can see going towards the ACE module).

I looked at the hub decoding logs from the other discussion thread and in the case of ACE/Freshwater, the connection hub does not seem to even try to decode anything at all. So, yes, super weird.

Now, I did add a new hidden feature in the integration, the ACE emulator. When enabled, it makes it look like a ACE system is installed. I will unplug my Freshwater module later today and enable this.

```
iq2020:
   uart_id: SpaConnection
   port: 1234
   ace_emulation: true         <---
```

I also added a new sensor to detect when the "Boost" is enabled or disabled. So, if all goes well, I hope to use this to control lights from the remote. I don't use the Freshwater module anyway. Also, I could set any value I want in emulation mode and see if the connection kit indicates anything, so, I could find values this way.

npinguin, 2024-06-19:
Just some speculation not sure, but looks to me there is a regular status update in the messages, maybe it holds some info that are regularly required for the display.
So test and boost might just do something but do not get a feedback

npinguin, 2024-06-19:
Look these statuses look to be interlinked and probably encoded in one or more status messages

`'mz_POL_light_intensity': 0, 
'mz_POL_color_state': 1, 
'mz_Exterior_light_intensity': 0, 
'mz_Accent_loop_speed': 0, 
'mz_24_hour_light_status': 1, 
'mz_Accent_wheel_status': 0, 
'mz_Main_loop_speed': 0, 
'mz_Main_wheel_status': 0, 
'mz_Main_light_intensity': 0, 
'mz_Accent_light_intensity': 0, 
'mz_model': 0, 
'mz_Exterior_color_state': 1, 
'mz_system_status': 0, 
'mz_Exterior_loop_speed': 0, 
'mz_version': 0, 
'mz_POL_wheel_status': 0, 
'mz_Accent_color_state': 1, 
'mz_POL_loop_speed': 0, 
'mz_Main_color_state': 1, 
'mz_Exterior_wheel_status': 0

'audio_radio_signal_strength': 0, 
'audio_play_pause_status': 0, 
'audio_power': 0, 
'audio_wireless_channel': 1, 
'audio_source_selection': 4, 
'audio_balance': 0, 
'audio_treble': 0, 
'audio_bass': 0, 
'audio_bluetooth_pairing': 0, 
'audio_subwoofer_volume': 11, 
'audio_volume': 25

'heat_pump_current_operational_status': 255,
'heat_pump_current_mode_setting': 255

dosing_state': 3,
'g3_sensor_data': 255,
'g3_chlor_test_data': 255,
'spa_size': 2,
'spa_usage': 4,
'g3_clrmtr_test_data': 0,
'salline_test': 72,
'g3_level2_errors': 0,
'g3_sensor_status': 255,
'g3_ph_test_data': 67 `

Ylianst, 2024-06-19:
Actually, looking at the logs again, I did find the decoding ACE/Freshwater data block... the `1E03` block of data.

```
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:buffer_to_parse 1C 1F 01 14 80 1E 03 00 03 FF FF 00 FF 00 68 00 FF 07 FF 00 00 B3 10 00 40 BA
Jun  3 02:20:01 local7.info pylog: parsing status payload ready
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:1E03
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:Handling Tanas Return Data Object/Command Response: 00 03 FF FF 00 FF 00 68 00 FF 07 FF 00 00 B3 10 00 40
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:channels_to_update: {'dosing_state': 0, 'g3_sensor_data': 255, 'g3_chlor_test_data': 255, 'spa_size': 3, 'spa_usage': 0, 'g3_clrmtr_test_data': 0, 'salline_test': 7, 'g3_level2_errors': 0, 'g3_sensor_status': 255, 'g3_ph_test_data': 0}
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:updating channel: dosing_state, with new value 0
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:updating channel: g3_sensor_data, with new value 255
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:updating channel: g3_chlor_test_data, with new value 255
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:updating channel: spa_size, with new value 3
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:updating channel: spa_usage, with new value 0
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:updating channel: g3_clrmtr_test_data, with new value 0
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:updating channel: salline_test, with new value 7
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:updating channel: g3_level2_errors, with new value 0
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:updating channel: g3_sensor_status, with new value 255
Jun  3 02:20:01 local7.info pylog: DEBUG:eagle:updating channel: g3_ph_test_data, with new value 0
```

Ylianst, 2024-06-19:
In the DataViewer, you should be able to sent this command `01 1F 40 1E03` and the your ACE status back.

[image]

In my case, I have get this:

```
 --> 01 1F 40 1E03
 <-- 1F 01 80 1E030003FFFF01FF006808FF0BFF0000B3100040
```

I should be able to decode this now.

Ylianst, 2024-06-19:
This `1E030003FFFF00FF006800FF07FF0000B3100040` decodes to this:

dosing_state, with new value 0
g3_sensor_data, with new value 255
g3_chlor_test_data, with new value 255
spa_size, with new value 3
spa_usage, with new value 0
g3_clrmtr_test_data, with new value 0
salline_test, with new value 7
g3_level2_errors, with new value 0
g3_sensor_status, with new value 255
g3_ph_test_data, with new value 0

Ylianst, 2024-06-19:
I sent a made up packet on the bus and the hub decoded it.

```
Jun 19 19:29:50 local7.info pylog: DEBUG:eagle:received message from: [00:13:a2:00:41:a6:0e:5b]!, buf is 1C 1F 01 14 80 1E 03 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 7F.
Jun 19 19:29:50 local7.info pylog: DEBUG:eagle:buffer_to_parse 1C 1F 01 14 80 1E 03 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 7F
Jun 19 19:29:50 local7.info pylog: parsing status payload ready
Jun 19 19:29:50 local7.info pylog: DEBUG:eagle:1E03
Jun 19 19:29:50 local7.info pylog: DEBUG:eagle:Handling Tanas Return Data Object/Command Response: 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12
Jun 19 19:29:51 local7.info pylog: DEBUG:eagle:channels_to_update: {'dosing_state': 5, 'g3_sensor_data': 3, 'g3_chlor_test_data': 12, 'spa_size': 2, 'spa_usage': 1, 'g3_clrmtr_test_data': 14, 'salline_test': 11, 'g3_level2_errors': 7, 'g3_sensor_status': 4, 'g3_ph_test_data': 13}
Jun 19 19:29:51 local7.info pylog: DEBUG:eagle:updating channel: dosing_state, with new value 5
Jun 19 19:29:51 local7.info pylog: DEBUG:eagle:updating channel: g3_sensor_data, with new value 3
Jun 19 19:29:51 local7.info pylog: DEBUG:eagle:updating channel: g3_chlor_test_data, with new value 12
Jun 19 19:29:51 local7.info pylog: DEBUG:eagle:updating channel: spa_size, with new value 2
Jun 19 19:29:51 local7.info pylog: DEBUG:eagle:updating channel: spa_usage, with new value 1
Jun 19 19:29:51 local7.info pylog: DEBUG:eagle:updating channel: g3_clrmtr_test_data, with new value 14
Jun 19 19:29:51 local7.info pylog: DEBUG:eagle:updating channel: salline_test, with new value 11
Jun 19 19:29:51 local7.info pylog: DEBUG:eagle:updating channel: g3_level2_errors, with new value 7
Jun 19 19:29:51 local7.info pylog: DEBUG:eagle:updating channel: g3_sensor_status, with new value 4
Jun 19 19:29:51 local7.info pylog: DEBUG:eagle:updating channel: g3_ph_test_data, with new value 13
```

Ylianst, 2024-06-19:
This `1E03 0102030405060708090A0B0C0D0E0F101112` decodes to:

```
dosing_state, with new value 5
g3_sensor_data, with new value 3
g3_chlor_test_data, with new value 12
spa_size, with new value 2
spa_usage, with new value 1
g3_clrmtr_test_data, with new value 14
salline_test, with new value 11
g3_level2_errors, with new value 7
g3_sensor_status, with new value 4
g3_ph_test_data, with new value 13
```

npinguin, 2024-06-19:
looking good, just send the command

 --> 01 1F 40 1E03
 <-- 1F 01 80 1E0303FFFFFF01FF0000FFFF34FFFFFFFFFFFFFF

and my bluerioth has the following measured water values
[image]

Ylianst, 2024-06-19:
So, your data `1E0303FFFFFF01FF0000FFFF34FFFFFFFFFFFFFF` would decode like this:

```
1E03
03 - spa_usage (3 of 10)
FF - spa_size -1
FF - g3_sensor_data -1
FF - g3_sensor_status -1
01 - dosing_state 1
FF - ?
00 - g3_level2_errors
00FFFF - ?
34 - salline_test 52
FF - g3_chlor_test_data -1
FF - g3_ph_test_data -1
FF - g3_clrmtr_test_data -1
FFFFFFFF - ?
```

The only interesting thing is the 52 for salline_test. I don't salt my tub and have the value of 11.

npinguin, 2024-06-19:
indeed which makes sense since it only needs to measure the salt level to see if there is enough salt for the ace cell to work

next question is what are the ranges for these levels ? currently it is in the green looking at the display bit of the middle about 40%

Ylianst, 2024-06-19:
@npinguin I am figuring out that Freshwater and ACE exactly the same data, but just different command numbers. This makes everything easy. I could try to raise the value of the salt test until the hub rejects the value... maybe that will give me a hit as to what the range is. Freshwater and ACE are just electrodes with current passing thru it, so, I am guessing they measure the current resistance... the more salt, the less the resistance since salt helps conduct electricity. However, just a guess.

Freshwater cartridges are very expensive, so, I just use bromine instead. At first I was under the impression ACE was something completely different, but now it seems it's also a salt system. ACE cartridges last a lot longer than Freshwater ones. I wonder if the intent was to make it a subscription service?

npinguin, 2024-06-19:
so next step would be automatic bromine feeder based on esp32 ;-)

Ylianst, 2024-06-19:
LOL! That's funny. No, I am good. Sometimes it's best to keep it simple.

Ylianst, 2024-06-19:
Updated the integration. Add the `salt_content` sensor and you can start tracking it long term and see how it goes.

```
sensor:
    salt_power:
      name: Salt System Power
    salt_content:
      name: Salt Content
```

Ylianst, 2024-06-23:
Ok. I just competed the ACE/Freshwater integration. The `salt_power` sensor in the `sensor` section no longer exists, instead there is the new `salt_power` in the `number` section like this:

```
sensor:
    salt_content:
      name: Salt Content

number:
  - platform: iq2020-dev
    id: salt_power
    name: Salt System Power
    datapoint: 5
```

This allows you to both see and control the 0 to 10 level of the salt system. Let me know if it works.

Ylianst, 2024-06-23:
Ok. I will close this since ACE support should now work. Open a new issue if something does not work right.

closed 2024-06-23 by Ylianst