Ylianst/ESP-IQ2020 #7: Add data polling rate adjustment from ESP32 to Home Assistant

itry2fly, 2024-06-21:
Feature request to adjust data polling rate from default ~60 seconds between Spa controller/ESP32 and Home Assistant.  Would like HA to be able to reflect spa jet speed/light/etc changes made from spa side remote control sooner.   Not all data would need to be polled as often (lifetime counters, etc.) , is it possible to poll more important data (Jet status, Lights, Set Temp) more frequently?

Ylianst, 2024-06-21:
Excellent suggestion, I will add this.

Ylianst, 2024-06-21:
Done. Add `polling_rate` with the number of seconds in the `iq2020` section of the ESP-Home config. The minimum value is 5. Let me know if that works.

```
iq2020:
   uart_id: SpaConnection
   polling_rate: 65                          <----
```

itry2fly, 2024-06-21:
Thanks!, However I am getting an invalid option error with the added line:
[image]

If I add another indent to fix, I receive this error on line 1:
[image]

Ylianst, 2024-06-21:
Use `refresh` in the config like this to force the update. Let me know if it works.

```
external_components:
  - source: github://ylianst/esp-iq2020
    refresh: 30s
```

If you don't do this, ESP Home will only check for integration update one a day. Once you got it, comment out the `refresh` line. Let me know if that works.

Ylianst, 2024-06-23:
Closing this since the feature was added. Open a new issue if needed.

closed 2024-06-23 by Ylianst