Ylianst/ESP-IQ2020 #8: Pumps power and heater power reading 0 W

rossbrouse, 2024-06-22:
I've noticed that it's not pulling any data for pumps power or heater power. Is that expected>
Here is that section from my config:

    power_on_counter:
      name: Power On Counter
    power_l1:
      name: Pumps Power
#    power_heater:
#      name: Power Heater
    power_l2:
      name: Heater Power

Ylianst, 2024-06-22:
Hi. These values are polled along with the other values in one go, I don't get to pick not to poll them. This said, various versions of the IQ2020 seem to offer different values. What model hot tub do you have?

rossbrouse, 2024-06-22:
It's a 2024 Hotspring relay and uses the IQ2020 Watkins 1303401-1 controller.

Ylianst, 2024-06-22:
Ok, add the following sensors and let me know what you see when the heaters and pumps are on:

```
sensor:
  - platform: iq2020-dev
    voltage_l1:
      name: Voltage L1
    voltage_heater:
      name: Voltage Heater
    voltage_l2:
      name: Voltage L2
    current_l1:
      name: Current L1
    current_heater:
      name: Current Heater
    current_l2:
      name: Current L2
```

You should see the added sensors here:

[image]

rossbrouse, 2024-06-23:
I can't recompile and install using the iq2020-dev platform. Is using the iq2020-dev platform required?

In file included from src/esphome.h:33,
                 from src/main.cpp:3:
src/esphome/components/iq2020-dev/iq2020.h:52:7: error: redefinition of 'class IQ2020Component'
 class IQ2020Component : public esphome::Component {
       ^~~~~~~~~~~~~~~
In file included from src/esphome.h:26,
                 from src/main.cpp:3:
src/esphome/components/iq2020/iq2020.h:52:7: note: previous definition of 'class IQ2020Component'
 class IQ2020Component : public esphome::Component {

rossbrouse, 2024-06-23:
Look like Current L2 went up to 15 amps when Jets 1 turned on as a result of the heater turning on. Current L1 still registers only 0 watts.

Ylianst, 2024-06-23:
Apologies. Yes, always use `iq2020` and not `iq2020-dev`, that is my own development version so I can test things without breaking the "production" version. You did the right thing. It does not look like you have a power meter. If you really want to investigate this further, add the following line to your config:

```
# Relay UART messages over TCP
iq2020:
   uart_id: SpaConnection
   port: 1234                               <---------- Add this line
```

Re-flash the ESP32 device then download the [Windows Data Viewer app](https://github.com/Ylianst/ESP-IQ2020/raw/main/DataViewer/IQ2020-DataViewer.exe) and connect it to the ESP32 using the IP address and port 1234. You will start seeing traffic like this:

[image]

You can then go in the "file" menu and select "log to file..." and start a capture. In "Actions" select "Ask Status (x0256)" and send the results back here. Basically, that will give us a complete status of your device and I will be able to decode all the values. You can also go in the "state" panel and see the entire decoded state:

[image]

joshs85, 2024-06-23:
I have the same issue with  112T b5e1A002 v3.0.  It's a hot spring 4 person hot tub by watkins.  I can try to get you some debug logs.. everything else seems to work except for current and power measurements.

Ylianst, 2024-06-23:
One thing I am thinking off is that if the voltage and current values are both not zero, but the power is zero, I will fill in the power and volts x current. It's not going to be as precise of a value as having the power meter, but it will be in the right ballpark.

rossbrouse, 2024-06-23:
I'll try this once I get my system back online. Surely coincidental but after running the windows app my HA web interface will no longer load. As soon as I connected on tcp port 1234, the Windows app would crash. I tried 3 or 4 times and since then I have not been able to get my system to load unless it's in recovery mode.

Ylianst, 2024-06-23:
@rossbrouse Oh dear. I am not sure what is going on there. Port 1234 is just a plan TCP port where all RS485 traffic is forward to the port and anything you send to port 1234 gets sent on the RS485 port. So, any generic TCP application can connect to port 1234 and see the RS485 traffic. The DataViewer app is just a basic C# app I built to connect to that port and decode the packets. It should work get fine while HA is running. Not sure what is going on.

Ylianst, 2024-06-23:
Ok, I just updated the IQ2020 integration to compute power as voltage*current if power is zero. Let me know if this works. However, getting a capture would be really nice.

rossbrouse, 2024-06-24:
I got my HA back online and updated my hot-tub.yaml file with your latest version. I will try the capture again later today and report back.

Ylianst, 2024-07-02:
I will close this once since no new data was provided. However, please re-open as needed.

closed 2024-07-02 by Ylianst