Ylianst/ESP-IQ2020 #10: Lifetime values spa and circulation pump seem to be wrong

npinguin, 2024-06-24:
Hi 

Looks like the circulation pump total lifetime values are not correct. spa is of 2017, hotspring envoy, fw 2.02

[image]

[image]

Ylianst, 2024-06-24:
@npinguin I am not sure there is anything I can do about this. We could look at the raw status (0x0256), you may have submitted it on a different issue? If that is the reported value in the raw command, I am not sure what correctly can be applied to it.

npinguin, 2024-06-24:
I will update the firmware later this week to 4.05 let’s see what that brings

Ylianst, 2024-06-24:
@npinguin Oh! Please take note of any changes. So far, this integration seems to work ok on all firmware versions, but would be nice to spot any differences.

Ylianst, 2024-06-29:
I will close this one since I imagine there is no change and nothing we can really do about this. However, please re-open if there is any additional information.

closed 2024-06-29 by Ylianst