Ylianst/ESP-IQ2020 #9: Extra Config

npinguin, 2024-06-24:
Might be usefull if we share our configs and improvements like that the best elements could be merged in the overall config ?

My config so far includes all sensors i could find and  config for the web server like that you could run it without home assistant

```
esphome:
  name: jacuzzi
  friendly_name: Hotspring Envoy 2017
  comment: "Luxury Spa"

esp32:
  board: m5stack-atom

logger:
  baud_rate: 0
  level: ERROR

# Enable Home Assistant API
api:
  encryption:
    key: "....."

ota:
  - platform: esphome
    password: "....."

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Jacuzzi Fallback Hotspot"
    password: "1231231231"

captive_portal:

external_components:
  - source: github://ylianst/esp-iq2020
    refresh: 30s

uart:
  id: SpaConnection
  tx_pin: GPIO26
  rx_pin: GPIO32
  baud_rate: 38400

iq2020:
   uart_id: SpaConnection
#   flow_control_pin: GPIO0
   port: 1234
   audio_emulation: true

# Example configuration entry
web_server:
  port: 80

button:
  - platform: restart
    name: "ESP Restart"
    
binary_sensor:    
  - platform: status
    name: ESP Status
  
# If using celsius units on the hot tub remote, replace _f_ with _c_ in the three entries below.
# Feel free to remove any sensor that are not relevent for your hot tub.
sensor:
  - platform: iq2020
    current_c_temperature:
      name: Current Temperature
    target_c_temperature:
      name: Target Temperature
    outlet_c_temperature:
      name: Heater Outlet
    heater_total_runtime:
      name: Heater Runtime
    jets1_total_runtime:
      name: Jets 1 Runtime
    lifetime_runtime:
      name: Lifetime Runtime
    jets2_total_runtime:
      name: Jets 2 Runtime
#    jets3_total_runtime:
#      name: Jets 3 Runtime
    lights_total_runtime:
      name: Lights Runtime
    power_on_counter:
      name: Power On Counter
#    salt_power:
#      name: Salt System Power
    circulation_pump_total_runtime:
      name: Circulation Pump Runtime
#    jet1_low_total_runtime:
#      name: Jets 1 Low Runtime
    jet2_low_total_runtime:
      name: Jets 2 Low Runtime
    power_l1:
      name: Pump Power
    power_heater:
      name: Small Heater Power
    power_l2:
      name: Heater Power
    voltage_l1:
      name: Voltage L1
    voltage_heater:
      name: Voltage Heater
    voltage_l2:
      name: Voltage L2
    current_l1:
      name: Current L1
    current_heater:
      name: Current Heater
    current_l2:
      name: Current L2
    pcb_c_temperature:
      name: Controller Temperature
    salt_content:
      name: Salt Content
    logo_lights:
      name: Logo Lights
    audio_buttons:
      name: Audio Buttons
    lights_intensity:
      name: Light Intensity
    lights_underwater_intensity:
      name: Underwater Light Intensity
    lights_bartop_intensity:
      name: Bartop Light Intensity
    lights_pillow_intensity:
      name: Pillow Light Intensity
    lights_exterior_intensity:
      name: Exterior Light Intensity
    lights_underwater_color:
      name: Underwater Light Color
    lights_bartop_color:
      name: Bartop Light Color
    lights_pillow_color:
      name: Pillow Light Color
    lights_exterior_color:
      name: Exterior Light Color

  - platform: wifi_signal # Reports the WiFi signal strength/RSSI in dB
    name: "ESP WiFi Signal dB"
    id: wifi_signal_db
    update_interval: 60s
    entity_category: "diagnostic"
  - platform: copy # Reports the WiFi signal strength in %
    source_id: wifi_signal_db
    name: "ESP WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"
    device_class: ""
  - platform: uptime
    name: ESP Uptime Sensor
  - platform: internal_temperature
    name: "ESP Internal Temperature"
  - platform: template
    id: esp_memory
    icon: mdi:memory
    name: ESP Free Memory
    lambda: return heap_caps_get_free_size(MALLOC_CAP_INTERNAL) / 1024;
    unit_of_measurement: 'kB'
    state_class: measurement
    entity_category: "diagnostic"

select:
  - platform: iq2020
    name: Audio Source
    id: audio_source
    datapoint: 0

number:
  - platform: iq2020
    name: Volume
    id: audio_volume
    datapoint: 0
  - platform: iq2020
    name: Tremble
    id: audio_tremble
    datapoint: 1
  - platform: iq2020
    name: Bass
    id: audio_bass
    datapoint: 2
  - platform: iq2020
    name: Balance
    id: audio_balance
    datapoint: 3
  - platform: iq2020
    name: Subwoofer
    id: audio_subwoofer
    datapoint: 4
  - platform: iq2020
    id: salt_power
    name: Salt System Power
    datapoint: 5

text:
  - platform: iq2020
    name: Song Title
    id: song_title
    datapoint: 0
    mode: text
    value: "Home Assistant"
  - platform: iq2020
    name: Artist Name
    id: artist_name
    datapoint: 1
    mode: text
    value: "Remote Control"
    
switch:
  - platform: iq2020
    name: Lights
    id: lights_switch
    icon: "mdi:lightbulb"
    datapoint: 0
  - platform: iq2020
    name: Spa Lock
    id: spa_lock_switch
    icon: "mdi:lock"
    datapoint: 1
  - platform: iq2020
    name: Temperature Lock
    id: temp_lock_switch
    icon: "mdi:lock"
    datapoint: 2
  - platform: iq2020
    name: Clean Cycle
    id: clean_cycle_switch
    icon: "mdi:vacuum"
    datapoint: 3
  - platform: iq2020
    name: Summer Timer
    id: summer_timer_switch
    icon: "mdi:sun-clock"
    datapoint: 4

fan:
  - platform: iq2020
    name: Jets 1
    id: jets1
    icon: "mdi:turbine"
    datapoint: 0
    speeds: 1
  - platform: iq2020
    name: Jets 2
    id: jets2
    icon: "mdi:turbine"
    datapoint: 1
    speeds: 2

# Set "celsius" to "true" if using celsius units.
climate:
  - platform: iq2020
    name: Temperature
    celsius: true

text_sensor:
  - platform: iq2020
    versionstr:
      name: Version
  # Retrieve ESPHome version information
  - platform: version
    name: "ESP Device Version"
    id: device_esphome_version
    internal: true
    icon: 'mdi:chevron-right'
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
    ssid:
      name: ESP Connected SSID
    mac_address:
      name: ESP Mac Wifi Address
    scan_results:
      name: ESP Latest Scan Results
    dns_address:
      name: ESP DNS Address
```

Ylianst, 2024-06-24:
@npinguin Oh wow! Thanks for posting this. That is some expert level configuration. Yesterday I simplified the base config on the main page and pushed a lot of sensors on the extras page. I sort of want people to start with the basics and not overload the main page... however, this full config is really amazing. Maybe I should turn this into a "full configuration"?

npinguin, 2024-06-24:
Thanks yes please do. 

I think it could simplify things for everyone 

you could put nearly all in a central yml and include it and only set the substitude variables that are needed

Ylianst, 2024-06-24:
Ok. Thanks for this. Yes, this is a great idea. For my own day-to-day use, I actually use a minimal amount of sensors, but having the full set is great. I will work on this in the next few days.

Ylianst, 2024-06-25:
I just added your [configuration in this documentation file](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/configuration.md).

npinguin, 2024-06-26:
I noticed that the esp uptime sensor is in seconds but home assistant displays it nicely in hours:minutes:seconds…

so maybe the runtime sensors could use the same date type like that no conversion to text is needed anymore

Ylianst, 2024-06-26:
@npinguin Oh. Interesting. Yes, that would be the best. I will search around and see what type they use.

Ylianst, 2024-06-27:
@npinguin I found the attributes of the esp-home update and applied it to lifetime counters on my dev version. I added:

```
    state_class=STATE_CLASS_TOTAL_INCREASING,
    device_class=DEVICE_CLASS_DURATION,
```

I now get this below. Let me know if this is what you expected. I can move that to the production version.

[image]

npinguin, 2024-06-27:
Looks good!

Ylianst, 2024-06-27:
@npinguin Thanks. It's now in the official `IQ2020` integration.

closed 2024-06-27 by Ylianst