Ylianst/ESP-IQ2020 #11: Conflict of some form with audio mod.

D3nkNugZ, 2024-06-27:
Hey there!

So I added the required lines to my yaml as per the documentation and this is the error I've been spat out:

```
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.4
Compiling .pioenvs/spa-control/src/main.cpp.o
In file included from src/esphome.h:33,
                 from src/main.cpp:3:
src/esphome/components/iq2020-dev/iq2020_climate.h:10:8: error: redefinition of 'class esphome::iq2020_climate::IQ2020Climate'
  class IQ2020Climate : public climate::Climate, public Component {
        ^~~~~~~~~~~~~
In file included from src/esphome.h:26,
                 from src/main.cpp:3:
src/esphome/components/iq2020/iq2020_climate.h:10:8: note: previous definition of 'class esphome::iq2020_climate::IQ2020Climate'
  class IQ2020Climate : public climate::Climate, public Component {
        ^~~~~~~~~~~~~
In file included from src/esphome.h:34,
                 from src/main.cpp:3:
src/esphome/components/iq2020-dev/iq2020_fan.h:10:8: error: redefinition of 'class esphome::iq2020_fan::IQ2020Fan'
  class IQ2020Fan : public fan::Fan, public Component {
        ^~~~~~~~~
In file included from src/esphome.h:27,
                 from src/main.cpp:3:
src/esphome/components/iq2020/iq2020_fan.h:10:8: note: previous definition of 'class esphome::iq2020_fan::IQ2020Fan'
  class IQ2020Fan : public fan::Fan, public Component {
        ^~~~~~~~~
In file included from src/esphome.h:37,
                 from src/main.cpp:3:
src/esphome/components/iq2020-dev/iq2020_switch.h:9:8: error: redefinition of 'class esphome::iq2020_switch::IQ2020Switch'
  class IQ2020Switch : public switch_::Switch, public Component {
        ^~~~~~~~~~~~
In file included from src/esphome.h:30,
                 from src/main.cpp:3:
src/esphome/components/iq2020/iq2020_switch.h:9:8: note: previous definition of 'class esphome::iq2020_switch::IQ2020Switch'
  class IQ2020Switch : public switch_::Switch, public Component {
        ^~~~~~~~~~~~
*** [.pioenvs/spa-control/src/main.cpp.o] Error 1
========================= [FAILED] Took 16.20 seconds =========================
```

My yaml looks as such: 

```
esphome:
  name: spa-control
  friendly_name: Spa Control

esp32:
  board: m5stack-atom

logger:
  baud_rate: 0
  level: ERROR

# Enable Home Assistant API
api:
  encryption:
    key: "yoloswaggins"

ota:
  password:"xxxo.oxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source: github://ylianst/esp-iq2020

# Make sure tx/rx pins are correct for your device.
# GPIO26/32 is ok for M5Stack-ATOM + Tail485, look in GitHub devices link for your device.
uart:
  id: SpaConnection
  tx_pin: GPIO26
  rx_pin: GPIO32
  baud_rate: 38400

iq2020:
   uart_id: SpaConnection
   audio_emulation: true
   polling_rate: 65
   port: 1234

select:
  - platform: iq2020-dev
    name: Audio Source
    id: audio_source
    datapoint: 0

number:
  - platform: iq2020-dev
    name: Volume
    id: audio_volume
    datapoint: 0
  - platform: iq2020-dev
    name: Tremble
    id: audio_tremble
    datapoint: 1
  - platform: iq2020-dev
    name: Bass
    id: audio_bass
    datapoint: 2
  - platform: iq2020-dev
    name: Balance
    id: audio_balance
    datapoint: 3
  - platform: iq2020-dev
    name: Subwoofer
    id: audio_subwoofer
    datapoint: 4

text:
  - platform: iq2020-dev
    name: Song Title
    id: song_title
    datapoint: 0
    mode: text
    value: "Home Assistant"
  - platform: iq2020-dev
    name: Artist Name
    id: artist_name
    datapoint: 1
    mode: text
    value: "Remote Control"

# If using celsius units on the hot tub remote, replace _f_ with _c_ in the three entries below.
# Feel free to remove any sensor that are not relevent for your hot tub.
sensor:
  - platform: iq2020
    current_c_temperature:
      name: Current Temperature
    target_c_temperature:
      name: Target Temperature
    outlet_c_temperature:
      name: Heater Outlet
    power_l1:
      name: Pumps Power
    power_heater:
      name: Power Heater
    power_l2:
      name: Heater Power
    pcb_c_temperature:
      name: Controller Temperature
    audio_buttons:
      name: Audio Buttons

switch:
  - platform: iq2020
    name: Lights
    id: lights_switch
    icon: "mdi:lightbulb"
    datapoint: 0
  - platform: iq2020
    name: Spa Lock
    id: spa_lock_switch
    icon: "mdi:lock"
    datapoint: 1
  - platform: iq2020
    name: Temperature Lock
    id: temp_lock_switch
    icon: "mdi:lock"
    datapoint: 2
  - platform: iq2020
    name: Clean Cycle
    id: clean_cycle_switch
    icon: "mdi:vacuum"
    datapoint: 3
  - platform: iq2020
    name: Summer Timer
    id: summer_timer_switch
    icon: "mdi:sun-clock"
    datapoint: 4

fan:
  - platform: iq2020
    name: Jets 1
    id: jets1
    icon: "mdi:turbine"
    datapoint: 0
    speeds: 1
  - platform: iq2020
    name: Jets 2
    id: jets2
    icon: "mdi:turbine"
    datapoint: 1
    speeds: 2

# Set "celsius" to "true" if using celsius units.
climate:
  - platform: iq2020
    name: Temperature
    celsius: false

text_sensor:
  - platform: iq2020
    versionstr:
      name: Version

```

Ylianst, 2024-06-27:
Replace all `iq2020-dev` with `iq2020` and it will fix it. I will go thru my documentation and fix this. I personally use `iq2020-dev` for my own development, but it should not be used by anyone else.

Ylianst, 2024-06-27:
Thanks. I just went over my documentation and fixed it.

closed 2024-06-27 by Ylianst