Ylianst/ESP-IQ2020 #13: Discussion

Ylianst, 2024-07-27:


elitemike, 2024-08-02:
I got my hardware yesterday and got it installed today.  Now I need to work on the dashboard. So far so good otherwise.

I have a new Dell xps 13 snapdragon machine.  Support for the Atom lite esphome didn't seem to exist. I didn't put any effort into making it work so I just used a different machine.  I run HA in docker so I ended up using the docker ESPHome dashboard to get it working.  

Thanks for all you have done with this

Ylianst, 2024-08-03:
Hi @elitemike. Nice, glad it's working for you. I did not think ESP-Home was picky so to what Home Assistant it could run on, this is new to me. I have a Intel NUC x86 and just figured it would work on anything.

For the Freshwater System, I do have it but I do not use it. I do not currently have a way to "acknowledge the output level", I did not know that was a thing. Basically, the integration can't push button on the remote so to speak. So, even if a proper command was sent, if the remote needs something, I am not sure it would get the remote acknowledge. That said, I am not sure I understand this area much.

elitemike, 2024-08-03:
I'm the one that commented on your short video 10 days ago by the way.  I have a 2023 Envoy with FWS so you can mark that as another tested unit.   Once I get some time, I may fork your project to experiment a bit.  I'm a .Net developer primarily and I'm the guy that does a lot of front end work on a project where looks don't matter much.  I have some Python experience.  

Is there a way to kill the power to both jets and lights using your integration, or would I just create a simple automation?

Ylianst, 2024-08-04:
Hi @elitemike. Nice to hear back from you. If you are a .NET dev, you may want to take a look [at the way debug the RS485 traffic](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/debugging.md), I use a C# application to do that. You can fully control the hot tub from a C# app if you like, it's not difficult once you enable the ESP32 traffic relay system.

Yes, you can turn on/off both lights and jets, generally you would create an automation and do it that way.

afp316, 2024-08-10:
- Hot Tub Model
Hotsprings Limelight Flash (2021)

- IQ2020 Enclosure Internals
[image]

- Yaml Changes Made in ESPHome
```
fan:
  - platform: iq2020
    name: Jets 1
    id: jets1
    icon: "mdi:turbine"
    datapoint: 0
    speeds: 2 #to 2 from 1
  - platform: iq2020
    name: Jets 2
    id: jets2
    icon: "mdi:turbine"
    datapoint: 1
    speeds: 1 #to 1 from 2
```

- Result
Jets 1 has off/low/high speeds
Jets 2 has off/high speeds

Figured I'd share my model specifics.

Thanks!

derhally, 2024-08-10:
> Hi @elitemike. Nice, glad it's working for you. I did not think ESP-Home was picky so to what Home Assistant it could run on, this is new to me. I have a Intel NUC x86 and just figured it would work [...]

I could never get the the Hot Spring app to confirm the salt setting after 15 days, but that doesn't mean it is impossible.  I just added an automation in HA to send me a notification when the "Logo lights" value switches to 1 to remind me to go out and press the button.

derhally, 2024-08-10:
@Ylianst what is needed to get the power entities to show up in the HA Energy dashboard?  The entities don't show up in the entity picker.

Ylianst, 2024-08-11:
@derhally Interesting point, I have not looked into this.

jdaleyftp, 2024-08-16:
@Ylianst I have a goal to sync the hottub light state with other outdoor lights. Rather than waiting for a poll (and have a delay) I am wondering if there is an event that can trigger a "poll"/update out of the ESP.  I am going to fire up the remote packet monitor this weekend, but do you know if there are any packets that send when a button is pressed on the unit (or equivalent).

jdaleyftp, 2024-08-16:
> @Ylianst what is needed to get the power entities to show up in the HA Energy dashboard? The entities don't show up in the entity picker.

The requirements to show up in the energy dashboard is for the device to have the following attributes.
```yaml
device_class: power
state_class: measurement
```
It looks like it has state_class, but i think it might need "device_class: power"

I dont know ESPHome that well yet, but another option is to add this to your customize.yaml in HA, which lets you modify the entitiy outside of the integration
```yaml
sensor.living_room_circuit:
  device_class: power
  state_class: measurement 
  ```

Ylianst, 2024-08-18:
Ok, I think I have a solution for displaying energy usage. The tub tub will send back how much power it's using in Watts, but you need to integrate this over time and ESPHome has helper to do this, [total_daily_energy](https://esphome.io/components/sensor/total_daily_energy.html). It will add up the wattage used over time and give you energy usages in KWh that you can then use in the energy panel.

The `sensor` section of my hot tub looked like this:

```
sensor:
  - platform: iq2020-dev
    current_f_temperature:
      name: Temperature
    target_f_temperature:
      name: Set Temperature
    outlet_f_temperature:
      name: Heater Outlet
    power_l1:
      name: Pump Power
    power_heater:
      name: Controller Power
    power_l2:
      name: Heater Power
    pcb_f_temperature:
      name: Controller Temperature
    buttons:
      name: Buttons
```

I changed it to this:

```
sensor:
  - platform: iq2020-dev
    current_f_temperature:
      name: Temperature
    target_f_temperature:
      name: Set Temperature
    outlet_f_temperature:
      name: Heater Outlet
    power_l1:
      name: Pump Power
      id: power_l1
    power_heater:
      name: Controller Power
      id: power_heater
    power_l2:
      name: Heater Power
      id: power_l2
    pcb_f_temperature:
      name: Controller Temperature
    buttons:
      name: Buttons
  - platform: total_daily_energy
    name: 'Pump Energy'
    power_id: power_l1
    unit_of_measurement: 'kWh'
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
  - platform: total_daily_energy
    name: 'Controller Energy'
    power_id: power_heater
    unit_of_measurement: 'kWh'
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
  - platform: total_daily_energy
    name: 'Heater Energy'
    power_id: power_l2
    unit_of_measurement: 'kWh'
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001

# Enable time component to reset energy at midnight
time:
  - platform: homeassistant
    id: homeassistant_time
```

Note that there is a new `time:` section in addition to the `sensor:` section we had before. Once added and re-flashed, you can add `Pump Energy`, `Heater Energy` and `Controller Energy` to the Home Assistant Energy Dashboard. I think you need to wait a while before data will start to be shown. Let me know if this works.

Ylianst, 2024-08-18:
One more thing, I added `internal: true` to the wattage sensor to not have then be reported in Home Assistant and so, only get the KWh values like this:

```
sensor:
  - platform: iq2020-dev
...
    power_l1:
      internal: true
      name: Pump Power
      id: power_l1
    power_heater:
      internal: true
      name: Controller Power
      id: power_heater
    power_l2:
      internal: true
      name: Heater Power
      id: power_l2
...
```

This is a matter of preference, personally I don't want Home Assistant tracking and storing data I will not be using, so this makes things a bit more efficient. I get just what I need:

[image]

Ylianst, 2024-08-18:
Thanks @afp316. I added your data to the [tested hot tubs page](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/tested.md).

HalLandry, 2024-08-18:
I think this is a firmware thing, so I didn't start an "issue" for it.

Using ACE emulation, then on/off button state does not change if the output level is set to zero.

I see retires for switch 8 in the sysout if the level is zero.

I use the on/off to control the outside lights and the output level to control the porch ceiling fan speed.

Hal

Ylianst, 2024-08-18:
@HalLandry You are exactly correct, the ACE emulation on/off does not work when the salt power level is at zero and I can't do anything about that, it's a IQ2020 firmware thing. However, the "Test" button does work for all values so, I personally select my scene using the power level and trigger it under the "Test" button.

I also do this with a single Home Assistant automation and can set this up in the user interface, no YAML required. I do an automation that if "Hot Tub Buttons" changes to 6 then in the action, I add a "If" block, if "Hot Tub Salt Power" state is "0.0" do a set of things, if state is "1.0" do something different... Note that the salt power state is a float, so "0.0", "1.0", "2.0"...

Ylianst, 2024-08-19:
Follow up on the hot tub energy. Today, my breaker box is indicating that the hot tub used `5.22 KWh`, the hot tub sensors are as follows:

Heaters: 2.92 KWh
Controller: 1.73 KWh
Pumps: 0.42 KWh

For a total of `5.07 KWh`. I will keep watching this. So far, it seems to be under reporting a bit.

dhellerstedt, 2024-08-20:
- Hot Tub Model:
-Caldera Vanto 2023

IQ2020 Enclosure Internals
Watkins 1303401-1 Rev K
[image]

Yaml Changes Made in ESPHome same as @afp316 

> fan:
>
>   - platform: iq2020
>     name: Jets 1
>     id: jets1
>     icon: "mdi:turbine"
>     datapoint: 0
>     speeds: 2 #to 2 from 1
>   - platform: iq2020
>     name: Jets 2
>     id: jets2
>     icon: [...]

gdbassett, 2024-08-21:
I'm hoping someone here can help with figuring out the chlorine sensor on a Caldera w/ Salt System and Watkins connected spa. 

Background:
The connected spa system has a pretty straight forward wireless transmitter and network attached receiver.  The network attached receiver has a simple web interface and uses a python script and a custom binary library (dia) to read the channels from the spaw and upload them to watkins connected spaw cloud service which then interfaces with the app.  The python script can be edited and replaced, allowing a way to shim the whole thing w/ stock hardware.

I added code to the python script to log the channel status as they're being read to upload to the cloud service.  You can see the results in the chart below.  "spa_usage" is the salt system setting.  "mz_main_light_intensity" is a good indicator of when I used the tub (minus the las blurb where I cycled the light to ensure I knew when I added 4oz chlorine bleach).

[image]

What I can't find is any channel recording the measured chlorine levels.  Given the settings and usage, from mid august 18 through august 21, I'd expect it to be getting progressively lower (form almost top value to lowest), and then shoot up to top (with the directly added chlorine).  I don't see a single channel like that.

The channel I'd expect (g3_chlor_test_data) stays at  255.  "peripheral_status_chlorine" and "peripheral_status_ace_gen_3" are both 0 (likely meaning missing since "peripheral_status_networking" is 1).  However "g3_ph_test_data" does have values (need to add some ph down).

Does anyone have any ideas? The hot tub console can clearly read the chlorine level and the app _says_ it can...

Here's the python script if anyone's interested (with a few of my edits): https://gist.github.com/gdbassett/5ee80a57c4eb80d9c5a481cbdb819a2c

Ylianst, 2024-08-21:
Hi @gdbassett. So... basic question maybe, but how to you know there is a chlorine level sensor? Do you have the ACE or FreshWater system? Can you send a picture of the hot tub console reading this value? What does it looks like on the screen? Thanks.

gdbassett, 2024-08-21:
@Ylianst  ^

Ylianst, 2024-08-21:
Ok, this is the FreshWater system so yes, none of the Python values you are looking at include data from FreshWater. FreshWater has a binary block of data that looks like this:

```
1C:01:29:0F:80:1E:01:00:00:03:35:00:00:00:00:B3:10:00:68:41:83
```

The address of the module is `0x29`. I have the Python your talking about but, if I recall correctly, that code does not decode the Freshwater data block and instead just sends it as-is to the Watkins server. I attempted to decode this data, it's in the [RS485 Protocol](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/protocol.md) page. I did not get super far with it, but it would be wonderful if you could complete the work. I have a Freshwater system myself, but don't use it and disconnected it. If you can decode that data block correctly, that would be wonderful.

A while back, I made this, this it's not 100% accurate.

[image]

Ylianst, 2024-08-22:
Yes, this is it.  The start of the packet...

```
1C 29 01 0F 40 1E...
```

The `0x29` is the source access (Freshwater Module) and `0x01` is the destination address (IQ2020). The Spa Connection Kit only considers data sent to it, with destination address `0x1F` so it will ignore everything else. IQ2020 will poll the Frashwater module periodically and you will need to change the Python code to decode that data. You are on the right track.

gdbassett, 2024-08-24:
Two questions from that.

If the spa connect kit ignores the message as it's not the destination, can I really capture the sensor reading from the python script (as the dia library is ignoring the message)?  (I can apparently kick off multiple python scripts so kicking off one that scrapes the python log for message may then be the only option.)

If the spa connect kit is ignoring the freshwater module messages, WTF is it showing in the app?

Ylianst, 2024-08-24:
@gdbassett Ha yes... Looking at the protocol page, the base station must be polling the state like this:

Read FreshWater/ACE Salt Module Data (Polled)
```
<-- 01 1F 40 1E03
<-- 1F 01 80 1E030003FFFF00FF006800FF07FF0000B3100040
```

When it does that, the response will code make to it (0x1F) and so, not be ignored. I new see that there was code decoding of the data, this is in my protocol page and have gotten this from Python:

FreshWater/ACE Salt Module Data Decoding

```
1E03
00 - spa_usage (0 to 10)
03 - spa_size
FF - g3_sensor_data
FF - g3_sensor_status
00 - dosing_state
FF - ?
00 - g3_level2_errors
6800FF - ?
07 - salline_test (salinity test)
FF - g3_chlor_test_data (chlorine test)
00 - g3_ph_test_data (Ph)
00 - g3_clrmtr_test_data (colorimeter test)
B3100040 - ?
```

Look for `g3` in the python code, you should see where values.

The IQ2020 will also poll for these values and when it does, the base station will ignore this data, but yes, you should be able to see these values.

gdbassett, 2024-08-24:
I got a script logging the freshwater system (https://gist.github.com/gdbassett/4e45cabf25e9a4e6c7950892fc0afbe4).  In this log I go from red (low) to red (high) by adding 3oz 10% bleach to a 360gal tub (probably enough to go 0 to 7 PPM).  I also decreased the salt system from 8 to 1

[freshwater (5).log](https://github.com/user-attachments/files/16737376/freshwater.5.log)

Charting out the logs, if we look at the IQ2020 -> freshwater, it looks like the last two bytes (or at least the last byte or some set of the bits) is the chlorine level with high being low chlorine and lower being more.  The 8th two bytes are the salt system setting:
[image]

We can also see it from the freshwater system to the IQ2020:
[image]

gdbassett, 2024-08-24:
If freshwater is 01 and IQ2020 is 29, I've got my charts above backwards.

Ylianst, 2024-08-24:
Nice!! So, IQ2020 is 0x01 and Freshwater is 0x29. The full list is:

```
0x01 - IQ2020 controller
0x21 - Coolzone module
0x24 - ACE Salt System module
0x29 - Freshwater Salt System module
0x33 - Audio/Music module
0x1D - Audio/Music module
0x1F - Spa Connection Kit
```

The `0x1E03` is the command, this has no chance of containing any data.

gdbassett, 2024-08-28:
[image]

Still looking at it but have more data.  Red are tub uses.  Green line is adding chlorine.  It appears COL 20 and 21 should be added.  (not sure why they're separated.)  

Going to test if the drop-outs in COL12, 20, 21, and peripheral current are cycle times on the chlorine setting.  moved it from 8 to 10.  Will see if those period shorten or disappear.

but still not seeing anything which would match the chlorine measurement. (decreases at the red lines, big increase at the green line, and progressively back down to where it was before the 2nd red line.)

Ylianst, 2024-08-28:
@gdbassett Interesting. Looking at your chart, are you decoding command `0x1E03`? It seems like this is the decoding of the `0x0256` command with voltage, etc. which will not have chlorine measurement. This said, maybe I am missing something?

gdbassett, 2024-08-28:
There are no 0x1E03 commands, only 0x1E01.  see attached log (scraped from dia debug logs).

[freshwater (17).log](https://github.com/user-attachments/files/16775495/freshwater.17.log)

Ylianst, 2024-08-28:
I have not seen the 0x1E01 command, maybe it's a short version of the 0x1E03. Decoding it:

"1C 01 29 0F 80 1E 01 01 03 44 42 00 03 00 00 51 05 00 69 41 9A"

You get this:
```
1E01
01 - spa_usage (0 to 10)
03 - spa_size
44 - g3_sensor_data
42 - g3_sensor_status
00 - dosing_state
03 - ?
00 - g3_level2_errors
005105 - ?
00 - salline_test (salinity test)
69 - g3_chlor_test_data (chlorine test)
41 - g3_ph_test_data (Ph)
9A - g3_clrmtr_test_data (colorimeter test)
```

Keep monitoring 0x1E01, this is where you should see your data.

gdbassett, 2024-08-29:
This resource ( https://acesaltcellreplacement.com/wp-content/uploads/2018/12/Ace-Trouble-Shooting.pdf ) says the ace system doesn't have a chlorine sensor.  Salt is a point estimate at run time (twice daily based on the peripheral current data).   (Bold sentence at end of paragraph on page 1.)  This leads me to believe g3_chlor_test_data will never return data.

right now for "1C 01 29 0F 80 1E 01 01 03 44 42 00 03 00 00 51 05 00 69 41 9A" I'm thinking something like...

```
1E01
01 - dosing_state (high confidence)
03 - date (med confidence - goes up 1 exactly every day)
44 - ?? salt_test_data ?? (low confidence - varies between 40, 44, and 1 F4. Mainly sits at 44)
42 - ?? (Goes up roughly once per day, range 42 to 46 (hex) )
00 - ?? ph_test_data ?? (low confidence.  went up after use.  varies in a limited range so far, 00-09)
03 - ?? (single events of "0B", otherwise "03".  Maybe indicates the beginning of a cycle?)
00 - ?? (always 00)
00 - ?? (always 00)
51 - ?? Chlorine test or Chlorine generator on ?? (goes up at constant rate when chlorine system on)
05 - ?? (always 05)
00 - ?? (always 00)
69 - ?? (always 69)
41 - ?? high bytes of chlorine test ?? (when combined with next two-byte, results in a linear progression interrupted by duty cycle of chlorine dosing)
9A - ?? low bytes of chlorine test ?? (when combined with previous two-byte, results in a linear progression interrupted by duty cycle of chlorine dosing)
```
(note, the only thing on the freshwateriq system varied here was the dosing)

I got tired of downloading the settings through the dia webapp and so built a small webserver to serve them to me: https://gist.github.com/gdbassett/a50c40bf8429d13de5d55ce5e5855452.  It works with the other scripts that collect the logs.  I'll also use it as the basis of a bigger webapp to control the salt system eventually.

Ylianst, 2024-08-29:
@gdbassett Nice work! I am not an expert at this, but for my hot tub both ACE or Freshwater can be installed and the remote control has two different user interfaces depending on the module you have installed. I was not able to figure out much about Freshwater, but ACE was a lot easier. It has a mode at the top, a salt level and a power level with a on/off and test button.

[image]

Because my code can emulate ACE, I figured out all of the basic useful values for it and indeed, it does not offer a lot of data. This screen is my favorite way right now to control lights scenes of my house using my hot tub remote. I select the scene number and hit test to activate it. Anyway, yes, I am not surprised ACE is very limited... but your device has a water care menu that seems to be way more capable and ACE.

Ylianst, 2024-08-31:
Oh, humm... interesting. You posted this image before:

[image]

You are saying that the values shown on that user interface are correct, like, going up when you add chlorine, but you see nothing change in the commands? I would have expected the Freshwater module to provide all this data. If it's not the case, it's very much a misery. Are there any other commands being regularly sent to the controller from a different address?

The other option is that it's not on the RS485 bus at all, but I would be very surprised. Everything seems to be on that bus.

gdbassett, 2024-08-31:
Exactly.  The screen seems to reflect what's added, but nothing in the data (or the app for that matter).

My next step was to try and scrape every message, but the difference in length is going to make it a bit harder, and the volume is going to mean being a bit more judicious with how I log it on the device.

My concern is that there's possibly a bug and the chlorine monitoring beyond the user interface is broke.  I may email them.

Ylianst, 2024-08-31:
If you index the commands by the destination/source identifier and command number, they should all have the same length. In this example:

```
1C 29 01 0F 40 1E 01 0A 02 FF FF 
```

You keep the 0x2901 and 0x1E01, create a int of 0x29021201 and store the packet.

By the way, I take it you never installed the ESP32 integration, I imagine this is because you don't use Home Assistant? In any case, if you have a way to get a large dump of all traffic for a long time, that would be interesting to look at and see if there is any other commands.

gdbassett, 2024-09-01:
Honestly? I asked for the watkins device w/ the hot tub. (Actually got it free because the sales guy thought the tub came with it and so didn't charge me extra.)

Also, I'm trying to avoid adminning a server at home.  I have a phillips hue bridge for lights server and apple TV for matter server.  Just too old to want to admin my own server these days.  (But I love doing data stuff.  Though I did wildly underestimate how difficult this would be.  I expected both chlorine status and salt system setting to be in the data sent to the server.  It's wild to me that the chlorine sensor clearly works given the user interface, but doesn't get read by the watkins device, sent to the cloud service, or displayed in the app.)

Ylianst, 2024-09-02:
@gdbassett ha yes, I completely understand. This is one of the big limitation of the ESP-IQ2020, you need to have Home Assistant which is a bridge too far for a lot of people and I completely understand it. At some point, I may look into have a standalone ESP32 firmware with a local web page, that way you can do what you are doing, but for around 25$.

Going a big gzip of a lot of packets is a good idea, you can do a lot of offline analysis which is great. If there is no success and it's true there is no chlorine data on the RS485 bus, It would then be interesting to see how that data is routed to the IQ2020. If you have pictures of your IQ2020, it would be interesting to see if you have an extra wire to an additional module that others do have.

Ylianst, 2024-09-02:
Searching online a bit, I found "Freshwater IQ" with this UI.

[image]

This is certainly what you have. I have the Freshwater module, but not IQ. If anyone else has information on this, how it works and how it's installed? Can it be installed after market?

mediamanrit, 2024-09-02:
I have the Freshwater IQ add on, and my ESP32 is in the mail to try to mess with it.

I have a HotSpring Envoy.  It’s a sensor that goes in the old mineral stick hole.  It appears to go to a connector on the expansion board, but I haven’t looked close enough to see if it’s rs485 or i2c yet.

Fair warning, I’ve got 2 little kids, so once my ESP32 gets here it’s going to be a bit before I get it set up and going.

Ylianst, 2024-09-02:
Thanks @mediamanrit. Absolutely no worries on timing. If you can even send a high res picture of the connectors you have like this.

[image]

The expansion connectors have 8 pins, but generally only 4 wires are connected. It would interesting to know what wires are used. Overall, having Home Assistant be able to connect and graph the IQ data would be great for people that have it.

gdbassett, 2024-09-07:
I spoke with watkins support and 0x37 is the freshwaterIQ.  My log scraping script died so having to restart it, but going to focus in on the 37->01 messages.  For the 2 hours or so I have, here's the columns (2-byte parts of the message) that have more than 1 value.

[image]

Ylianst, 2024-09-07:
Wow, you can talk to support and get that information? Well, did not know `0x37` was a module, I just added it to my list.

```
0x01 - IQ2020 controller
0x21 - Coolzone module
0x24 - ACE Salt System module
0x29 - Freshwater Salt System module
0x33 - Audio/Music module
0x37 - Freshwater IQ
0x1D - Audio/Music module
0x1F - Spa Connection Kit
```

gdbassett, 2024-09-08:
I'm starting to think the 23 d1 and 23 dc commands are the chlorine.   See below  There seem to be several columns carrying relatively smooth curves.  (I sat in the hot tub today, Sept 7th for a few hours and it went from near max chlorine to about mid.)  Will have to see if it continues to track.  

23 d1
[image]
23 dc
[image]

gdbassett, 2024-09-08:
23 D5 on the other hand has far more signal, but mainly seems to be more binary values and a few counters around Col68 - 73

23 D5
[image]

gdbassett, 2024-09-08:
The 23 a1 also kinda tracks teh d1 and dc messages.

23 a1
[image]

gdbassett, 2024-09-09:
Looking good for 23 D1.  I was in the hot tub sept 7th for a while.  The added 4oz chlorine sept 8th.  System is set to 10 so should be going up.

COL 27 + 28 are likely related.  Col 36 looks like it's scaled to 0-128 or such.  Col 40 looks like it might  be the bars on the screen... (80 = high red?,  70 = low red?)
[image]
Will continue monitoring and using it.  Eventually will add the salt system setting and lights/jets (which help indicate what I was doing...) to see if it all aligns.

gdbassett, 2024-09-14:
I'm pretty confident COL36 is chlorine in 10ths of PPM.  Col 40 looks to me to be Ph in 10ths as well.  Going to try and automate chlorine control with something like this today:

```
### less than 45, go full
if chlorine["chlorine"] < 30:
    json_rec[key] = 10
    print("Chlorine low. Going to 10")
### more than 60, go minimal
elif chlorine["chlorine"] > 50:
    json_rec[key] = 6
    print("Chlorine high. Going to 6.")
### if we're less than 50 and going down, go up to maintain
### remain in maintain until we hit full or minimal
elif json_rec[key] <= 6 and chlorine["chlorine"] < 40:
    json_rec[key] = 8
    print("Chlorine coming down, go to 8 to maintain.")
### if we're over 50 and going up, go to maintain
### remain in maintain until we hit full or minimal
if json_rec[key] == 10 and chlorine["chlorine"] > 40:
    json_rec[key] = 8
    print("Chlorine coming up, go to 8 to maintain.")
```

[image]

gdbassett, 2024-09-14:
Manually sending values to the salline_test value that gets surfaced in the mobile app.  It appears it only goes low_yellow (at 30) to high yellow (89).

Everything else takes the system offline.  (It looks like 20's are also low yellow, but in offline - low status state)

Here's what I got fiddling with it:

salline_test
0,grey,"offline"
1,grey,"offline"
60,low_green
50,low_yellow
40,low_yellow?
30,low_yellow?
20,grey,"offline"
25,Low_yellow,"offline - low status"
29,Low_yellow,"offline - low status"
70,mid_green
80,high_green
90,grey,"offline"
85,high_green
87,high_green
89,grey,off
88,high_yellow

gdbassett, 2024-09-15:
ok, I think I have it automated.  set_tanas_menu_entry in the thinkworx channels is the salt system setting (value + 256, so a salt setting of 10 is 266).

gdbassett, 2024-09-17:
[image]

Ylianst, 2024-09-18:
This is amazing! I could probably figure it out, but if there is anyway you can take an example data packet and decode it, that would be great. I don't have Freshwater IQ, but if there is interest in having it added to the ESP-IQ2020 integration, I am willing to put in the work. I would need someone with this module and Home Assistant to confirm that it works.

robertsze, 2024-09-20:
Works like a Charme!, Hotspring Grandee, just three minor issues/requests

1. Motor Power consumption appears to measure 1 motor only? ( Shelly is reporting around 2kwh for each motor if turned on fully), but the ESP32 reports approx 1.95kwh only, just a minor issue

2. Are there any plans to support fresh water system? (salt system), e.g. changing boost mode, level, ...

3. After a few days, with the fresh water system, the pool stops producing chlor and waits for a user confirmation, any plans on supporting this? (this would be help a lot if not at home and you get a notification when the whirlpool stops)

Anyway, great work and really great instructions how to setup, very appreciated!!

Ylianst, 2024-09-20:
@robertsze Thanks for the feedback. Can you post the year of your hot tub and the version string? I will add your device to the [tested page](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/tested.md).

1. For the hot tub power sensors, I have no real idea what the numbers measure and just took guesses. Since you find what the measurement is exactly for, let me know and I will change the name / unit. This value may also change depending on your tub model. So, your likely right, not sure what the value truly is.
2. Yes, take a look at the [extra's page here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/extras.md#ace--freshwater-salt-systems). You can add extra sensors for Freshwater (Note that Freshwater IQ is something different). Let me know if that works.
3. Yes, we have the "Salt Level Confirmed" sensor that tells you confirmation is needed. I do not have a command to remotely confirm, but it's not impossible it exists and has not been captured yet.

Again, thanks for the report. The [extras page](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/extras.md) has more controls for lights, etc. Feel free to check it out.

robertsze, 2024-09-20:
@Ylianst 

> Thanks for the feedback. Can you post the year of your hot tub and the version string? I will add your device to the [tested page](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/tested.md).
Hotspring Grandee Model Year 2021

> 1. For the hot tub power sensors, I have no real idea what the numbers measure and just took guesses. If you find what the measurement is exactly for, let me know and I will change the name / unit. This value may also change depending on your tub model. So, you're likely right, not sure what the value truly is
I can debug this a bit and try the find the correct values / meaning

> 2. Yes, take a look at the [extra's page here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/extras.md#ace--freshwater-salt-systems). You can add extra sensors for Freshwater (Note that Freshwater IQ is something different). Let me know if that works.
Awesome, will give it a try

> 3. Yes, we have the "Salt Level Confirmed" sensor that tells you confirmation is needed. I do not have a command to remotely confirm, but it's not impossible it exists and has not been captured yet.
Will give it a try too!

Thanks a lot, I will dig into the source a bit if something is missing and try to provide patches if I find something

nigeljg, 2024-09-22:
This is great! This has saved me £500.00 as I was about to by the wifi controller from Hot Springs in the UK. 
My hot tub is HotSpring HotSpot SX 2017 version 110T 7b54A002 v2.6

There is one little issue, is that the light switch is a bit unstable and turns on off as and when it likes. Sometimes its comes on its own. The light in the hot tub is just one blue colour, but you can control the intensity, by pressing the light button, once for dimmed, and then again, for full power and then press again, its off. I have tried to tweak the yaml code, but I hasnt worked.  I remove jets 2 and 3, as there is only one pump on the tub. 

On Debug I do get a warning now again the Component IQ2020 took a long time for operation (63ms).

Also The pump power doesnt show any power when the pump is on.

I am also using this, without HA, and just logging into the web interface, using portfowarding on the router. This hot tub is our holiday home, and I dont want to set up another smart home server. I also use Openhab, in our main home, and it gets a bit tricky to multi cast over VPN so that openhab can control it. 

Many thanks for figuring this out!! 

[image]

Ylianst, 2024-09-22:
Hi @nigeljg and thank you for sending the model, version and image. That is super useful.

For the "pump power", that is only a guess on my part as to what that sensor is and many hot tubs will not have this sensors at all. If it's always zero, just remove it from the YAML.

For the light, your hot tub does seems to have a different light configuration for sure. If you want to debug it, follow the [debugging guide here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/debugging.md) and you will see all the RS485 traffic. If I had to guess, you hot tub has only the underwater light with brightness 0 to 2. Again, complete guess but if this is correct, you can adjust the YAML to this. I will guess any brightness over 2 will turn the lights off.

Not using Home Assistant is perfectly ok. If you have suggestions for stand alone usage, let me know. I would be interested in providing stand alone usage since not everyone uses HA.

nigeljg, 2024-09-24:
Hi @Ylianst, I have just removed the light switch YAML code, and just left in the YAML code to change the intensity, to 3, ie giving 3 positions. Trouble is I am not there at our holiday home at the moment, and will not be down this weekend, so I dont know if it is really working and controlling the light, but is not throwing up errors on the debug. I dont notice on the Debug, that setSwitchState, switchid=0, status = 0, and I cant seem to figure out what that switch ID is in the YAML code.

I have found a ESPHome Link, on here, github, where someone has developed a standalone app for ESPHome, and all you need to do is point the app at it. It appears to work OK. I attach a screen shot, with my IP address greyed out. I think you can encrypt it, but Im not using the encrypted bit, and Im NOT that really worried, if someone hacks into the control of the hot tub! 

I will take a picture next time I am down there of the ESP32 connected to the pins on the board and post it. 

Many thanks once again on figuring this out!!! 

[image]

Ylianst, 2024-09-24:
Thanks @nigeljg. I did a quick search and found [ESPHome Link](https://github.com/DanielBaulig/esphome-web-app/), indeed this is very interesting as it allows access to a ESPHome device without the need for HomeAssistant. Good find, I will take a look at it in more detail when I get a chance.

Ylianst, 2024-09-24:
Hi @gdbassett. I just took a look at the logs your provided to decode Freshwater IQ data. So far, I have this:

```
<-- 37 01 40 23D500
<-- 01 37 80 23D5000000003234313130304E30A76EBD3915FF7E9C30303744000000040000000400000004000031500000271000002710000007E2000007E2
                                                                                                                 AAAAAAAABBBBBBBB

AAAAAAAA - Cartridge use in hours (Small-Endian)
BBBBBBBB - Cartridge use in hours (Small-Endian)
```

The one week log you provided has many `0x37` commands, both polling and response, and I see the two last 4 byte values increase every hour. However, I don't see any other changes. Can you point to exactly where are the Ph and Chlorine values and how you decode them? I will update the documentation and add support for it in the integration. Thanks.

[105 more replies]

closed 2025-04-24 by Ylianst