Ylianst/ESP-IQ2020 #14: Tiger River Spa Caspian

mark-r-cullen, 2024-07-30:
Just tried this with the recommended Atom and Tail485. It powers on and I get "Poll Version" logs after flashing, but that's about it. Sensors are showing "unknown" in Home Assistant.

I measured the voltage on pin 2 as 9v though. Perhaps there's something a bit amiss with my board if this should be 12v?

Ylianst, 2024-07-31:
Hi. What year is your hot tub? I see Tiger River Spa Caspian tubs dating back from 2003 before the IQ2020 existed. Are you sure you don't have a IQ2000? I found this [video on replacing the IQ2000 with the IQ2020](https://www.youtube.com/watch?v=7U2TA-5Is2c). If this is the case, I am not suggesting you do this, just use the video to see if this is your case.

The IQ2000 has a similar connector but it's not pin compatible. If you can send a picture of your board.

mark-r-cullen, 2024-07-31:
I'm not 100% sure what year as I got the tub second hand, but I think it's from about 2008. I'm fairly sure it's a iq2020 though. The board looks exactly like in your first image showing where to connect the pins to the esp.

I do have an oscilloscope so could check the pins to see if anything is happening. I'm not familiar with rs485. Just pin a and b to ground?

Ylianst, 2024-08-01:
It looks like to have a board used from 2001 to 2009.5. If you look just to the right of the expansion connector, it will say I2C. It also says this on my board but I do wonder if you board is only I2C and RS485 was added later.

In doing my past research, I found on ebay some devices that seem like the connector would fit, but the wires don't make any sense, like the image below. It seems to show the older wiring. In once of the picture on my main page, I used a random IQ2020 board image to show the connections, but that may have been a mistake since it's showing an older board and not the board I have (which is on the second image). I will fix this, but still, it would be nice to know when RS485 was added.

You could go thru the same steps I have done and intercept traffic to a expansion module and figure out the pins. If you can find the I2C pins, you may be able to start controlling the hot tub with I2C get/set commands. It's possible the modern IQ2020 also have I2C, there are 2 free pins I suspect may be I2C, but never looked further into this.

[image]

Ylianst, 2024-08-01:
I hope it's ok, I updated the [Tested Hot Tubs](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/tested.md) page with your image to warn people. I don't think we had anyone try such an older hot tub before.

The ESP32 does support I2C, so, it's not an impossible task that something could be made to work with it, however, I don't have such a hot tub controller.

mark-r-cullen, 2024-08-02:
I think it's either my board is faulty in some strange way, or RS485 was added later on. I guess we'll have to wait for someone else with an older board to try.

I should add that on the i2c connector I did have a 4 port expansion board which I removed, as it only passed through pin 2-5 (the i2c pins and power) as per the wiring on that iCAST device. I believe the RGB light for this era is also i2c.

When I get some time, I'll see if I can get anything out of i2c instead. There seemed to be activity on pin 3 and 5 at least. I suspect pin 3 is the clock and 5 is data. I did start looking yesterday, but I've never touched esphome before and I think I need to do a little reading up :)

Ylianst, 2024-08-03:
@mark-r-cullen Ha yes, if I had to guess, you have the older wiring style. I have the 8 port expansion board myself and in my case, the board passes thru all pins, however pin 4 and 8 on the diagram below are short (connected) on the expansion board. This gave me a really good hint that 4 and 8 where both ground.

[image]

I labeled pins 3 and 5 as I2C, but this is speculation on my part as I have not tested these much. If I recall, I think I did see them as being "high" which would be consistent with I2C pins, but did not investigate further since the official internet connection key does not connect to these pins.

Ylianst, 2024-10-30:
Closing this. Please re-open if there is any new on this topic.

closed 2024-10-30 by Ylianst

mark-r-cullen, 2025-04-04:
So, an update from me. I didn't ever get around in to looking in to i2c, though I'm fairly sure something was happening on the pins. 

However, I bought this tub second hand and it's always had an intermittent issue when switching between jets, more often when it was trying to run both jets on high, where sometimes it would reset the power (everything off) then the display would flicker briefly, the jets try to spin, and repeat. It seemed to be getting worse, and I found one post with a similar sounding issue saying that a new control board fixed the problem for them.

I decided to retrofit the newer 50hz Eagle box (part 76837) and hot springs control panel (part 76845 - this is visually identical to the discontinued tiger river spa control panel and slotted straight in). I believe the board I had before was an older Pug board for reference, at least judging from pictures. It was a very straightforward swap. I also upgraded the heater from 1.5kw to 4kw about a year ago which was equally as straight forward and doesn't seem to cause any issues (I intentionally had over spec wiring put in as I knew I would end up doing this!)

I plugged in the esp on this board and it just works! Well, aside from I seem to have no low speed jets any more, but I believe this is likely a software problem (low speed is just high speed on this integration, and the button on the control panel only seems to cycle between pump 1 high, pump 2 high and pump 1&2 high now, where before I had five settings). I have gone back to the supplier to query this.

Thanks for the work on this. It's excellent!

mark-r-cullen, 2025-04-05:
The supplier of the board got back to me about the low speed pumps not working and gave me a soft jumper sheet. Pressing the white button on the board to go to service mode, then mode button up on the control panel until you reach J12 and set that to + by pressing the on / up button and then power cycling has done the trick!