Ylianst/ESP-IQ2020 #15: Unable to import component Iq2020.sensor

Schjonning, 2024-07-30:
i have issues installing the using ESPHOME, i get the following error:

INFO Reading configuration /config/esphome/esphome-web-15814c.yaml...
ERROR Unable to import component iq2020.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 165, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/24b0febb/components/iq2020/sensor.py", line 4, in <module>
    from esphome.const import (
ImportError: cannot import name 'ICON_HEATING_COIL' from 'esphome.const' (/esphome/esphome/const.py)
Failed config

text: [source /config/esphome/esphome-web-15814c.yaml:48]
  
  Component not found: text.
sensor.iq2020: [source /config/esphome/esphome-web-15814c.yaml:54]
  
  Platform not found: 'sensor.iq2020'.
  platform: iq2020
  current_f_temperature: 
    name: Current Temperature
  target_f_temperature: 
    name: Target Temperature
  outlet_f_temperature: 
    name: Heater Outlet
  power_l1: 
    name: Pumps Power
  power_heater: 
    name: Power Heater
  power_l2: 
    name: Heater Power
  pcb_f_temperature: 
    name: Controller Temperature

Ylianst, 2024-07-31:
I took a quick look at this and is it possible you are running an older version of ESP-Home? 'ICON_HEATING_COIL' was added to the ESP-Home source code on Aug 31, 2023 and so, was probably released sometime later. For my use, the [link to the code is here](https://github.com/esphome/esphome/blob/3920029aff9dd85fd5f0775945b8426c03da471c/esphome/const.py#L969).

Schjonning, 2024-07-31:
hi, thanks noted and corrected. now with a different issue that you might be able to assist with 

[logs_badekar_compile (1).txt](https://github.com/user-attachments/files/16447730/logs_badekar_compile.1.txt)

```c++
esphome:
  name: badekar

esp32:
  board: m5stack-atom
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0
  level: ERROR

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source: github://ylianst/esp-iq2020

# Make sure tx/rx pins are correct for your device.
# GPIO26/32 is ok for M5Stack-ATOM + Tail485, look in GitHub devices link for your device.
uart:
  id: SpaConnection
  tx_pin: GPIO26
  rx_pin: GPIO32
  baud_rate: 38400

iq2020:
   uart_id: SpaConnection
   polling_rate: 65
   port: 1234

select:
text:
number:

# If using celsius units on the hot tub remote, replace _f_ with _c_ in the three entries below.
# Feel free to remove any sensor that are not relevent for your hot tub.
sensor:
  - platform: iq2020
    current_c_temperature:
      name: Current Temperature
    target_c_temperature:
      name: Target Temperature
    outlet_c_temperature:
      name: Heater Outlet
    power_l1:
      name: Pumps Power
    power_heater:
      name: Power Heater
    power_l2:
      name: Heater Power
    pcb_c_temperature:
      name: Controller Temperature

switch:
  - platform: iq2020
    name: Lights
    id: lights_switch
    icon: "mdi:lightbulb"
    datapoint: 0
  - platform: iq2020
    name: Spa Lock
    id: spa_lock_switch
    icon: "mdi:lock"
    datapoint: 1
  - platform: iq2020
    name: Temperature Lock
    id: temp_lock_switch
    icon: "mdi:lock"
    datapoint: 2
  - platform: iq2020
    name: Clean Cycle
    id: clean_cycle_switch
    icon: "mdi:vacuum"
    datapoint: 3
  - platform: iq2020
    name: Summer Timer
    id: summer_timer_switch
    icon: "mdi:sun-clock"
    datapoint: 4

fan:
  - platform: iq2020
    name: Jets 1
    id: jets1
    icon: "mdi:turbine"
    datapoint: 0
    speeds: 1
  - platform: iq2020
    name: Jets 2
    id: jets2
    icon: "mdi:turbine"
    datapoint: 1
    speeds: 2

# Set "celsius" to "true" if using celsius units.
climate:
  - platform: iq2020
    name: Temperature
    celsius: true

text_sensor:
  - platform: iq2020
    versionstr:
      name: Version
    
```

Ylianst, 2024-07-31:
It looks like your ESP-Home is not installed correctly, I am not a big expert in ESP-home, but try to install a basic firmware like this below and see it works before adding the IQ2020 integration.

```
esphome:
  name: badekar

esp32:
  board: m5stack-atom
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0
  level: ERROR

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

Ylianst, 2024-08-18:
No more info. Closing. Please re-open if needed.

closed 2024-08-18 by Ylianst