Ylianst/ESP-IQ2020 #17: Audio Emulation Power Button

creemerica, 2024-08-18:
First of all want to say great work. I have this installed and working on a Grandee 2021 and it works great! 

I do have a question regarding the Audio Emulation Mode. I am curious if there is an event sent when the Power Button is tapped on this screen? I would like to be able to turn on/off my outdoor speakers based on the power state of the audio mode on the hot tub controller if possible.

Ylianst, 2024-08-19:
Hi and thanks. Interesting you ask this. If you hit the play button in the audio screen and later hit the power button, the audio "Stop" button will get pressed. My first experience was to setup 4 lights and have `previous`, `stop`, `play` and `next` each toggle a different light. It turns out, pressing the power button does call `stop` and so, one of my lights would toggle. I am not sure now, maybe powering on again would press `play`.

In any case, there is no direct event for the power button (sadly, that would be nice, but I can't change the firmware). However, because of the `stop` on power off, you may have exactly the behavior you want. Let me know if that helps.

creemerica, 2024-08-19:
Thank you, I think I can work with that. For now I am just turning the speakers off after the pause event is passed and turning them on with the play event. While this means the speakers may turn off when I press pause and not just turn off, I think it’ll be fine and work as I want. Thanks!

closed 2024-08-19 by creemerica