Ylianst/ESP-IQ2020 #18: My fork for a 2016 Hot Springs Flair

john159753, 2024-08-21:
I don't know if this should technically be an "issue", but I wanted to put my information out there. First off - thanks again for all your work on this. I am sure tons of time went into it. I made some edits specifically for my hot tub. I really don't mind just having my fork exist in parallel with your base. I don't know how we can get my unique changes could be merged into your code, and i don't know if its worth it to make it a "one size fits all".

I have all my changes sitting in the iq2020-dev component, and I'm probably going to keep it there for now. Let me know if you have any questions on any of this. Close out this issue if you want, but i just wanted to share what I have done. Oh yeah -- I have a 2016 HotSprings Limelight Flair

- Fixed all instances of Tremble spelling error
- Added trigger poll pin logic, when this pin goes high it triggers a poll. I am using this with a pullup resistor to another ESP that (with optocouplers) watches LED light changes to  trigger polls. The code isn't perfect, detecting these changes was proving to be a pain (without being noisy and overflowing on poll requests) but it also gets the job done.
- Changed Audio Address in all instances from 0x33 to 0x1D, my IQ2020 was looking for an audio device on 1D
- Audio volume logic needed to be changed, the data I was getting from my box was simply 0-40 0x00 to 0x28 (rather than 0x0F to 0x18). So no need to do any maths/bitshifting to translate.
- Light Changes -
1. I had different colors so i updated the color list with my colors. 
2. I was getting a crash on the ESP when cycling through the color list, i think what was happening is my firmware loops from 0x08 to 0x00, whichthen got passed to publish_state_ex, where value-1 was getting applied to 0x00 and going to a negative index, i just added a quick line to make sure value == 0 get set to "cycle"
3. I also noticed my color cycle start command is not 0x08 like yours. 0x08 and 0x09 didn't seem to do anything for me. 0x07 started the cycle and 0x06 stopped it. This is where the only open bug exists for me at this point. I dont know what to do about it. When I send the command using your packet tool, the lights cycle normally. When the ESP tries to do, for some reason the esp starts looping, with poll requests and the tub's color loop stutters around to a root color while the loop continues running in the background. That explanation probably isn't that good, but its the best i can do. The loops are not from my trigger logic - even though that seems like an easy thing to point at and blame. Something that sticks out as odd is below. I am throwing in the towel on this one.
```
11:07:24	[D]	[iq2020:1088]	setSwitchState, switchid = 0, status = -127
11:07:24	[D]	[iq2020:1157]	Poll
```
- Adding the iPOD to the audio source list. Then also changing the logic in publish_state and publish_state_ex to handle ipod fitting in position 1 of the select_audio_source list

Ylianst, 2024-08-22:
Nice!! Fully encouraged. Would you be ok sending me over your version so I can try to merge? In some cases like the audio, I can make the address configurable, etc. I would rather try to make a single component if possible.

Ylianst, 2024-08-24:
Hi @john159753. I just merged your changes in my "dev" version and testing it now. It seems that you have an audio modules attached since I don't think you are using the audio emulation.

The is a specific command that I really would like a capture of, it's `0x1901`.

In my code, you may have noticed this line:

```
unsigned char cmd[] = { 0x19, 0x01, 0x00, 0x19, 0x00, 0x00, 0x00, 0x0B, 0x00, 0x04, 0x01, 0x00, 0x00 };
```

It's sent to the IQ2020 to indicate the audio module is attached, however, this is a complete guess on my part. I suspect this command indicates that input sources are available, volume range, etc. If I can have a capture of your audio traffic including this command, that would be super helpful. I would then be able to emulate your audio 0x1D device and make the code better.

I am testing my code so it supports both 0x33 and 0x1D audio devices, but not sure if it's perfect or not.

john159753, 2024-08-25:
I am using audio emulation, I don't have an audio unit. My clue to my audio address being 1D was seeing 190100 over and over again in my captures. Like this:
 <-- 1D 01 40 190100

I figured that's the IQ2020 polling for the presence of  the audio unit. In your examples on your protocol page, i was seeing 0x1901 all over the place so that's how i figured it is asking something for the audio unit. Once i changed all instances of 0x33 to 0x1D, your command you crafted up above, totally worked for my unit too, the IQ2020 saw an audio unit connected. Attached are two logs, were my tub is idle, the only difference is the audio emulation function being enabled (and all the extra sensors/text/select associated with it). They kind of look the same to me, i was hoping to capture the handshake between the ESP an and mainboard, but it doesn't seem like that's getting captured through over the UDP UART stream.

Rather than defining it in the config, or doing some sort of firmware/model version detection....It would be slick if the ESP code could detect the IQ2020 was asking for an audio unit on 1D, and then automatically change all the required 0x33 to 0x1D, including the volume differences and the select menu to include iPod as an audio source. Also just as an FYI I usually just keep it on iPOD because my bartop controller (not the wireless one) doesn't show song artist/title unless its on ipod mode. This must have been the age before Bluetooth was able to send song data along with the music 😆.

Let me know how i could be any help further!

[logfile-idle-audioemulation.txt](https://github.com/user-attachments/files/16741037/logfile-idle-audioemulation.txt)

[logfile-idle-noaudioemulation.txt](https://github.com/user-attachments/files/16741036/logfile-idle-noaudioemulation.txt)

Ylianst, 2024-08-25:
Ha! Got it. Thanks for being so helpful. In my merge, I have a new variable for the audio module address and do auto-detect it and, so, it should handle both cases now.

It's interesting that in your case the iPod option showed up. I don't see that once. I have to keep it on Bluetooth to have artist/song title and iPod is not shown. I added `iPod` to the selection, but I need to see what happens when I select it from Home Assistant and it's not supported. You can override the selection options in yaml, so, I may have to document this.

Don't worry about capturing anything, I was only asking if you had the actual audio module. Yes, any data sent from the ESP32 is not recorded in the capture. I could fix this, but never bothered since it's not been of any interest to me.

I also made it so that depending on 0x33 or 0x1D the volume is encoded/decoded differently. So, hopefully that will handle both cases.

I am still going to be on the lookout for anyone with the audio module, it seems to be very uncommon.

Thank you so much for your fixes (and spell checking!) that was wonderful.

Oh, I did want to ask the purpose of the trigger pin? What ESP32 setup do you have??

Ylianst, 2024-09-06:
I just put your changes and a few changes into production. If you spot any issues, let me know.

john159753, 2024-09-06:
Sorry I haven't gotten back to you yet. Just been busy.

The trigger pin is connected to another esp that runs some pretty rudimentary code. It reads the led signal through optocouplers and when a change is detected it triggers the pin high which tells the esp to re-poll the IQ2020

Ylianst, 2024-09-06:
No worries on timing. What LED is the other ESP looking at? The LED of the hot to logo?

john159753, 2024-09-07:
Nope. I spliced the optocouplers in with the main lighting circuit 😆. I wanted real time response when I changed lights in the tub for my backyard lights to follow the same color (and vice versa).

Ylianst, 2024-09-22:
@john159753 The latest ESP-IQ2020 integration should have all your changes, if you ever test it let me know if it works and any more fixes I need to make. Trigger pin code is included.

Ylianst, 2024-10-30:
Closing this since I added a lot of fixes to this integration thanks for your fixes.

closed 2024-10-30 by Ylianst