Ylianst/ESP-IQ2020 #19: Logo light sensor not reporting updated value

derhally, 2024-08-23:
I know this was working before because I saw the sensor at one point report the value of "1" when the logo light was yellow.

Yesterday I noticed the light was yellow but the value being reported was 2.  Here is what the dataviewer app showed when I queried for lights

With lights on
1F 01 80 17050003000003000000000000000007010104010000

With lights off
1F 01 80 17050000000000000000000000000007010107000000

HA says a value of 2 was reported multiple times between august 21st and 23rd.

```csv
entity_id,state,last_changed
sensor.hot_tub_logo_lights,2,2024-08-21T04:00:00.000Z
sensor.hot_tub_logo_lights,unavailable,2024-08-21T11:08:42.605Z
sensor.hot_tub_logo_lights,unknown,2024-08-21T11:08:51.943Z
sensor.hot_tub_logo_lights,2,2024-08-21T11:08:55.204Z
sensor.hot_tub_logo_lights,unavailable,2024-08-21T19:06:52.696Z
sensor.hot_tub_logo_lights,2,2024-08-21T19:06:54.407Z
sensor.hot_tub_logo_lights,unavailable,2024-08-22T09:04:58.790Z
sensor.hot_tub_logo_lights,2,2024-08-22T09:05:00.834Z
sensor.hot_tub_logo_lights,unavailable,2024-08-23T14:22:58.275Z
sensor.hot_tub_logo_lights,2,2024-08-23T14:23:08.099Z
```

I'm on the latest release of ESP-IQ2020

Ylianst, 2024-08-23:
Hi. The logo lights are [explained in the extras here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/extras.md). The values are:

```
0 = Unknown
1 = Power on
2 = Power on & ready on
3 = Ready flash
4 = Power flash
5 = Power and ready flash
6 = Power and ready alternate
7 = Power and ready salt error
```

The logo lights is the blue/green logo (circled in red):

[image]

This command is for the main colored lights:

```
1F 01 80 17050003000003000000000000000007010104010000
```

It does not encode the logo light. Hope this helps.

derhally, 2024-08-23:
Yeah, mine is yellow because it asking me to confirm the salt level

Ylianst, 2024-08-23:
The [protocol page here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/protocol.md) has the location and values on the logo lights. Post your latest 0x0256 status command to see what the values are.

derhally, 2024-08-24:
here it is

 --> 01 1F 40 0256
 <-- 1F 01 80 02560008000400000604000A0822F11206201C201C201C8403605400000000203936466A7A1E006BD90A00B6D8E0011B0000000000000005C20100000000000000000085781100030051C9DD016BB8060000000000203936462039364678000001000100000000000000000000000046000B000000003C001E00006269003706041B01D10701
 <-- 1F 01 80 17050000000000000000000000000007010107000000

Ylianst, 2024-08-24:
Thanks @derhally. Lets see here:

```
 <-- 1F 01 80 0256000800040000060400 0A...
```

The logo light value is `0x0A` for you on both cases, which is (in my book) both blue and green on. In your case it's the same before and after. The `0x1705...` lights are the same, this I would expect.

An interesting change is this one:

```
 <-- 1F 01 80 02560008 40...
```

I have not seen `0x40` at this position before.

```
0x01 = Temp Lock
0x02 = Spa Lock
0x04 = Jet1 Full
0x08 = Jet2 Full
0x10 = Clean Cycle
0x20 = Summer Timer
0x40 = Salt water confirmed ??   <------
```

I can add a new sensor for this flag if you like and you can monitor it to see what it does longer term?

derhally, 2024-08-25:
I checked the status again today and it is back to showing 0x40 again.

```
<-- 1F 01 80 02560008400400000604000A0822F11206201C201C201C840360540000000020393846E1901E0055F10A00C438E2011B0000000000000030C601000000000000000000CC91110003005F29DF01B3C706000000000020393846203938467800000100010000000000000000000000004A000A000000003C001E0000631A010C09051C01D10701
```

I will play around with the settings some more and try to notice a pattern.

Ylianst, 2024-08-25:
Yes, monitor that `0x40`. If you have a good idea what the significance is, I will add it as a binary sensor. Let me know.

derhally, 2024-09-05:
I can confirm that 0x40 indicates the salt level is confirmed and that 0x00 means it needs confirmation

When Salt Level is confirmed 

```text
 <-- 1F 01 80 02560008400400000604000A0822F11206201C201C201C84036054000000002039354699ED1E00B2230B00C3A8F0011C00000000000000A3CB01000000000000000000999F110003001B99ED0195ED06000000000020393546203935467900000100010000000000000000000000004D0004000000003C001E0000601901010A050B02D10701
```

When Salt Level needs confirmation

```text
 <-- 1F 01 80 02560008000400000604000A0822F11206201C201C201C84036054000000002039354699ED1E00B2230B0041A8F0011C00000000000000A3CB01000000000000000000999F110003009998ED0195ED06000000000020393546203935467800FF00FF000000000000000000000000004B0005000000003C001E00006096003307050B02D10701
```

Ylianst, 2024-09-06:
Nice. Ok, I will add it as a extra optional sensor when I get a chance. It's tool bad it's not the opposite values, but this will do for sure.

derhally, 2024-09-06:
I've added the sensor and it is reporting.

I'm hoping there is one that may let you confirm the salt level and one that triggers the "salt level test" .   Is the `clean cycle` a command as well?

Ylianst, 2024-09-06:
That is also what I am hoping, so, you can try the commands I gave above and see if it makes a difference.

Yes, `clean cycle`, `summer timer`, `spa lock` all have status flags and also commands to change them. This is why I suspect that if the salt confirmation has a status flag, it may also have a command.

derhally, 2024-09-15:
@Ylianst So I would say that monitoring for the salt confirmation is working great.  We can close this issue and we can continue the discussion on the commands on the general thread

derhally, 2024-09-15:
@Ylianst i don't know if you wanted me to put the info in the general thread, but here is what I found

01 1F 40 0B2702 <- Decreases Interior lights Dim level
01 1F 40 0B2703 <- Increases Interior lights Dim Level

Ylianst, 2024-09-15:
Thanks you @derhally. That is wonderful to hear, also that you found the use for another command! I just updated the documentation with your interior dim level find. Excellent!

closed 2024-09-15 by Ylianst