Ylianst/ESP-IQ2020 #20: Status Light For Atom M5?

mediamanrit, 2024-09-09:
I've been trying to make this work and just can't.  My ESPHome skills aren't up to par yet.  Is there a way to be able to light the RGB LED when the Atom boots, and leave it lit?

Why?  Because it seems like my Atom is just not working, BUT, I can write software to it.  It's not coming up on my home WiFi, but it also isn't advertising its own AP/Captive Portal either.  So, having an "I'm booted" status light would just prove it's at least on.

Better would be a thing like:

- Blue: Booted (Step 1)
- Green: Connected to WiFi (Step 2)

Ylianst, 2024-09-10:
Is it this device you have?

[image]

If so, add this:

```
light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: 27
    num_leds: 1
    rgb_order: GRB
    id: status_led
    name: M5Atom Light
    effects:
      - random:
      - flicker:
      - addressable_rainbow:

binary_sensor:
    platform: gpio
    pin:
      number: 39
      inverted: true
    name: M5Atom Button
    on_press:
      then:
        - light.toggle: status_led
        - homeassistant.event:
            event: esphome.button_pressed
            data:
              button: M5Atom Button
    on_press:
      then:
      - light.toggle: status_led
```

This should make it so the LED will do something when you press the button and so, signal that something is alive.

Also, the ATOM has only 2.4Ghz WIFI, so, if your router is 5Ghz, it will not connect.

mediamanrit, 2024-09-10:
I had to remove that last on_press: stanza (it was a dup from above a few lines), but that gave me something to prove it's on.  This will help a lot.  Thanks!  On to the next problem...

closed 2024-09-10 by mediamanrit

Ylianst, 2024-09-10:
Nice! For WIFI, lots to say, but if it can't connect, you can have it be a WIFI hotspot with a captive portal, that could be the next step.

mediamanrit, 2024-09-10:
It's connecting to WiFi now.  Sorry to say, bad access point.  My challenge now is, no free expansion ports.  I closed it up for the day, but, I'll take some photos later in the week.  Since RS485 is just taps on a bus, I may make a splitter cable of sorts...but I'll need to find the proper connectors to do that.

Ylianst, 2024-09-10:
Yes, a splitter cable should work fine. Do you have 2 or two expansion ports? What modules do you have?

Normally [it's not a cheap board](https://www.myhotspring.com/products/i2c-comm-5-port). I have one and can confirm it's a dumb splitter for the RS485 pins. The two I2C pins seems to have plenty of pull up resistors, but does not matter for this use case.