Ylianst/ESP-IQ2020 #21: 2017 Grandee NXT

cheungbe, 2024-09-20:
Hi, I have a 2017 Grandee NXT that I tried to connect to via the peripheral port and it did not work. Seemingly caused the board to crash/reboot as I could hear the power relay clicking on/off for the ozonator every 10 seconds.

Potentially a bit reckless to try and connect without testing the pin outs for 12v, that will be my next step.

Anything else you would recommend in terms of low-risk troubleshooting? 
[image]

Ylianst, 2024-09-20:
Hi @cheungbe. Thanks for reporting. I would expect this to work and your wiring looks correct. The first thing I would do is disconnect the ESP32 and test the 4 wires to make sure there is no shorts. Check each pair. If resistance goes to zero, there is a problem. Next, find a 12v source (old power brick) and connect it to make sure it works as expected, that is connects to WIFI, etc.

You can also check that you have 12v on the hot tub side, if that is ok, you should able to connect only ground and 12v and see the device connect to WIFI. Lastly, connect the RS485 pins.

I don't see anything wrong with your setup, 2017 model should have no issue that I know of.

closed 2024-10-30 by Ylianst

cheungbe, 2024-10-30:
@Ylianst - Turns out it was a faulty ATOM Lite, replaced it and it worked like a charm.

Ylianst, 2024-10-30:
@cheungbe Oh wow. Thanks for reporting back on this. I am very happy it's working for you now.