Ylianst/ESP-IQ2020 #22: Where to get the extension board?

RWag-AC, 2024-09-21:
Hi Ylianst,

We own an 2024 JetsetterLX with Freshwater and IQ.
When I tried to mount the interface, I discovered that both interface terminals are in use.
So I need to get this extension board.
Where can I get one of these?

Regards.

RWag-AC

Ylianst, 2024-09-22:
Hi. Search for "Watkins 74150" online. I think the older versions have 8 ports, the new ones have 5 ports but both should work. Note that you could make your own by just splitting each pin, if both modules use RS485. If the I2C pins are using there are extra pull up resistors. In any case, there are some links I found ([link](https://www.myhotspring.com/products/i2c-comm-5-port)) ([link](https://www.walmart.com/ip/Watkins-Caldera-Spas-74150-1117701-8-Port-Communication-Hub-1117701-01/572751234)) ([link](https://www.ebay.com/itm/116303826926)) ([link](https://backyardplus.com/parts/74150-i2-c-rs485-5-port-comm-hub-iq-2020/))

I personally got mine off eBay for around 40$, I did not really need it, but it did allow me to study the wiring of it. Hope this helps.

Ylianst, 2024-09-22:
Oh, you did mention you have Freshwater IQ. I don't have this module myself, but that is a module I may add support for and will need someone to test it.

RWag-AC, 2024-09-22:
Thank You very much for the info. This is pretty expensive for a passiv board like this. I try to find something here in Europe, and if I do not succed, I will think about building my own only for RS485.
On the other hand, this is a brand new spa and my wife will certainly kill me if I spoil it.
We have the latest technology from Hot Spring with all they can put into a Jetsetter LX, so if I manage to solve the interface problem, I surely want to give it a try...

Ylianst, 2024-09-22:
Yes, everything hot tub related has a high profit margin. This is a passive board that if someone designed and sent to PCBWay, could make for a few dollars. If you are ever tempted to make your own wired splitter, look for the following two connectors online: `3-644540-8` and `3-647000-8`. I have never ordered these, but I got these numbers from someone on a forum how previously had tried to build a control module.

RWag-AC, 2024-09-23:
I have orderd one board from a shop in the Netherlands for 75€ plus pp. I hope it isn't a fake one...

Ylianst, 2024-09-23:
Nice. When you get it, please take pictures and report back.

spangenberg, 2024-09-24:
> Oh, you did mention you have Freshwater IQ. I don't have this module myself, but that is a module I may add support for and will need someone to test it.

@Ylianst signing up as a guinea pig to test Freshwater and IQ as well, just got my Aria yesterday.

Ylianst, 2024-09-24:
Thank you @spangenberg. I will start working on the integration in the coming days, hopefully it should not be too difficult.

RWag-AC, 2024-09-25:
I got the board today, and I think, it is the original one.

[image]

If its dry outside, I will mount it on the weekend...

Ylianst, 2024-09-25:
Yes, this is the original 8 port one. Looks good, this is the same one I have.

Ylianst, 2024-09-26:
@spangenberg  Initial Freshwater IQ support was added to the integration, [instructions here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/extras.md#freshwater-iq-experimental). Please report back anything and everything as I don't have this module and so, need community feedback.

Ylianst, 2024-10-30:
Close this since it's resolved.

closed 2024-10-30 by Ylianst