Ylianst/ESP-IQ2020 #23: 2004 Hotspring Prodigy-P not working

hgontijo, 2024-10-15:
Hello, I just want to report that the ESP-IQ2020 didn't work for the 2004 Hotspring Prodigy-P.
In my case, it looks like the ESP32 didn't even get power because it didn't show up on the WIFI network. 

[image]

Ylianst, 2024-10-15:
Arg. Sorry about that. I do have a note that models 2001 to mid-2009 do not work. It's not impossible that someone could make it work, but since I do not have this tub model, I can't try it.

In these older models, the power is clearly on different pins. On eBay, I see the "WATKINS 74294 iCAST I2C Interface" that looks like this:

[image]

This is clearly a module for older models and the pins on the connector do not match the new models. I would guess that this module would work on your tub and we could discover the voltage and interface.

Ylianst, 2024-10-15:
Ok, just added it to the front page:

[image]

Thank you for reporting this.

hgontijo, 2024-10-16:
Do you have some tips on how to discover the voltage and interface? Using raspberry pi or some serial to USB cable?

Ylianst, 2024-10-16:
Hi @hgontijo. Sure thing. If you have a bit of knowledge in this space, I will gladly help. For 2014 and after tubs, this is the connector pins I found. Note that I have never used the I2C pins, put they do seem to have pull up resistors all over the line for both these pins so, I am pretty sure they are I2C. I also did put a oscilloscope on the I2C pins and did notice they are high.

[image]

For pre-2014, in looking at the image above, I see (again, this is my guess):

- 8 - 
- 7 -
- 6 -
- 5 - Green (I2C)
- 4 - Black (Ground)
- 3 - White (I2C)
- 2 - Red (V+)
- 1 - 

So, now that I look at this, it's exactly what I expected. I know the ESP32 is not powering up, so, I would test pins 2 and 4 to see if the voltage is 12v. I would not be surprised if the voltage was 5v instead (but this is a complete guess). If you do have 5v, it would explain why the ESP32 is not powering up but that would be great since you could connect the ESP32 directly to 5v and connect the ESP32 I2C pins directly to the pins 3 and 5. You can use the ATOM you got, just remove the RS485 module and get a cable from the ESP32 5v/ground/i2c direct to the pins. I think the M5Stack has a GROVE connector, so [something like this would work](https://my.cytron.io/p-grove-4-pin-buckled-to-female-cable).

[image]

For looking at the I2C protocol, a friend did give me for a while this [I2C Driver](https://i2cdriver.com/). It had good software to scan the I2C addresses, but you may be able to do the same with the ESP32 or a Raspberry Pi.

I will note that for RS485, I did buy the internet connection module and so, was able to listen to all the traffic on the bus and figure it out. Once you do connect correctly to I2C, I am not sure how to go about figuring out the commands or what set of commands are even available. So, this is a completely new world.

Hope this helps. Please report back your findings if any. I (and many others) would be super interested.

Ylianst, 2024-10-16:
FYI. Old sheet I made when starting out this project to figure out the pins. I just looks at every picture of modules I could find:

[image]

Ylianst, 2024-10-30:
Closing, please re-open if there is more on this topic.

hgontijo, 2024-10-30:
I'm still waiting for the Grove connector to arrive. Once I receive it, I will test and report the results.

Ylianst, 2024-10-30:
@hgontijo Oh nice!! Thanks for the update.

Ylianst, 2025-01-02:
Ok. closing. Please re-open anytime if there is new data.

closed 2025-01-02 by Ylianst

crisperfx, 2025-06-15:
Hello, I have the same motherboard but for the Hotspring Jetsetter model from 2005. I have done the test with the RS485 module on the pins indicated on the example. However, there was no way to get this working with RS485. 
Note: on pin 2 and pin 4 there is 9 volts. So it should work with the RS485 Tail to power up the M5Stack ATOM Lite ESP32. 
Conclusion this is a motherboard that can only be read with I2C signal. In the meantime I have the setup ready to let the sniffer do its thing on pin 3 and pin 5. 
I will keep you informed. 
Kind regards Dempsy

Ylianst, 2025-06-16:
@crisperfx Nice. Early on when trying to figure out the connector, a friend let me borrow a I2C debugging board and since you can see the pull up resistors on the expansion board, I had a good idea what pins where the I2C pins. In my case having a later generation board, I never got any signal on these pins. However, I bet it would be different on your board. A long time ago, I looks all over the internet to see what pictures I could find for what module and recorded it in a Excel sheet.

[image]

Because the I2C protocol is well defined, I bet you could try a lot of different read/write commands and find out everything that is possible with it. Indeed, if you can power the M5Stack ATOM Lite ESP32 and wire it's I2C pins to the IQ2020, it's possible you can git it working.