Ylianst/ESP-IQ2020 #24: Component Already Defined Errors?

efaden, 2024-11-17:
See error below.  Any ideas?

Config

```
external_components:
  - source: github://ylianst/esp-iq2020
    refresh: 30s

uart:
  id: SpaConnection
  tx_pin: GPIO26
  rx_pin: GPIO32
  baud_rate: 38400

iq2020:
  uart_id: SpaConnection
#   flow_control_pin: GPIO0
  port: 1234
  audio_emulation: true

binary_sensor:    
  - platform: iq2020-dev
    salt_level_confirmed:
      name: Salt Level Confirmed

# If using celsius units on the hot tub remote, replace _f_ with _c_ in the three entries below.
# Feel free to remove any sensor that are not relevent for your hot tub.
sensor:
    #measurement counters
  - platform: iq2020
    current_${celcius_farenheit}_temperature:
      name: Current Temperature

    target_${celcius_farenheit}_temperature:
      name: Target Temperature

    outlet_${celcius_farenheit}_temperature:
      name: Heater Outlet

    pcb_${celcius_farenheit}_temperature:
      name: Controller Temperature

    salt_content:
      name: Salt Content

    #Lifetime counters
    lifetime_runtime:
      name: Lifetime Runtime
      on_value:
        then:
          - lambda: |-
              ESP_LOGD("main", "The current version is %s", x.c_str());
      id: jacuzzi_total_runtime

    heater_total_runtime:
      name: Heater Runtime
      id: heater_total_runtime

    circulation_pump_total_runtime:
      name: Circulation Pump Runtime
      id: circulation_pump_total_runtime

    lights_total_runtime:
      name: Lights Runtime
      id: lights_total_runtime      

    jets1_total_runtime:
      name: Jets 1 Runtime
      id: jets1_total_runtime
[... 490 lines cut ...]
In file included from src/esphome.h:36,
                 from src/main.cpp:3:
src/esphome/components/iq2020/iq2020_select.h:9:8: note: previous definition of 'class esphome::iq2020_select::IQ2020Select'
  class IQ2020Select : public select::Select, public Component {
        ^~~~~~~~~~~~
In file included from src/esphome.h:44,
                 from src/main.cpp:3:
src/esphome/components/iq2020-dev/iq2020_switch.h:9:8: error: redefinition of 'class esphome::iq2020_switch::IQ2020Switch'
  class IQ2020Switch : public switch_::Switch, public Component {
        ^~~~~~~~~~~~
In file included from src/esphome.h:37,
                 from src/main.cpp:3:
src/esphome/components/iq2020/iq2020_switch.h:9:8: note: previous definition of 'class esphome::iq2020_switch::IQ2020Switch'
  class IQ2020Switch : public switch_::Switch, public Component {
        ^~~~~~~~~~~~
In file included from src/esphome.h:45,
                 from src/main.cpp:3:
src/esphome/components/iq2020-dev/iq2020_text.h:9:8: error: redefinition of 'class esphome::iq2020_text::IQ2020Text'
  class IQ2020Text : public text::Text, public Component {
        ^~~~~~~~~~
In file included from src/esphome.h:38,
                 from src/main.cpp:3:
src/esphome/components/iq2020/iq2020_text.h:9:8: note: previous definition of 'class esphome::iq2020_text::IQ2020Text'
  class IQ2020Text : public text::Text, public Component {
```

efaden, 2024-11-17:
It was 
```

binary_sensor:    
 - platform: iq2020-dev
   salt_level_confirmed:
   name: Salt Level Confirmed
```

Ylianst, 2024-11-18:
@efaden Your correct, use `iq2020`, do not use `iq2020-dev`. So, just replace `iq2020-dev` with `iq2020` and that should fix it. The dev version is for my own development, it's sort of a early access version and it's the one I personally use for my own development. Once things work, I move features over to the non-dev version.

In the case above, because both dev and non-dev versions where used at the same time, it would cause the duplicate errors.

If there is any `-dev` in the documentation, let me know, I need to fix it. Thanks.

Ylianst, 2024-11-18:
Closing this since it's fixed. Please re-open if anything need to be fixed. Thanks.

closed 2024-11-18 by Ylianst