Ylianst/ESP-IQ2020 #25: Issues with 2024 Pulse

efaden, 2024-11-18:
1) In IQ2020_select.cpp on line 28 to 32 it has 

```
		case SELECT_LIGHTS4_COLOR:
			if ((this->traits.get_options().size() < 7) || (this->traits.get_options().size() > 8)) {
				this->traits.set_options(lights_colors_values);
			}
			break;
```
to check if the color selection parameters are valid.  Issue is that with the pulse there are 9.... 

        - 1 -> Blue
        - 2 -> Aqua
        - 3 -> Green
        - 4 -> White
        - 5 -> Yellow
        - 6 -> Red
        - 7 -> Magenta
        - 8 -> Color Wheel On
        - 9 -> Color Wheel Off

So that prevents you from using them all because if you do its >8 so it uses the defaults.

2) Even though the intensity values for the tub are 0 (off) and 1 to 3 ... it still seems to report 5 as the default intensity for the sensors. Not sure why?

Thanks!

efaden, 2024-11-18:
Also... if interested you can use text templating to make the values report something more useful....

Sensors
```
    #light sensors      
    logo_lights:
      name: Logo Lights
      on_value:
        then:
          - text_sensor.template.publish:
              id: logo_lights_text
              state: !lambda 'return (to_string(int(x)).c_str());'

    lights_intensity:
      name: Light Intensity
      on_value:
        then:
          - text_sensor.template.publish:
              id: lights_intensity_text
              state: !lambda 'return (to_string(int(x)).c_str());'

    lights_underwater_intensity:
      name: Underwater Light Intensity
      on_value:
        then:
          - text_sensor.template.publish:
              id: lights_underwater_intensity_text
              state: !lambda 'return (to_string(int(x)).c_str());'

    lights_bartop_intensity:
      name: Bartop Light Intensity
      on_value:
        then:
          - text_sensor.template.publish:
              id: lights_bartop_intensity_text
              state: !lambda 'return (to_string(int(x)).c_str());'

    lights_pillow_intensity:
      name: Pillow Light Intensity
      on_value:
        then:
          - text_sensor.template.publish:
              id: lights_pillow_intensity_text
              state: !lambda 'return (to_string(int(x)).c_str());'

    lights_exterior_intensity:
      name: Exterior Light Intensity
      on_value:
        then:
          - text_sensor.template.publish:
              id: lights_exterior_intensity_text
              state: !lambda 'return (to_string(int(x)).c_str());'

    lights_underwater_color:
      name: Underwater Light Color
      on_value:
        then:
          - text_sensor.template.publish:
              id: lights_underwater_color_text
              state: !lambda 'return (to_string(int(x)).c_str());'

    lights_bartop_color:
      name: Bartop Light Color
      on_value:
        then:
          - text_sensor.template.publish:
              id: lights_bartop_color_text
              state: !lambda 'return (to_string(int(x)).c_str());'

    lights_pillow_color:
      name: Pillow Light Color
      on_value:
        then:
          - text_sensor.template.publish:
              id: lights_pillow_color_text
              state: !lambda 'return (to_string(int(x)).c_str());'

    lights_exterior_color:
      name: Exterior Light Color
      on_value:
        then:
          - text_sensor.template.publish:
              id: lights_exterior_color_text
              state: !lambda 'return (to_string(int(x)).c_str());'
```

and the text_sensors:
```
  # lights
  - platform: template
    id: logo_lights_text
    name: Logo Lights (Text)
    filters:
      - map:
        - 0 -> Unknown
        - 1 -> Power on
        - 2 -> Power on & ready on
        - 3 -> Ready flash
        - 4 -> Power flash
        - 5 -> Power and ready flash
        - 6 -> Power and ready alternate
        - 7 -> Power and ready salt error

  - platform: template
    id: lights_intensity_text
    name: Light Intensity (Text)
    filters: &LIGHT_INTENSITY_FILTER
      - map:
        - 0 -> Off 
        - 1 -> Low
        - 2 -> Mid Low
        - 3 -> Mid
        - 4 -> Mid High
        - 5 -> High

  - platform: template
    id: lights_underwater_intensity_text
    name: Underwater Light Intensity (Text)
    filters: *LIGHT_INTENSITY_FILTER

  - platform: template
    id: lights_bartop_intensity_text
    name: Bartop Light Intensity (Text)
    filters: *LIGHT_INTENSITY_FILTER

  - platform: template
    id: lights_pillow_intensity_text
    name: Pillow Light Intensity (Text)
    filters: *LIGHT_INTENSITY_FILTER

  - platform: template
    id: lights_exterior_intensity_text
    name: Exterior Light Intensity (Text)
    filters: *LIGHT_INTENSITY_FILTER

  - platform: template
    id: lights_underwater_color_text
    name: Underwater Light Color (Text)
    filters: &LIGHT_COLOR_FILTERS
      - map:
        - 1 -> Blue
        - 2 -> Aqua
        - 3 -> Green
        - 4 -> White
        - 5 -> Yellow
        - 6 -> Red
        - 7 -> Magenta
        - 8 -> Color Wheel On
        - 9 -> Color Wheel Off

  - platform: template
    id: lights_bartop_color_text
    name: Bartop Light Color (Text)
    filters: *LIGHT_COLOR_FILTERS

  - platform: template
    id: lights_pillow_color_text
    name: Pillow Light Color (Text)
    filters: *LIGHT_COLOR_FILTERS

  - platform: template
    id: lights_exterior_color_text
    name: Exterior Light Color (Text)
    filters: *LIGHT_COLOR_FILTERS
```

Ylianst, 2024-11-19:
Hi @efaden. Thanks for the text templating! I did not know about this at all. When I get a change, I will look into this and update the installation script.

For the colors, I have a Hotspring Grandee and on mine, the user interface will show 9 color options, but internally, the color and cycling are different variables. So, you really have 7 color options + cycling options. Did you really try to change the code to allow 9 colors and used colors 1 to 9 as you indicated? I don't think that would work.

Instead, I think you should do this:

```
select:
  - platform: iq2020
    name: Color Underwater
    id: lights1_color
    datapoint: 1
    options:
      - Blue
      - Aqua
      - Green
      - White
      - Yellow
      - Red
      - Magenta
      - Cycle
  - platform: iq2020
    name: Color Bartop
    id: lights2_color
    datapoint: 2
    options:
      - Blue
      - Aqua
      - Green
      - White
      - Yellow
      - Red
      - Magenta
      - Cycle
  - platform: iq2020
    name: Color Pillow
    id: lights3_color
    datapoint: 3
    options:
      - Blue
      - Aqua
      - Green
      - White
      - Yellow
      - Red
      - Magenta
      - Cycle
  - platform: iq2020
    name: Color Exterior
    id: lights4_color
    datapoint: 4
    options:
      - Blue
      - Aqua
      - Green
      - White
      - Yellow
      - Red
      - Magenta
      - Cycle
  - platform: iq2020
    name: Color Cycle Speed
    id: lights_cycle_speed
    datapoint: 5

number:
  - platform: iq2020
    id: lights1_intensity
    name: Intensity Underwater
    datapoint: 7
    maximum: 3
  - platform: iq2020
    id: lights2_intensity
    name: Intensity Bartop
    datapoint: 8
    maximum: 3
  - platform: iq2020
    id: lights3_intensity
    name: Intensity Pillow
    datapoint: 9
    maximum: 3
  - platform: iq2020
    id: lights4_intensity
    name: Intensity Exterior
    datapoint: 10
    maximum: 3
```

So, the colors will have one "cycle" color, then you need to change the "cycle speed" to to change the cycle mode. Also, for light intensity, use `maximum: 3` for you model. It's a bit weird, but that is how it works internally.

Obviously, I could be completely wrong but the lights color values, intensity levels and cycle modes seem to be completely different between hot tub models, so, you need to adjust the values yourself. Let me know if this works for you.

Ylianst, 2024-11-19:
I updated the logo lights sensor to a text version in the documentation. That is much better. The other values, like light color and light intensity are read-only sensors that I was not sure if they are useful to keep given I have the selectors now.

efaden, 2024-11-19:
I didn't change the code.  That's the line preventing me from having the same number of options as are on my tub though.  I haven't fully gone through all the menus yet.  May have to debug some of the protocol though.  

As for the color options those are the ones I have .... 

        - 1 -> Blue
        - 2 -> Aqua
        - 3 -> Green
        - 4 -> White
        - 5 -> Yellow
        - 6 -> Red
        - 7 -> Magenta
        - 8 -> Color Wheel On
        - 9 -> Color Wheel Off

I'll play with it more later and see what options there actually are on the tub.  I may have to pull the RS485 data to see how the relate.

I almost got the grande... but got a great deal on this one off the floor....

Ylianst, 2024-11-20:
Sure, I just changed the code to this, let me know what you find:

```
			if ((this->traits.get_options().size() < 7) || (this->traits.get_options().size() > 9)) {
				this->traits.set_options(lights_colors_values);
			}
```

efaden, 2024-11-20:
Should work.  I guess I'm just not clear why even check the count... Or are
you thinking it's like a sanity check?

Ylianst, 2024-11-20:
Well, let me know if it does work.

efaden, 2024-11-20:
These are the values I am getting....

        - 1 -> Magenta
        - 2 -> Blue
        - 3 -> Aqua
        - 4 -> Green
        - 5 -> White
        - 6 -> Yellow
        - 7 -> Red
        - 8 -> Color Wheel On
        - 0 -> Color Wheel Off

        - 0 -> Off 
        - 1 -> Low
        - 3 -> Medium
        - 5 -> High

I don't think I can actually do exactly that the way it is setup.  For example number needs to allow 4 values for intensity.... 0,1, 3,5 for example....   And the color seems to set the value to 0 for off.

Ylianst, 2025-04-24:
Ok, I will close this since it's solved. Please re-open if needed. Lots of useful data here to people with this setup.

closed 2025-04-24 by Ylianst