Ylianst/ESP-IQ2020 #27: error after an ESP update today

CHief-Wiggum1203, 2024-11-28:
Hi, I'm getting this error after an ESP update today

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.6
Compiling .pioenvs/whirlpool/src/esphome/components/iq2020/iq2020.cpp.o
Compiling .pioenvs/whirlpool/src/esphome/components/iq2020/iq2020_number.cpp.o
Compiling .pioenvs/whirlpool/src/esphome/components/iq2020/iq2020_text.cpp.o
Compiling .pioenvs/whirlpool/src/esphome/components/logger/logger.cpp.o
In file included from src/esphome/components/iq2020/iq2020_number.cpp:2:
src/esphome/components/iq2020/iq2020_number.h:4:10: fatal error: esphome/components/number/number.h: No such file or directory
 #include "esphome/components/number/number.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/whirlpool/src/esphome/components/iq2020/iq2020_number.cpp.o] Error 1
src/esphome/components/iq2020/iq2020.cpp: In member function 'virtual void IQ2020Component::setup()':
src/esphome/components/iq2020/iq2020.cpp:53:42: error: 'number_state' was not declared in this scope
  for (int i = 0; i < NUMBERCOUNT; i++) { number_state[i] = number_pending[i] = NOT_SET; }
                                          ^~~~~~~~~~~~
src/esphome/components/iq2020/iq2020.cpp:53:42: note: suggested alternative: 'switch_state'
  for (int i = 0; i < NUMBERCOUNT; i++) { number_state[i] = number_pending[i] = NOT_SET; }
                                          ^~~~~~~~~~~~
                                          switch_state
src/esphome/components/iq2020/iq2020.cpp:53:60: error: 'number_pending' was not declared in this scope
  for (int i = 0; i < NUMBERCOUNT; i++) { number_state[i] = number_pending[i] = NOT_SET; }
                                                            ^~~~~~~~~~~~~~
src/esphome/components/iq2020/iq2020.cpp:53:60: note: suggested alternative: 'switch_pending'
  for (int i = 0; i < NUMBERCOUNT; i++) { number_state[i] = number_pending[i] = NOT_SET; }
                                                            ^~~~~~~~~~~~~~
                                                            switch_pending
*** [.pioenvs/whirlpool/src/esphome/components/iq2020/iq2020.cpp.o] Error 1
========================== [FAILED] Took 5.44 seconds ==========================

// Ursprünglich (die fehlerhaften Zeilen):
for (int i = 0; i < NUMBERCOUNT; i++) {
  number_state[i] = number_pending[i] = NOT_SET;
}

// Ersetzt durch:
for (int i = 0; i < NUMBERCOUNT; i++) {
  switch_state[i] = switch_pending[i] = NOT_SET;
}

thanks Erich

Ylianst, 2024-11-29:
i just updated my Home Assistant and ESP Home to the latest this morning and tried updating and everything worked. just for sanity, can you "Clean Build Files" and try again:

[image]

Let me know if it fixes it or not. The key is `esphome/components/number/number.h`, looks like a file in ESPHome is missing, this is a pretty basic one.

CHief-Wiggum1203, 2024-12-01:
> Von "Ylian Saint-Hilaire" ***@***.***>
> Cc "CHief-Wiggum1203" ***@***.***>; "Author"
> Datum 29.11.2024 22:58:29
> Betreff Re: [Ylianst/ESP-IQ2020] error after an ESP update today (Issue
> #27)

Hi, oh sorry, it was my mistake, it works great

Thx Erich

closed 2024-12-01 by Ylianst