Ylianst/ESP-IQ2020 #28: Color loop

dhellerstedt, 2024-11-28:
My Caldera 2023 has the standard underwater colors, but instead of selecting the color cycle in the color selection, there is another option called Loop with 0-3 to cycle through the colors. 0-3 just controls the speed of the color cycle. How can I set loop or a color?

Ylianst, 2024-12-01:
Hi @dhellerstedt. Thanks for installing this integration. So, for hot tub lights, colors and cycling, there does not seem to be any "standard" way this is handled internally. This is probably why the Watkins app does not allow the remote user to control the lights, every model seems to be doing it a bit differently with different color mapping, etc.

If you have not done so already, looks the [Controlling Lights Color, Intensity and Cycle Speed](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/extras.md#controlling-lights-color-intensity-and-cycle-speed) section to add the base configuration to control the lights and cycling. Then, you will need to edit the configuration to match your hot tub model.

If you work it out, please come back and post the configuration you came up with so I can post it in the [tested page here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/tested.md). Let me know if this helps.

dhellerstedt, 2024-12-08:
Thanks @Ylianst, 

Caldera Vanto 2023
Status: Works great. Lights, temperature, Pump1 with 2 speeds, Pump2 with 1 speed, cleaning cycle.
Version: 	112T b5e1A002 v2.6
Interface: M5Stack-ATOM + Tail485
Watkins P/N: 1303401-1 Rev K
Notable Settings:
```
uart:
  id: SpaConnection
  tx_pin: GPIO26
  rx_pin: GPIO32
  baud_rate: 38400

iq2020:
   uart_id: SpaConnection
#   flow_control_pin: GPIO0
   port: 1234
   audio_emulation: false

# Example configuration entry
web_server:
  port: 80

# Sync time with Home Assistant. Enable time component to reset energy at midnight.
time:
  - platform: homeassistant
    id: homeassistant_time

button:
  - platform: restart
    name: "ESP Restart"
    
binary_sensor:    
  - platform: status
    name: ESP Status

# If using celsius units on the hot tub remote, replace _f_ with _c_ in the three entries below.
# Feel free to remove any sensor that are not relevent for your hot tub.
sensor:
    #measurement counters
  - platform: iq2020
    current_${celcius_farenheit}_temperature:
      name: Current Temperature
    target_${celcius_farenheit}_temperature:
      name: Target Temperature
    outlet_${celcius_farenheit}_temperature:
      name: Heater Outlet
    pcb_${celcius_farenheit}_temperature:
      name: Controller Temperature

    #Lifetime counters
    lifetime_runtime:
      name: Lifetime Runtime
      on_value:
        then:
          - lambda: |-
              ESP_LOGD("main", "The current version is %s", x.c_str());
      id: jacuzzi_total_runtime
    heater_total_runtime:
      name: Heater Runtime
      id: heater_total_runtime
    circulation_pump_total_runtime:
      name: Circulation Pump Runtime
      id: circulation_pump_total_runtime
    lights_total_runtime:
      name: Lights Runtime
      id: lights_total_runtime      
    jets1_total_runtime:
[... 325 lines cut ...]
          (String(seconds)) 
          ).c_str()};
  - platform: template
    id: total_runtime_$device_6
    name: "Total Runtime_$device_6"
    update_interval: $sensor_update_frequency
    icon: "mdi:timer-sand"
    lambda: |-
      int seconds = round(id(${device_6}_total_runtime).state);
      int years = seconds / (24 * 3600);
      int months = seconds / (24 * 3600);
      int days = seconds / (24 * 3600);
      seconds = seconds % (24 * 3600);
      int hours = seconds / 3600;
      seconds = seconds % 3600;
      int minutes = seconds /  60;
      seconds = seconds % 60;
        return {
          ((days ? String(days) + " days, ": "") +
          (hours ? String(hours) + ":": "") +
          (minutes ? String(minutes) + ":": "") +
          (String(seconds)) 
          ).c_str()};
```
[image]

Ylianst, 2024-12-09:
Thanks @dhellerstedt. I just added it to the tested page. This is the first time I see a picture of the new updated IQ2020 board. I need the expansion connectors are now horizontal. Thanks for that pic.

mark-r-cullen, 2025-04-17:
To add to the list, I have retrofit my Tiger River Spa Caspian (posted an update on my old closed issue) with a 50hz Eagle board and this is the light config that works for me. 

The difference from the document version is the colours were off by one, cycle didn't work until I took out the Cycle colour and 3 intensity levels instead of 5.

`select:
  - platform: iq2020
    name: Color Underwater
    id: lights1_color
    datapoint: 1
    options:
      - Blue
      - Cyan
      - Green
      - White
      - Yellow
      - Red
      - Violet
  - platform: iq2020
    name: Color Cycle Speed
    id: lights_cycle_speed
    datapoint: 5
number:
  - platform: iq2020
    id: lights1_intensity
    name: Intensity Underwater
    datapoint: 7
    maximum: 3`

Also maybe to note, I've fit a cheap (£35) Sloan 12 led light as part of the retrofit as the old light doesn't work with the newer board, and I can't find the cheapest genuine led light in stock in the UK anywhere. Its a bit dimmer I think, but there is a 27 led version which is still only about £60.

Ylianst, 2025-04-17:
Very interesting @mark-r-cullen. Thanks for that report. I am interested in your light setup. Is [this what you got](https://sloanled.com/downloads/SpecSheet-SloanLEDPrism12-US-Canada.pdf)?.

For your configuration, I will post this in the list of tested ones noting the board update.

mark-r-cullen, 2025-04-21:
@Ylianst this is the led that I have working. https://parts4tubs.co.uk/product/sloan-led-12-led-cluster-liqualed/?gad_source=1&gbraid=0AAAAA-LHA3qtDPCE3HhIZgLnqr9K-UwE7&gclid=Cj0KCQjw2ZfABhDBARIsAHFTxGwfNXJY-aCEwGte4ewIszvtxdIy32lJg4g0KYppxbNplL5FVwN3cFcaAjpdEALw_wcB

The supplier of my board didn't initially send the adapter cables that should work with this so I just used a 9v battery and a bit of common sense to figure out which was the +ve pin, and which pins were red, green and blue. I now have the adapter cables after proving to them that it would work for my retrofit.

I thought since my old light was 12 led that it should be bright enough, but it's definitely dimmer. I've ordered the 21 led version and will let you know if that gets the brightness back.

mark-r-cullen, 2025-04-23:
@Ylianst 

The 21 led version also works fine and I'm now also using the proper adapter cable (https://backyardplus.com/parts/76860-light-adapter-hot-spring-limelight-hot-spot/?srsltid=AfmBOooK07d-N2Z2aes_5ebp-vduxhn1_83PYcb1fEE8A9rwPqzrm_iK - provided with the Eagle control box). 

It's not dark yet, but I opened the lid a smidge and it definitely seems much brighter, probably about the same as the old blue only LED.

The actual light seems to be nothing special at all, it's just 12v LEDs and some resistors. You could make an even brighter one pretty easily (and I might if I get bored!)

closed 2025-06-28 by Ylianst