Ylianst/ESP-IQ2020 #31: Audio buttons is missing in full example

pettab, 2025-01-24:
Want to react to play/pause buttons but don’t have the audio button entity. I guess the full config example is missing this?

Ylianst, 2025-01-25:
No, it's there in [the documentation](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/audio.md). There is a sensor called "Buttons":

```
sensor:
    buttons:
      name: Buttons
```

It will switch to one of these values when you press a button on the remote:

```
1 - Play
2 - Pause
3 - Next
4 - Back
```

I will note that PLAY/STOP are a bit weird because when you power off the hot tub, it will hit STOP for you. When you power back on, it will try to go back to the previous state and may hit PLAY. So, these two buttons may cause odd effects when connected to lights, etc. So, just test it out.

Let me know if this works.

Ylianst, 2025-01-27:
I will close this since to response, but please report back anytime. Thanks!

closed 2025-01-27 by Ylianst