Ylianst/ESP-IQ2020 #33: RS485

mrjjomen, 2025-01-26:
Is it okay too connect this ESP device to same RS485 pins light controller is already using on my IQ2020 ?  I have only one aux port.
I looked expansion ports and they just set several connectors parallel to electrically the same as connecting wires together from different units.

Very nice project and very usefull too !

mediamanrit, 2025-01-27:
I effectively did the same thing.  My expansion board was full, so I bought some connectors and made a splitter for the rs485 and power pins.  Worked like a champ.

Ylianst, 2025-01-27:
@mrjjomen Yes. this is not only ok, but the expansion board is exactly this. Like routes the same lines to all the devices. On the expansion board, the two I2C have pull up resistors, but we don't use I2C so, yes, just split the pins and connect the ESP32.

closed 2025-01-27 by Ylianst