Ylianst/ESP-IQ2020 #34: Energy consumption for energy dashboard

milesroper, 2025-02-03:
Hi,
Trying to get power use showing in energy dashboard for the water heater sensor.  I think there is a  device_class: Energy setting missing.

https://community.home-assistant.io/t/integration-riemann-sum-integral-for-solar-panels-showing-unexpected-device-class/551996/15

Has a bunch of info, basically the riemann sensor is coming up with an error in the energy dashboard, which i think the root cause is the entity doesnt have the device class included.  Ive tried adding the device class into the esphone yaml config/and recompiled onto the esphome device.  I've checked the states in developer tools and the device class doesnt show up.  I figure somewhere in your code it defines those values.  Hopefully this makes sense if you can point me in the write direction would be appreciated.

Regards
Miles

milesroper, 2025-02-03:
Found some instructions for energy use in the other general discussion which i think will work.

milesroper, 2025-02-03:
Immm, says the new entity is unavailable, will leave it overnight and see if it self resolves.

Ylianst, 2025-02-03:
@milesroper - Give this a try, this is was I use. Add these sensors if you don't have then already. Let me know how it goes.

```
sensor:
  - platform: iq2020
    power_l1:
      internal: true
      name: Pump Power
      id: power_l1
    power_heater:
      internal: true
      name: Controller Power
      id: power_heater
    power_l2:
      internal: true
      name: Heater Power
      id: power_l2
  - platform: total_daily_energy
    name: 'Pump Energy'
    power_id: power_l1
    unit_of_measurement: 'kWh'
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
  - platform: total_daily_energy
    name: 'Controller Energy'
    power_id: power_heater
    unit_of_measurement: 'kWh'
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
  - platform: total_daily_energy
    name: 'Heater Energy'
    power_id: power_l2
    unit_of_measurement: 'kWh'
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    filters:
      # Multiplication factor from W to kW is 0.001
      - multiply: 0.001
```

milesroper, 2025-02-05:
The above is what I have, kind of, i used your full config example, so has lights/every setting you can think of.  Its working fine, other than the entitys are saying not responsive but they still update and show energy use.  Its a bit odd.

[image]

milesroper, 2025-02-05:
Heres my config

[esphome-web-fa1f1c.yaml.txt](https://github.com/user-attachments/files/18668507/esphome-web-fa1f1c.yaml.txt)

Ylianst, 2025-02-06:
To add to what I previously posted, my setting above should allow you to add your HotTub to the Energy monitoring panel of Home Assistant. Here is my screenshot. Note that my solar is showing as positive instead of negative... but regardless, you can see the Hotbu values.

[image]

If you go to "Settings", "Dashboards", "Energy". You should be able to add the values. Let me know if this works.

[image]

milesroper, 2025-02-06:
Yeah i managed to do this.  Its all working fine, shows the sensors as greyed out sometimes.  Its working now, and not greyed out, so maybe it just takes a day or so to sort itself out when you restart the esphome device.

Btw, its a really awesome integration.  Your pretty amazing to put this together.  Must have taken a bit.  Ill close this as sorted as at worse it seems to be a problem on restart for approx 1 day.

closed 2025-02-06 by milesroper