Ylianst/ESP-IQ2020 #35: Compiling problems

mrjjomen, 2025-02-16:
I am a bit confused what is wrong now.
I have my devices connected to PC at the moment and trying to install software via Home assistant ESP Add-on.

It first first time due to "audio_buttons:" defination I replaced that with just  "buttons:" and it goes further

INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/hotspring-sovereign.yaml...
INFO Detected timezone 'Europe/Helsinki'
Failed config
sensor.iq2020: [source /config/esphome/hotspring-sovereign.yaml:85]
  platform: iq2020
  **[audio_buttons] is an invalid option for [sensor.iq2020]. Did you mean [buttons]?
  audio_buttons:** 
    name: Audio Buttons
  lights_intensity: 

But then it fails again and now I have no clue what is wrong ??

INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/hotspring-sovereign.yaml...
INFO Detected timezone 'Europe/Helsinki'
INFO Generating C++ source...
INFO Compiling app...
Processing hotspring-sovereign (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Removing unused dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/api/api_connection.cpp.o
xtensa-esp32-elf-g++: fatal error: Killed signal terminated program cc1plus
**compilation terminated.
*** [.pioenvs/hotspring-sovereign/src/esphome/components/api/api_connection.cpp.o] Error 1
========**================= [FAILED] Took 35.54 seconds =========================

mrjjomen, 2025-02-16:
Is this compiling ongoing in RPi or ESP Atom board ?
I am using Raspberry Pi3B+ which has of course very limited memory, could this be because of it ?

Ylianst, 2025-02-16:
HI @mrjjomen. First, I looked in the repo and you are right, "audio_buttons" changed to "buttons" a long time ago and [this configuration](https://github.com/Ylianst/ESP-IQ2020/blob/3a2d5879fe2bfc787d5dc53f0f667e7b694008f8/documentation/configuration.md) still had the old name. I just fixed it.

The compile happens on the Raspberry Pi where Home Assistant is running, the ESP32 is so small, it has 320KB RAM and no file system, it could not run a compiler.

I see this in your logs:

```
xtensa-esp32-elf-g++: fatal error: Killed signal terminated program cc1plus
```

I pasted you log in ChatGPT and asked what could be the problem:

- The error message xtensa-esp32-elf-g++: fatal error: Killed signal terminated program cc1plus typically means that the compilation process was terminated by the operating system due to running out of memory (RAM). This can happen if your Home Assistant system (probably running on a Raspberry Pi or another low-memory system) doesn't have enough RAM to compile ESPHome.

So yes, you probably right. ChatGPT has suggestions for freeing RAM on the PI. You can ask it.

mrjjomen, 2025-02-16:
This second error was due to lack of memory in RPI3B+. Solved by inceasing swap file size: 
https://community.home-assistant.io/t/esphome-with-pi3b/313688/5

However there is still something wrong with audio_buttons -defination.

Compiling took some long time indeed:

INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/hotspring-sovereign.yaml...
INFO Updating https://github.com/ylianst/esp-iq2020.git@None
INFO Detected timezone 'Europe/Helsinki'
INFO Generating C++ source...
INFO Compiling app...
Processing hotspring-sovereign (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/binary_sensor/automation.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/binary_sensor/filter.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/button/button.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/climate/climate.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/climate/climate_mode.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/climate/climate_traits.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/copy/sensor/copy_sensor.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/esphome/ota/ota_esphome.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/fan/fan.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/fan/fan_state.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/internal_temperature/internal_temperature.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/iq2020/iq2020.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/iq2020/iq2020_climate.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/iq2020/iq2020_fan.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/iq2020/iq2020_number.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/iq2020/iq2020_select.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/iq2020/iq2020_switch.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/iq2020/iq2020_text.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/logger/logger_esp32.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/logger/logger_esp8266.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/logger/logger_host.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/logger/logger_libretiny.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/logger/logger_rp2040.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/hotspring-sovereign/src/esphome/components/mdns/mdns_esp32.cpp.o
[... 237 lines cut ...]
Compiling .pioenvs/hotspring-sovereign/FrameworkArduino/esp32-hal-spi.c.o
Compiling .pioenvs/hotspring-sovereign/FrameworkArduino/esp32-hal-time.c.o
Compiling .pioenvs/hotspring-sovereign/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pioenvs/hotspring-sovereign/FrameworkArduino/esp32-hal-tinyusb.c.o
Compiling .pioenvs/hotspring-sovereign/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pioenvs/hotspring-sovereign/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pioenvs/hotspring-sovereign/FrameworkArduino/firmware_msc_fat.c.o
Compiling .pioenvs/hotspring-sovereign/FrameworkArduino/libb64/cdecode.c.o
Compiling .pioenvs/hotspring-sovereign/FrameworkArduino/libb64/cencode.c.o
[... cut ...]
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/hotspring-sovereign/firmware.bin"], [".pioenvs/hotspring-sovereign/firmware.elf"])
Wrote 0x113bd0 bytes to file /data/build/hotspring-sovereign/.pioenvs/hotspring-sovereign/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/hotspring-sovereign/firmware.bin"], [".pioenvs/hotspring-sovereign/firmware.elf"])
======================== [SUCCESS] Took 1301.64 seconds ========================
INFO Successfully compiled program.

Ylianst, 2025-02-16:
Hi @mrjjomen. Yes, I did mention above that you where right about "audio_buttons" being replaced with "buttons" and I fixed this already. Let me know if you are talking about something else I missed. Thanks.

mrjjomen, 2025-02-16:
Okay, thanks for confirmation.
Regarding audio features, how do I remove them disturbing controls and sensor lists ? 
I tried to turn off audio emulation but it did not help. This could be documented as an example in configuration file.
  #audio_emulation: true
   audio_emulation: false

Ylianst, 2025-02-16:
@mrjjomen. Not sure what you mean, but on the main page of this integration, there is a "Sample ESP-Home configuration file". You a section to can open up. It's a minimal configuration and probably what you want to start with. I am going to paste it below. This gets you started with the basics and you can then add to this as needed. Most people do not need all the crazy features like Audio/ACE emulation and extra sensors, the basics keep it nice and simple. Let me know if this works for you.

```
esphome:
  name: hot-tub
  friendly_name: Hot Tub
  comment: "Luxury Spa"

esp32:
  board: m5stack-atom

logger:
  baud_rate: 0
  level: ERROR

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source: github://ylianst/esp-iq2020

# Make sure tx/rx pins are correct for your device.
# GPIO26/32 is ok for M5Stack-ATOM + Tail485, look in GitHub devices link for your device.
uart:
  id: SpaConnection
  tx_pin: GPIO26
  rx_pin: GPIO32
  baud_rate: 38400

iq2020:
   uart_id: SpaConnection
   polling_rate: 65
   port: 1234

select:
text:
number:

# If using Celsius units on the hot tub remote, replace _f_ with _c_ in the three entries below.
# Feel free to remove any sensor that are not relevant for your hot tub.
sensor:
  - platform: iq2020
    current_f_temperature:
      name: Current Temperature
    target_f_temperature:
      name: Target Temperature
    outlet_f_temperature:
      name: Heater Outlet
    power_l1:
      name: Pumps Power
    power_heater:
      name: Power Heater
    power_l2:
      name: Heater Power
    pcb_f_temperature:
      name: Controller Temperature

switch:
  - platform: iq2020
    name: Lights
    id: lights_switch
    icon: "mdi:lightbulb"
    datapoint: 0
  - platform: iq2020
    name: Spa Lock
    id: spa_lock_switch
    icon: "mdi:lock"
    datapoint: 1
  - platform: iq2020
    name: Temperature Lock
    id: temp_lock_switch
    icon: "mdi:lock"
    datapoint: 2
  - platform: iq2020
    name: Clean Cycle
    id: clean_cycle_switch
    icon: "mdi:vacuum"
    datapoint: 3
  - platform: iq2020
    name: Summer Timer
    id: summer_timer_switch
    icon: "mdi:sun-clock"
    datapoint: 4

fan:
  - platform: iq2020
    name: Jets 1
    id: jets1
    icon: "mdi:turbine"
    datapoint: 0
    speeds: 1
  - platform: iq2020
    name: Jets 2
    id: jets2
    icon: "mdi:turbine"
    datapoint: 1
    speeds: 2

# Set "celsius" to "true" if using celsius units.
climate:
  - platform: iq2020
    name: Temperature
    celsius: false

text_sensor:
  - platform: iq2020
    versionstr:
      name: Version
```

mrjjomen, 2025-02-16:
Don´t get me wring this absolutely fantastic and at the same usefull DIY.
But I wish I could get it how to get rid foff these audio controls ( I do not have audio features)

[image]

Ylianst, 2025-02-17:
It's not clear what you wrote, but for removing audio control: post your configuration here, make sure you compile and update your firmware and if you know the firmware was updated, you may need to remove the device in Home Assistant and have it detected again. However, it looks like you may have removed the audio controls and not updated the device's flash.

Ylianst, 2025-03-03:
Closing since this if fixed. Please re-open if needed

closed 2025-03-03 by Ylianst