Ylianst/ESP-IQ2020 #39: Monitoring a Freshwater Salt system

fclauson, 2025-03-02:
I am readimng you protocol file and trying to see if I can help witht the Freshwater system

all that I have on the key pad is 
a dial guage which  think shows "conductivity" (or how salty is the water 
a setting 1->9 of the "power level" 
a boost setting on or off
a change salt catridge option - which turns off the salt pump , waits, tells to change the catridge, waits for you to change, and then restarts the system

not sure what debug I can help with - but let me know

fclauson, 2025-03-02:
Looking at this - https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/protocol.md

trying to figure out how I can pull some of these values back to the Home Assistant

the dial gauage value would be useful (tells you if you need to top up with salt) 
the power level is useful - the tub auto decrements this if the tub is not used over time 
boost on/off is sort of interesting 
the catridge change would be useful to forwarn you when its needed - You can then either spoof it (fake a change in cartridge) or actually do it 

they seem to last longer than the suggested time limit based on tub usage

Ylianst, 2025-03-03:
Hi. Thanks exploring this. So, I can emulate the ACE system and figured out the values for that one, but I have a Freshwater system and it's been crazy, I have not been able to figure stuff out.

Do you have the DataViewer app setup and connected to the debug port on the ESP32? If so, you should be able to see all the traffic and record it. You can press buttons and see what messages get sent on the serial bus, etc. I did try to document what I [found here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/protocol.md#freshwater-salt-system). Since I don't personally use this module, I disconnected it in my hot tub.

If you use the module, probably the best is to setup the DataViewer and document of the coming weeks the values.

fclauson, 2025-03-03:
A quick monitor plus the manuel 

[Utopia Series 2020 Pre Delivery Instructions.pdf](https://github.com/user-attachments/files/19049231/Utopia.Series.2020.Pre.Delivery.Instructions.pdf)
[salt debug file 1.txt](https://github.com/user-attachments/files/19049232/salt.debug.file.1.txt)

Largosofthj, 2025-03-03:
Hi, First of all thank for starting this project. Great help to get my HotTub online.
Much appreciated!

I noticed that boost ON is working, but it is not possible for me to turn it OFF again.
Workaround is setting the FreshWater output level to "0". That will turn Boost OFF.

Created a logfile where I turn boost on/off with panel and app + adjusting the level.
There is also a hidden menu for adjusting the Power level of the salt system. Don't remember what I pushed for that menu.
I think it was level 1-4 connected to the size of the pool. Will try to find it and upload logfile.

[Applogfile.txt](https://github.com/user-attachments/files/19050537/Applogfile.txt)
[Panellogfile.txt](https://github.com/user-attachments/files/19050541/Panellogfile.txt)

homeontherocks, 2025-03-16:
> Hi, First of all thank for starting this project. Great help to get my HotTub online. Much appreciated!
>
> I noticed that boost ON is working, but it is not possible for me to turn it OFF again. [...]

Can confirm I have this same issue with boost being unable to turn it off.  Also I have the Spa Connection Kit (which is horrible by the way) and the values I change with the HA integration do not sync or show up in the SCK app on my iPhone (including set temp, salt output level and boost).

Ylianst, 2025-03-16:
I could be way off on this, but when you hit boost on the hot tub display, is that not just a timed thing? You turn it on and it goes back off after a few minutes? I don't think there is a way to turn it off.

Yes, the Spa Connection Kit is terrible. It's worst once you understanding how it's sort of hacked together. It makes underlying design of the ESP32+HA solution look crazy good by comparison.

Ylianst, 2025-03-16:
@fclauson, @Largosofthj - If you find exact values in the data you want me to convert into sensors in Home Assistant, please let me know. I see you have log files with RS485 data, that is exactly all the data you have to look at. If you make any sense of it, let me know.

homeontherocks, 2025-03-16:
Yes, hitting boost does up the Salt output to 10 for 24 hours in the system.  Was going to wait 24 to see if it returns to "off" automatically.  I do get values for IQ Unknown D.  It has gone successfully up from 675 to 765 over about 6 hours.  Will continue to watch but seems like a timer related function as so far it is only going up.

homeontherocks, 2025-03-17:
Can confirm that the Salt output does return to "off" as expected (24 hours of boost).  Ylianst, it might be better to replace the on/off switch with an activate button to prevent confusion and unneeded posts.  BTW my IQ Unknown D value continues to rise to 877 at noon today but has since decreased to 865.

fclauson, 2025-03-20:
Hi
how do I display these "ID Unknown" details - is there a YAML I need to show this

Ylianst, 2025-03-21:
I am not a hot tub export at all, very much far from it. I think it's a great theory and we can change it to another value easily.

homeontherocks, 2025-03-21:
Okay fantastic.  I will let you know when I get a change to confirm with further testing.  Is there any other data to pull out or is that it?  It seems like A and B are always zero so not sure if it's useful to have those in there.  Do you know if anyone has constructed a dashboard to mimic the screen of the Freshwater Salt system yet?

[image]

Ylianst, 2025-03-21:
@homeontherocks So, in an ideal world (and I do not have a Freshwater IQ so I can't do this, otherwise I would), you could setup the DataViewer and send out fake Freshwater IQ commands with various values and see how the hot tub remote display reacts to the values. We could also exactly mark that is the max, min, yellow, red area for each value. If this could be done, we would be able to exactly replicate the hot tub Freshwater IQ dashboard in home assistant and know for sure it's perfect. I am guessing that as soon as a correct command is sent, the UI will flip back, so you have to send the fakes at the right times.

Having the exact values for the 5 bars would be the ultimate.

homeontherocks, 2025-03-21:
Chlorine: 1-5 (3ppm)
pH: 7.2-7.8 (7.4)
Salt: 1500-2000 (1750ppm)

Pretty sure the above ranges are the green in the display.
For chlorine I'd make lower red 0-0.5 and lower yellow 0.5-1 and upper yellow 5-5.5 and upper red 5..5-6.0
For pH lower red 7.0-7.1 and lower yellow 7.1-7.2 and upper yellow 7.8-7.9 and upper red 7.9-8.0
For Salt lower red 1000-1250 and lower yellow 1250-1500 and upper yellow 2000-2250 and upper red 2250-2500
Anything out of the range should just stay in red.
We can deduce Cartridge days remaining based upon IQ hours left - can easily come up with the initial number with replacement of a new one.
I also think it would be useful to display current salt output level (ie 0-10) and maybe alkalinity?

The 2024 and later versions of Highlife Freshwater IQ Salt system also have an automatic dispenser system for some chemicals.  I don't know the details as mine is a 2023.

Interestingly, my chlorine continues to fall despite boosting each of the last 3 days.  So I'll be interested to see the control panel once I get up there as the pH is low and chlorine is very low.

homeontherocks, 2025-03-21:
Here is a quick and dirty dashboard that will work fine for me.

[image]

fclauson, 2025-03-22:
Here is some data from a freshwater system (not an IQ)

steps whre
1 - start monitoring 
2 - ramp level up from 4 to 9 - then to 0 
3 - select cartridge change
4 - change cartrigede 
5 - restart salt 
6 - set to level 6 

[salt log.txt](https://github.com/user-attachments/files/19403261/salt.log.txt)

the second image file is a pictire of "packet store" (what is this - is it the unique packets ??) 
ignore the blue 

[image]

homeontherocks, 2025-03-22:
Something strange is going on with my ESP32.  If you look at the sample dashboard I posted above, the values are abnormal.  Thinking my chlorine level was dangerously low, I ran boost 3 days in a row (I wasn't at the property).  My daughter went up last night and sent me a picture of the control panel and the chlorine was super high but all the other parameters were normal (ie completely different than this integration).  They used the hot tub but then the integration numbers changed to their correct values (see attached gauges).  I will watch this closely over the next few days but am worried about the accuracy of the sensor data over time.

[image]

Ylianst, 2025-03-23:
@homeontherocks - Interesting. Right now, the ESP32 is not polling any Freshwater IQ values, it's just listening on the RS485 bus for the IQ2020 to poll the values from the Freshwater IQ module. One theory could be that IQ2020 is only polling when the remote is on and so, if that's the case, no values would be obtained most of the time. If this is correct, the ESP32 is going to have to do some active polling.... this said, the Freshwater IQ module may have a limited capability to make measurements and polling it all the time could (???) cause it to need faster replacement (???).

Maybe the best would be to keep track of the last time the values where received.

homeontherocks, 2025-03-23:
Yeah the hot tub was used this morning and the values still seem accurate.  I've tried running a clean cycle to see if that updates the values but it does not seem to.  I think when they are read, the chlorine and pH values are accurate.  I am almost certain the Salt value is inaccurate (ESP32 reading way higher than control panel) and I still have not seen the Salt Level change from 4 over the last week.

I was told by the Hot Springs rep that they will be releasing a new version of the Spa Connection Kit that was developed in house in the next few months.  He promised to install for me since my SCA has never worked reliably.  We may be able to reverse engineer or intercept the data that way.  About 2 months ago the SCA had normal readings but when I checked the spa after being away for awhile it was at 65 degrees and in emergency mode.  So I don't think the polling of that system is good either.  Your point about the Freshwater IQ module making limited measurements (maybe a certain time of day?) would be useful to know.

Looking at historical values, in my system it's measuring every hour (about 2-3 minutes before the top of the hour for chlorine, pH, IQ D and Salt).  Salt Level is measuring once daily (random times).  IQ A and B are every few hours.

fclauson, 2025-03-24:
Did another Catridged changed (or rather went through the motions of doing one)

[salt log 20240324.txt](https://github.com/user-attachments/files/19428015/salt.log.20240324.txt)

homeontherocks, 2025-03-24:
> [@homeontherocks](https://github.com/homeontherocks) - Interesting. Right now, the ESP32 is not polling any Freshwater IQ values, it's just listening on the RS485 bus for the IQ2020 to poll the [...]

@ylinst Okay this is weird.  My high chlorine slowly decreased over the last few days and was 3.0 this AM.  I hit the boost button and now the chlorine level is at 1.0.  Is it possible that the data to the ESP32 is the inverse of the value?  Last week when I kept hitting the boost button the chlorine was still very low and I'm wondering now if the measurement is inverted.

Ylianst, 2025-03-26:
@homeontherocks - Sending fake values is not difficult (I guess it depends how technical you are). You need a Windows computer. [The instructions are here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/debugging.md). In your ESP32 configuration, add the line `port: 1234" like this if you do not have it already:

```
# Relay UART messages over TCP
iq2020:
   uart_id: SpaConnection
   port: 1234                               <---------- Add this line
```

Then, download the [IQ2020 Data Viewer Executable](https://github.com/Ylianst/ESP-IQ2020/raw/main/DataViewer/IQ2020-DataViewer.exe) and run it, figure out the IP address of the ESP32 and connect to the IP address port 1234.

[image]

Like magic, you will see everything going on in your Hot Tub in near real time. I hope your familiar with HEX values, if not, please let me know. The IQ2020 is '01' and Freshwater IQ is `37`. The first value is where the message is going and the next is where it's coming from. Just note the `01 37` messages, that is Freshwater to IQ2020. If you get that working, I can tell you exactly what to send. It's as easy and pasting it in the box and hitting "send".

homeontherocks, 2025-03-26:
I do not have a Windows computer but am able to install a Windows VM on my MacBook.  I assume that will work fine as it's just sending signals over TCP.  Not used to working with HEX so some pointers on that would be helpful.  I may go up this weekend and can experiment if you let me know exactly what to send and look for.  BTW, my chlorine levels have normalized now so the Boost caused them to go down and now back up so almost certainly that ESP32 value is inverted.

Ylianst, 2025-03-26:
@homeontherocks - Ok, yes, a Windows VM will work just fine. I will get sample data ready for you.

Ylianst, 2025-03-26:
@homeontherocks. I made it easy for you. In Data Viewer, you can select "FIQ Test 1" and "FIQ Test 2" in the action menu and put in fake value and hit send. You  should probably get a little familiar with the [protocol here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/protocol.md#freshwater-iq), lots to be discovered.

[image]

I would focus on "FIQ Test 1", I put in sample values that we have seen before. You can change the values, hit send and take lots of notes!

homeontherocks, 2025-03-27:
Okay I'm connected!  I'm getting lots of data but don't have the option of FIQ Test1 or FIQ Test2 in my downloaded IQ2020 Data Viewer.  Help!

Ylianst, 2025-03-27:
@homeontherocks - Your right, 2 minutes, I will have it updated.

homeontherocks, 2025-03-27:
@Ylianst Would this be easier if I shared my Tailscale exit node with you?  That is how I am accessing the ESP32 (from 100 miles away) over a Tailscale network.  That would make it about 100x faster for you to troubleshoot?

Ylianst, 2025-03-27:
@homeontherocks - I would be ok with it, but the point is to send the fake data and look at the result on the remote control. So, I am not sure I can help. If I am missing something, do feel free to mail me at my GitHub email.

Ylianst, 2025-03-27:
@homeontherocks - For FIQ1, the values I think I know are:

```
A = Unknown, always zero.
B = Unknown, always zero.
C = Unknown, 32 bit small-endian.
D = Unknown, 32 bit small-endian.
E = Chlorine in 10ths of PPM, 32 bit small-endian, Convert to decimal and divide by 10.
F = Ph in 10ths, 32 bit small-endian, Convert to decimal and divide by 10.
G = Hours remaining count-down, 32 bit small-endian.
```

You probably want to change C and D and see what that does. But also, find the low and high values for E and F. I would not change G and I can't imagine A and B do anything, but you can check.

Ylianst, 2025-03-27:
@homeontherocks - Also, in "File" select log to file and capture everything and send it over. After your experimenting, we may be able to look at the log and find new things. If you can, leave it capturing for a long time so we can capture all the internals and investigate. Thanks.

arthurfloh, 2025-03-27:
using IQ Freshwater Saltsystem - here my logfile perhaps it helps you.
component 237 + 238 needs to long time . What is that?

[logs_hotspring_logs.txt](https://github.com/user-attachments/files/19491386/logs_hotspring_logs.txt)

arthurfloh, 2025-03-28:
[homeontherocks](https://github.com/homeontherocks) it seems that you get the right values from the hot tube.
I have pH 8,6, CL 7,6, Alkality 727, Salt 2500.
Maybe that I have wrong settings in the config?

homeontherocks, 2025-03-29:
> [@homeontherocks](https://github.com/homeontherocks) it seems that you get the right values from your hot tube. I have pH 8,6, CL 7,6, Alkality 727, Salt 2500. Did you divide pH and CL by 10 and Salt by 2 or other config? Maybe that I have incomplete settings in my config?

Hi @arthurfloh no my values are not yet correct.  We are still trying to determine accurate values for the Freshwater Salt system.  I am going to try to make it up today to gather more data on my system.  Stay tuned.

arthurfloh, 2025-03-29:
@homeontherocks just for info:
when I enter the line for binary_sensor in config -> validate was OK -> that error message came after install - so I removed and wait! -->thanks

[image]

homeontherocks, 2025-03-29:
@arthurfloh You have some errors in your yaml file.  Look at the examples and use the exact syntax and spacing.

homeontherocks, 2025-03-30:
Placing this here for future reference and troubleshooting:

On the Freshwater IQ control panel, the green range for chlorine is stated as 1-5 and for pH 7.2-7.8.

I replaced my salt cartridge and sensor cartridge to make sure that all measurements are accurate (I needed to do it anyway).  It takes 24 hours to calibrate so I was unable to do any testing with the Data Viewer.  Of note, when you put in a new cartridge and sensor the IQ Hours Left defaults to 9999.  Also can confirm that changing Salt level in the ESP32 is reflected on the control panel.  If you hit Boost, the system is supposed to change Salt Level to 10 for 24 hours and then turn off.  This does not appear to be reflected in the Salt Level reported by the ESP32 (it just stays at the level set prior to hitting Boost).  It is unclear to me what "Salt Content" measures as it has not varied for me from 4 since installation.  In looking at History, the Salt Content went to Zero for 2 minutes after replacing the Sensor but then immediately went back to 4.

Chemical measurements via dipstick is as follows (obviously estimated based upon color coding):
Hardess 150, pH 7.8, Alkalinity 80, Chlorine 4, Bromine 5, Salt 1875

ESP32 measurements: Chlorine 1.5, pH 7.2, Salt 2132, "D" 813
Panel estimation: Chlorine 1.25, pH 7.25, Salt 1625

It seems that the Chlorine and pH data from the ESP32 are accurate.  It does not seem that the Salt is accurate (it might be "D" x 2?

gdbassett, 2025-04-05:
Hi y'all.  I did some of the original freshwaterIQ debugging, but don't have HA, only python code running on the connect spa kit.

One thing I've noticed is there is a direct relationship between channel D and what we assume is the chlorine channel.  when used, channel D will go down potentially 100 and then will slowly rise back up.  During this time, the chlorine channel tends to bottom out or almost bottom out.  Sometimes, D will drop quickly and chlorine will rise a bit, but then go back down.  Eventually channel D will level out around 750 and chlorine will start to go up very quickly. 

If y'all want the sqlite database I've been maintaining of the chlorine message logs, I'd be happy to share it.

gdbassett, 2025-04-05:
The reason I'm back is the webapp I'm running on the connected spa broke.

Up until March 25th it worked fine with commands looking like this (pardon the log details around it):
```
Date,Time,Level,Host Name,Category,Program,Messages
2025-03-15,09:13:22,Debug,192.168.200.245,user,hottub,"2025/03/15T14:12:15,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 77 00 00 03 08 00 00 00 09 00 00 00 4B 00 00 0D 0E 3C"
2025-03-15,08:15:38,Debug,192.168.200.245,user,hottub,"2025/03/15T13:12:39,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 71 00 00 03 06 00 00 00 0A 00 00 00 4B 00 00 0D 0F 42"
2025-03-15,07:12:40,Debug,192.168.200.245,user,hottub,"2025/03/15T12:12:30,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 6F 00 00 03 03 00 00 00 0B 00 00 00 4B 00 00 0D 10 45"
2025-03-15,06:15:51,Debug,192.168.200.245,user,hottub,"2025/03/15T11:12:15,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 70 00 00 02 FF 00 00 00 0A 00 00 00 4B 00 00 0D 11 49"
2025-03-15,05:13:33,Debug,192.168.200.245,user,hottub,"2025/03/15T10:12:30,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 71 00 00 02 FB 00 00 00 0B 00 00 00 4B 00 00 0D 12 4A"
2025-03-15,04:15:36,Debug,192.168.200.245,user,hottub,"2025/03/15T09:12:26,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 66 00 00 02 F9 00 00 00 0B 00 00 00 4B 00 00 0D 13 56"
2025-03-15,03:12:38,Debug,192.168.200.245,user,hottub,"2025/03/15T08:12:18,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 63 00 00 02 F9 00 00 00 0B 00 00 00 4B 00 00 0D 14 58"
2025-03-15,02:14:56,Debug,192.168.200.245,user,hottub,"2025/03/15T07:12:40,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 67 00 00 02 F8 00 00 00 0B 00 00 00 4B 00 00 0D 15 54"
2025-03-15,01:16:48,Debug,192.168.200.245,user,hottub,"2025/03/15T06:12:36,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 64 00 00 02 F8 00 00 00 0A 00 00 00 4B 00 00 0D 16 57"
2025-03-15,01:16:48,Debug,192.168.200.245,user,hottub,"2025/03/15T06:12:33,1C 37 01 03 40 23 D1 00 90 37"
2025-03-15,00:14:43,Debug,192.168.200.245,user,hottub,"2025/03/15T05:12:24,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 66 00 00 02 F7 00 00 00 0B 00 00 00 4B 00 00 0D 17 54"
2025-03-14,22:13:49,Debug,192.168.200.245,user,hottub,"2025/03/15T03:12:30,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 63 00 00 02 F5 00 00 00 0B 00 00 00 4B 00 00 0D 19 57"
2025-03-14,21:16:05,Debug,192.168.200.245,user,hottub,"2025/03/15T02:12:22,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 63 00 00 02 F3 00 00 00 0B 00 00 00 4B 00 00 0D 1A 58"
2025-03-14,20:14:05,Debug,192.168.200.245,user,hottub,"2025/03/15T01:12:43,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 62 00 00 02 F0 00 00 00 0C 00 00 00 4B 00 00 0D 1B 5A"
2025-03-14,19:16:54,Debug,192.168.200.245,user,hottub,"2025/03/15T00:12:31,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 5F 00 00 02 EE 00 00 00 0B 00 00 00 4B 00 00 0D 1C 5F"
2025-03-14,18:14:06,Debug,192.168.200.245,user,hottub,"2025/03/14T23:12:23,1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 08 5C 00 00 02 EC 00 00 00 0B 00 00 00 4B 00 00 0D 1D 63"
```

However, since then the messages have been scrambled:
```
Date,Time,Level,Host Name,Category,Program,Messages
2025-03-29,10:13:04,Debug,192.168.200.245,user,hottub,"2025/03/29T15:12:34,01 04 02 FF FF FF 00 FF 01 FF FF FF FF FF 69 1C 01 29 0F 80 1E 01 04 08 34 28 0B 03 00 00 CF 11 00 69 01 67 1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 07 E7 00 00 02 ED 00 00 00 1F 00"
2025-03-29,09:48:08,Debug,192.168.200.245,user,hottub,"2025/03/29T14:45:42,40 23 D1 91 1C 01 37 1E 80"
2025-03-29,09:13:18,Debug,192.168.200.245,user,hottub,"2025/03/29T14:13:02,03 40 19 01 00 6E 1C 38 01 02 40 24 01 5F 1C 38 01 03 40 24 05 0F 4B 1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 07 EB 00 00 02 F4 00 00 00 20 00 00 00 4D 00 00 0B BD 18 1C 29 01 0F 40"
2025-03-29,08:57:33,Debug,192.168.200.245,user,hottub,"2025/03/29T13:55:40,0F 4B 1C 37 01 02 40 23 D1 91 1C 01 37 1E 80 23 D1 00 00 00 00 00 00 00 00 00 00 07 ED 00 00 02 F4 00 00 00 21 00 00 00 4D 00 00 0B BE 14 1C 29 01 0F 40 1E 01 05 02 FF FF FF 00 FF 01 FF FF FF FF FF 68 1C"
2025-03-29,08:14:11,Debug,192.168.200.245,user,hottub,"2025/03/29T13:12:43,00 03 00 00 CF 11 00 69 41 31 1C 38 01 02 40 24 01 5F 1C 38 01 03 40 24 05 0F 4B 1C 38 01 06 40 24 04 00 80 00 D0 08 1C 38 01 02 40 24 01 5F 1C 38 01 03 40 24 05 0F 4B 1C 33 01 03 40 19 01 00 6E 1C 37 01 03 40 23 D1 00"
2025-03-29,07:16:07,Debug,192.168.200.245,user,hottub,"2025/03/29T12:12:39,A4 07 00 20 39 36 46 20 39 36 46 76 00 FF 00 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 4F 00 00 00 00 00 00 3C 00 1E 00 00 3A FE 00 36 24 06 08 03 00 08 01 CA 1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00 00 00 00"
2025-03-29,06:14:14,Debug,192.168.200.245,user,hottub,"2025/03/29T11:12:58,04 00 80 00 D0 08 1C 38 01 02 40 24 01 5F 1C 38 01 03 40 24 05 0F 4B 1C 29 01 0F 40 1E 01 01 02 FF FF FF 00 FF 01 FF FF FF FF FF 6C 1C 01 29 0F 80 1E 01 01 07 38 28 00 03 00 00 CF 11 00 69 41 32 1C 37 01 03 40 23 D1 00"
[... cut ...]
2025-03-28,12:12:56,Debug,192.168.200.245,user,hottub,"2025/03/28T17:12:47,FF FF FF 6C 1C 01 29 0F 80 1E 01 01 07 38 27 00 03 00 00 CE 11 00 69 41 34 1C 33 01 03 40 19 01 00 6E 1C 38 01 02 40 24 01 5F 1C 38 01 03 40 24 05 0F 4B 1C 37 01 03 40 23 D1 00 90 1C 01 37 1E 80 23 D1 00"
```

No clue why.  Rebooted the connected spa kit and the hot tub.  It's like the python log on the connected spa is no longer correctly capturing the messages.

(btw, these logs are off my log server.  it keeps completely logs, but for a much shorter time.  Happy to share those too.)

homeontherocks, 2025-04-05:
Hi @gdbassett I've been monitoring my Freshwater IQ system now for a couple of weeks.  Now that I've got a fresh salt and sensor (I think some of my previous issues were from having bad cartridges) the current chlorine and pH track well with the control panel.  The current salt, however, is not correlating with control panel salt.  It appears that D is tracking with Salt better but it would be 2x D = Salt.  But it is possible that one of the values correlates with something else like alkalinity.  Unfortunately, I will not be able to really determine this until I am back up at the hot tub to send false signals via the Data Viewer hopefully next weekend.

gdbassett, 2025-04-05:
@homeontherocks My salt signal tracks very well with my salt.  It goes up slowly over time as water evaporates and then returns quickly to the expected level when adding water.  D tracks very well with my usage (going down when in use, then slowly building back up).   I suspect there are some fluctuations in D and chlorine due to the circulation pump coming on.  They are also very sensitive to the water temperature.

As I think about it though, I wonder if salt is alkalinity.  Salt actually should come off the salt system as it uses the electrode to measure.  However doesn't salt usage effect Ph/alkalinity?  And I fill through a charcoal filter so the water I put in should be low alkalinity (hence the drop when adding water).

github won't let me drop a sqlite database here, but let me know if you want me to send it somewhere to look at.  It's useful to graph the channels and then look at how they change in comparison to each other.

Ylianst, 2025-04-05:
@gdbassett Is it correct your using the Spa Connection Kit to capture these logs (not the ESP32)? It looks like frame start/end is not longer being detected or something is transmitting constantly. I still see the data frames in the scrambled data, but yes, it's all bunched up together. I have not seen the ESP32 do this except when multiple modules are transmitting at the same time, in that case, you see two frames sort of merged together like you see. Generally when this happens, the CRC fails and you move on.

In the case of the SPA connection kit, all frames on the RS485 bus are sent to the base station (which is a bit wasteful). So, there are two opportunities to mess things up. Anyway, I don't have a solid recommendation (Flash a ESP32 and use DataViewer?)

One thing that would be interesting someday is to connect a sufficiently capable device to the hot tub that is can host it's own web site and not need Home Assistant. I think the ESP32 can't quite make it, it would need more RAM.

gdbassett, 2025-04-05:
@Ylianst honest, if it weren't so expensive, the spa connection kit could do the same thing.  I've just never dug into the library it uses (https://github.com/digidotcom/xbgw-app) to figure out how to configure and control it more directly.  I suppose my best bet is to shut down the whole thing for a while and bring it back up later.  I could try and extract the D1 packet out of string, but I think it's getting cut off as well.

Ylianst, 2025-04-05:
@gdbassett You are absolutely right, they connection kit is very capable. My main issue with it (and this is just theory, in practice it may not matter at all) is that the hot tub side is so dumb. It just relays everything and the smarts is in the base. If there was a way to connect a RPi Pico easily, I think it would be an interesting way to go. This said, the M5Stack's package is so perfect (4 pins, exactly what we need), maybe they will come out with a more capable ESP32 in the same package at some point.

gdbassett, 2025-04-05:
Would a device sitting directly on the bus be different than sitting on the other end of the wireless bridge?  Or is it a matter of getting lower in the connection kit so we can send/receive our own messages?

Ylianst, 2025-04-05:
It's just that having a bit more smarts closer to the bus makes things more efficient.  For example, if you want to turn on Jet1, the ESP32 can send the command, see on the next status poll if the jets are on and if not, try again. The RS485 bus is not crazy reliable so, having the check and retry close to the bus is nice. You can still do it form the base, it's more wireless round trips. Again, in practice I don't think anyone would notice.

homeontherocks, 2025-04-06:
I have the Spa Connection Kit, and it is a worthless POS.  I've turned it off as @Ylianst  ESP32 solution is far superior, even now that the readings are incomplete.  I have been told that a new "in house" app is under development with plans for release in Q3, which I only care about if it helps us extract more data via this approach.  Next weekend I will try to figure out C and D and report back.

Ylianst, 2025-04-06:
@homeontherocks - To be fair, @gdbassett hacked the Spa Connection Kit to pull raw data from it. However, it's an expensive way to do this compared to ESP32. As far and I am concerned, no "In House" app will ever match a Home Assistant integration. I really don't want another app, this said, their app is currently so bad, anything would be better.

gdbassett, 2025-04-06:
Also, there's no changes to the hot tub which hopefully helps ensure my warranty :-).

Ylianst, 2025-04-06:
By the way, in the US at least, the warranty indicates that if you see your house, the warranty is void for new owners. This is really messed up in my view. In any case, I have not generally been too worries about it in my case. This said, I want to be understanding to people that buy 500$+ equipment :) Having used both, the ESP32 does a better technical job in my view. I would not be surprised if the manufacturer ends up selling a 500$ ESP32 at some point.

homeontherocks, 2025-04-09:
> I have reduced the config only to IQ-watervalues and light. The value shown on the tub-display do not correspond with the data in HA. Chlor on tube-display low red down (more or less nothing) and in HA 2.2. pH can be correct. Measured ~8, on tub- display 7,4, in HA 8,5 Can I hope, if I change the water to get the right values? Maybe the iq sensor is the problem ?

Yes, my measurements made much more sense after I replaced the sensor AND the salt cartridges.  My sensor was 10 months old (supposed to replace every year) and the salt cartridge was older than the 4 month predicted lifespan.

arthurfloh, 2025-04-19:
@Ylianst Hot Tub is running very well so far! Another question.
I want to connect a BME680 via I²C directly with the grove port (G26/G32) on M5stack Atom lite.
Please help me what I have to write in the yaml-file to connect via this port. thanks

Ylianst, 2025-04-19:
@arthurfloh - You can get the BME680 on a board with the grove connector already, [like this](https://wiki.seeedstudio.com/Grove-Temperature_Humidity_Pressure_Gas_Sensor_BME680/). Then just connect it using standard connector. I would have to go look up standard YAML configs for it when I get a chance.

arthurfloh, 2025-04-19:
I have this one and prepare all - thanks 
[

[image]

](url)

Ylianst, 2025-04-20:
@arthurfloh Yes, that one will work. You can get yourself a grove connector like the one below for easy wiring. My big question is WHY? You would see the air quality, temperature and moisture level within the IQ2020 controller box? I will note that for many, there is already a temperature sensor on the IQ2020 board that you can read. Why the extra one?

I also used to sort of hoard data from my smart home, but now, I do somewhat the opposite and try to avoid having HA track data I don't need and optimize it a bit more. All this said, if you have a novel used, I want to hear it. Should I be doing the same?

[image]

arthurfloh, 2025-04-20:
That is a different project and has nothing to do with IQ2020.  I got this BME 680 and try to connect with a separate Atom Lite with the white 4-pin plug on it. 

+5, minus are clear pin G26 ->SDA, Pin G32 ->SCL
what i dont know is how to cofigure that port in  yaml-config file 
and BME680 libary 

is that correct and enough?
i2c:
  - id: grove
    sda: 26  
    scl: 32
    scan: true

with this sensor i want to analyse  the influence of indoor air quality. 
The IQ2020 is running very well so I need a new challenge with Atom Lite! Thanks for support!

Ylianst, 2025-04-20:
@arthurfloh - Oh, I see! Thanks for the details. I have never used the BME680 personally, but I have used many I2C modules and it should not be much of an issue getting it to work.

arthurfloh, 2025-04-20:
Do you think that the above I²C config is correct for the grove connector?

Ylianst, 2025-04-24:
Nice. Congrats @arthurfloh. Good work. I am going to close this thread since it's been a while. But please re-open or create a new thread for updated Freshwater IQ data. I am very interested to see is someone would feed fake data and collect the local display's reaction.

closed 2025-04-24 by Ylianst

arthurfloh, 2025-06-23:
@Ylianst : today I update the HA 2025.6.0 - following message came:

WARNING Using `select.SELECT_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `select.select_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: iq2020_select
WARNING Using `select.SELECT_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `select.select_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. 

Do I have to change something? Thanks

[2 more replies]