Ylianst/ESP-IQ2020 #40: Extending the range of the ATOM

fclauson, 2025-03-03:
Hi

I have read this - https://stevep.xyz/2021/03/13/increase-atom-lite-range-with-external-antenna/

My tub is "just" in range of the house but drops out once in a while 
I have a couple of options 
1 - put an external Wifi AP on the house
2 - put power line internet - one inside the tub (carfully !!!!!) and - one in the house 
3 - hack the atom as per the above post 

what approches have people taken

mediamanrit, 2025-03-03:
I had an AP already outside the house.  If I had to do it again, I’d do the
same thing.  Simple and then gives you coverage for other stuff as well.

_-=-_-=-_-=-_-=-_-=-_-=-_-=-_-=-_-=-_-=-_-=-_
Brian McCann
***@***.***

Ylianst, 2025-03-03:
Lots of factors. The ATOM is 2.4Ghz WIFI only and sometimes that band is used quite a bit. Moving as much traffic to 5Ghz can help. Also, I have been into a house when, each time they ran the microwave, all 2.4G WIFI would drop. My first option would be to put a WIFI relay (wired or not) closer to the hot tub.

Ylianst, 2025-03-06:
I am going to close this, but I do want to put the picture of the article linked above. Adding an external antenna to a M5Stack is a bit crazy.

[image]

closed 2025-03-06 by Ylianst