Ylianst/ESP-IQ2020 #41: Climate error when adjusting to 80F, value not in range.

Scott8586, 2025-04-18:
First off, new install - thanks so much for this work!

spa: 2023 hot springs pulse

I am working in a Fahrenheit environment.  When we go on extended travel I turn down the temp to 80F.  When I tried to do this through the climate control in home assistant, it gave the following error:  "Provided temperature 26.666666666666664 is not valid. Accepted range is 26.66667 to 40.0."

Relevant part of HA configuration (2025.4.1):

```
unit_system: us_cusomary
```

Relevant parts of the esp configuration (2025.3.3):

```
sensor:
  - platform: iq2020
    current_f_temperature:
      name: Current Temperature
    target_f_temperature:
      name: Target Temperature
    outlet_f_temperature:
      name: Heater Outlet

climate:
  - platform: iq2020
    name: Temperature
    celsius: false
```

Ylianst, 2025-04-18:
Interesting. Good catch. I see the same problem.

Ylianst, 2025-04-18:
@Scott8586 - Fixed it. You will need to re-flash the ESP32, you can do that over wireless so, no need to remove anything from the hot tub. Let me know if it works.

Ylianst, 2025-04-24:
Ok, this is fixed. Closing. Reopen if needed.

closed 2025-04-24 by Ylianst