Ylianst/ESP-IQ2020 #42: General Discussion

Ylianst, 2025-04-24:
Please post there anything you like. Issues, comments, feedback. Previous discussion thread was #13.

pdrakeweb, 2025-05-05:
I'm not sure about earlier versions, but the 2024 Limelight series offers 2 "eco" timers which run for up to 4 hours each.  I think this behaves similarly to the Summer Timer in that it shuts off the circulator and heaters and prevents the clean cycle from running.  Is there a way to expand this integration to cover those timers?  I checked and the summer timer switch does not appear to reflect the status of the eco timers.

fclauson, 2025-05-05:
In regard to eco timers 

I have the approach of using the tub as a thermal battery 
39.5c overnight on cheap electric
The drop set point to 37 day time 

Is working well

Ylianst, 2025-05-06:
@fclauson I don't have ECO timers, but I use Home Assistant to vary the set temperature depending on electric prices and it's been wonderful. Yes, hot tub as a battery, totally. [Page on this here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/electric.md).

sbenwell55, 2025-05-09:
Brilliant integration, well done. It worked easily for me. I had a small issue that I used Atom S3 and that has G1/G2 pins.
I seem to have it fully working, very cool.
My project is that same as you, I will control the temperature to ensure heating occurs during off-peak hours.

A question, perhaps I am too inexperienced with HA though, when I try to add an Automation, I do not get Temperature as one of the Actions I can perform on the Device. The closest is to change the HVAC mode. Where might I have gone wrong ?

[image]

[image]

Ylianst, 2025-05-10:
@sbenwell55 - Ha yes, I was going to post exactly this. For anyone else, this is what is looks like in the user interface.

[image]

Good for you. I save quite a bit of money with time of use. I also have Home Assistant look at the maximum temperature of the day at 6am and pre-heat the hot tub before mid-peak if the day's temperature is low. This said, plenty to experiment with.

derhally, 2025-05-16:
I keep meaning to ask, what does the "Controller Energy" metric represent?  It consumes about 1.44kwh every day.

Ylianst, 2025-05-16:
@derhally I am not exactly sure. I saw that number and put "Controller Energy" on it thinking it was the power used by the IQ2020 itself, but that seems high. There is a re-circulation pump that runs all the time (except stopped with summer timer) and I imagine it's the power used by the pump. If there is confirmation of this, I should rename it.

derhally, 2025-05-17:
> @derhally I am not exactly sure. I saw that number and put "Controller Energy" on it thinking it was the power used by the IQ2020 itself, but that seems high. There is a re-circulation pump that [...]

I think it must be the circulation pump.  I will turn on the summer timer, but according to the docs it says that will turn off both heat and the circulation pump.

fclauson, 2025-05-17:
Would it also be the salt system pump ?

Maybe it's any electrics used when the tub just sits there

nedguy, 2025-05-21:
Thanks for all the work on this, great integration. 

I have a 2014 Hotspring Aria and I’m having an issue with polling frequency. I have the polling interval set to 65 per the basic config, but it appears to be polling every 5 seconds. I’ve tried to change it and while the setting is picked up (I can see it in the log) it doesn’t have an actual impact on the frequency it polls.  During the poll I’m also getting a warning about the component taking too long. I tried legacy polling and it didn’t affect the outcome. Any ideas?

[logs_hot-tub_run.txt](https://github.com/user-attachments/files/20368533/logs_hot-tub_run.txt)

[logs_hot-tub_run.txt](https://github.com/user-attachments/files/20368598/logs_hot-tub_run.txt)

Ylianst, 2025-05-22:
@nedguy - Interesting. Can you send me your version string? Mine looks like this...

[image]

I went back and looked at the code and if the version string is empty, it will keep polling at 5 seconds interval. In your case, it never seems to poll for the version string? So, I am guessing you have that value empty?

nedguy, 2025-05-22:
[image]

That matches the version that is written on my controller 1.06 H03. So it does appear to be polling it and retrieving it. Maybe it’s the weird character at the end?

Ylianst, 2025-05-27:
Hi @arthurfloh and thanks. Yes, this is a nice project. If I recall, homeassistant_time will just have the ESP32 sync with Home Assistant so you can do energy tracking. This does not have anything to do with the time of the hot tub itself.

Starting in 2015/16 it looks like hot tubs do have a real-time clock. It's possible to read it and set it to the correct time value, when you get your hot tub it does not seem to be set to the correct time at all. You can read and sync using the DataViewer debug tool, however there does not seem to be any use at all for the real time clock and so, I did not add any support for it in the Home Assistant integration. If in the future they decide to show the current time on the remote or have a use for the clock, I will certainly add support for it as we know exactly how to read/set it using RS485 commands.

Let me know if that helps.

nedguy, 2025-05-28:
Hi @Ylianst, I think I might know what is going on but I don't fully understand the code around the audio module and how a tub without an audio module is supposed to respond, I wanted to see what you thought and if you could confirm what I'm thinking.

I believe it is polling for the version string and retrieving it, I don't know how to capture the full log, so on startup I think I miss the first few polling cycles before the remote logger is connected. Assuming that the version string is in-fact set prior to my log viewer connecting I looked to see what else might affect it. 

Using the polling interval appears to be dependent on both the version string being populated and the select audio source being set to something that is not equal to NOT_SET when USE_SELECT is defined.

`#ifdef USE_SELECT
			if (!versionstr.empty() && (select_state[SELECT_AUDIO_SOURCE] != NOT_SET)) { next_poll = ::millis() + (this->polling_rate_ * 1000); } // Next poll
#else
			if (!versionstr.empty()) { next_poll = ::millis() + (this->polling_rate_ * 1000); } // Next poll
#endif`

Looking at my log file and the code it appears to poll for audio 3 times, I see the Poll Audio message in the log file.  I'm not sure how/where the USE_SELECT compiler directive is set, but since I see the log message I'm assuming it is set.  Since I see the Poll Audio log message multiple times (stops after the third attempt), I'm also assuming that the select_state[SELECT_AUDIO_SOURCE] state remains set to NOT_SET after each poll.  I don't think the tub responds to this command at all as there is a delay after each audio status command is sent. I added the audio source state to my UI and did confirm that nothing is selected.

`#ifdef USE_SELECT
	// If we don't have the audio status, fetch it now.
	if ((got_audio_data < 3) && (select_state[SELECT_AUDIO_SOURCE] == NOT_SET || switch_state[SWITCH_AUDIO_POWER] == NOT_SET)) {
		ESP_LOGD(TAG, "Poll Audio");
		got_audio_data++;
		unsigned char cmd[] = { 0x19, 0x01 };
		sendIQ2020Command(0x01, 0x1F, 0x40, cmd, sizeof(cmd)); // Get audio status
		return;
	}
#endif`

This is where I'm unclear on why others are not seeing this.  Looking at the code if the select_state[SELECT_AUDIO_SOURCE] remains as NOT_SET then the polling rate will never be added to the next poll and will remain at the default 5 seconds.  This is where I feel like I might be missing something in my interpretation, as from what I see the polling rate would only be used if the tub responds to the audio status command and sets the audio source to something other than NOT_SET.  I'm not sure how the USE_SELECT is controlled, maybe this is not set for others and I'm doing something wrong?  If the USE_SELECT is not set then it wouldn't poll for audio and the next poll would only be based on the version string existing.  I appreciate any help you can provide.  Thanks!

Ylianst, 2025-05-28:
@nedguy - Thank you so much for that and I think you are on to something. I would not be surprise at all if others are having this fast polling and not reporting it. I may not be seeing it in my case because I turned on audio module emulation so I could use the audio controls to perform home automation tasks (control lights) on my patio. Your explication is exactly correct and I need to fix this.

For USE_SELECT, if you have `select:` in your configuration yaml, it's going to be set to true. It's that simple. So in my case I have a big `select:` section that starts with:

```
select:
  - platform: iq2020-dev
    name: Audio Source
    id: audio_source
    datapoint: 0
  - platform: iq2020-dev
    name: Color Underwater
    id: lights1_color
...
```

So, it will be enabled for me. So, if you have `select:` and no not have the audio module and do not have `audio_emulation: true`, you will have rapid polling. I will get myself setup in the correct config in the coming days and fix this. Indeed, if the audio module fails to respond 3 times, it should set the audio source to something other than NOT_SET.

Let me know if this makes sense, yes, this is a bug for sure.

arthurfloh, 2025-05-28:
yes, it is exactly as shown above. no date only time. A daily re-sync would be the solution.

nedguy, 2025-05-29:
@Ylianst - Yes that makes sense, I did try enabling audio emulation and it solved the polling frequency issue, so the bug is related to not having an audio controller and not enabling audio emulation.

nedguy, 2025-05-29:
Does anyone have any hints on getting the lighting controls to work?  My tub is a 2014 Highlife Aria, it has a single intensity setting for the underwater lights and a color setting for underwater lights.  It then has an "Additional Lighting Menu" where you can set the color of all the additional lighting, separate from the underwater lights.  You are not able to control the intensity of the additional lights, but can turn the bartop/pillow/water feature lights on and off independently.  I tried to add all the lighting selectors, numbers, and sensors, just to see what's returned, but I don't see any of them being updated.  They just stay in the unknown state. Its acting like the light poll command is not returning any information.  Any one have a similar model year tub that has seen this and gotten it to work?

Ylianst, 2025-05-29:
@nedguy - Ok, that confirms it. Thank you so much for all your data and participation. It's very much appreciated. I will look at fixing the polling this week, it should not be too difficult for me to get into the fast polling state. As for the lights, I am going to take a guess (huge guess) that being a 2014 model, you lights may be controlled by a separate modules that is connected to the RS485 bus and so, your lights are not controlled by the IQ2020 board itself. If this is true, you should see commands from the IQ2020 (0x01) to the external lights module (0x20).

Now, I know of the existence of this external lights module, but I don't have one and someone would need to decode the commands to/from that module. You would need to run the DataViewer on a Windows machine, connect it to your ESP32 over WIFI (just need to know the ESP32 IP address) and you could verify this theory.

Ylianst, 2025-05-29:
@arthurfloh - Ha yes, got it. Yes, indeed, having a way for HomeAssistant to set the time on the tub would solve this. If I can create a method to set the date/time you would them create a automation to set it once a day. Before I do any work on this, would it be possible for you to run DavaViewer, connect to your hot tub over WIFI and hit "Set Current Time" to check that it does update the time on the tub?

[image]

I imagine you want to set a wrong time then go on DataViewer and sync it to see if it works. If it does, I will add it in HA.

arthurfloh, 2025-05-29:
@Ylianst  -today early morning I did it as per your instruction and it works - excellent! That is what I´m looking for. 
A daily sync seems to be enough!

nedguy, 2025-05-29:
@Ylianst I spent some time looking at the lights today and there doesn't appear to be an external module.  The 4 lights are plugged directly into the controller.  The controller is this model: [https://shop.leisuretimeinc.com/products/hot-spring-control-box-iq2020-76836](url) which has 4 headers along the top for the 4 light zones and it's still used in some pretty new tubs.  I compared some of the operating manuals for the various models and the primary difference seems to be the controller software and how it allows the lights be controlled.  On mine I don't have as much control over the "additional lighting" (such as intensity or independent colors) as the newer software.  It think that also translates into the protocol being different. I hooked up the data viewer and when I query for the light status, I do get a response back, a single byte that appears to be either 00 if the lights are on and 01 if the lights are off.  
`--> 01 1F 40 1705
<-- 1F 01 80 170500`
I also noticed that the light status is not reflected in the general polling message, its always 00.
I ordered some updated software, as mine hasn't been touched since the tub was new in 2014, so I'll see if that updates the protocol.  If not I'll play with it some more and see if I can figure out what, if any, commands work to control lights 2-4.  Thanks for all your help and I'll update with what I find.

Ylianst, 2025-05-31:
@nedguy I put in a fix for the fast polling issue. I was not able to create a setup the would cause this issue, but I am pretty confident the fix should work. Please let me know if you can test and confirm it's ok. Thanks!

Ylianst, 2025-05-31:
@nedguy Ha yes, updating your firmware may do the trick. It's clear your not getting the light status thru IQ2020 commands. Please let me know what firmware version string you have before and after the update, so we can take note of this for others.

Ylianst, 2025-06-01:
@arthurfloh - Ok, I got a prototype implemented for settings the clock. In your configuration, replace all `platform: iq2020` with `platform: iq2020-dev`. Also, replace the section title 'iq2020:' with 'iq2020-dev:'. This way, your using my early development version. Next, change the `api:` section to look like this:

```
api:
  encryption:
    key: "xxxxxxxxxx"
  services:
    - service: set_hot_tub_time
      variables:
        hour: int
        minute: int
        second: int
        year: int
        month: int
        day: int
      then:
        - lambda: |-
            if (g_iq2020_main != nullptr) {
              g_iq2020_main->setTime(hour, minute, second, year, month, day);
            }
```

You probably have the `encryption` with a `key` already, all the `services` sub-section like shown. Compile and reflash.

Now, you can call the `set_hot_tub_time` service in a Home Automation.

[image]

You click "Add Action..." in the Automation and search for "hot", you should see it right away. Now, you can set the date and time, obviously the hot tub will only really use the time and you can but nothing interesting for the date. One each things to do is to set the time at 3:00am each morning to 3am, this way you also past daylight savings.

If you enable debug logs, it should show "setTime, time(h:m:s) = %d:%d:%d, date(y:m:d) = %d:%d:%d" in the log. As with any command, there is a tiny chance the command will be sent while another module is also sending some other command, but setting once a day should be fine.

Let me know if this works, I will move this to the production version.

arthurfloh, 2025-06-01:
error: Platform not found: 'fan.iq2020-dev'
fan:
  - platform: iq2020-dev
    name: Jets 1
    id: jets1
    icon: "mdi:turbine"
    datapoint: 0
    speeds: 1
  
what can I do?

Ylianst, 2025-06-01:
@arthurfloh - This is something unrelated. If you update to the latest ESP-Home, it will fix this. New ESP-Home versions started indicating some schema deprecation and the dev version has the fix for this. Let me move the setTime to the production version since it seems to be working pretty well.

arthurfloh, 2025-06-01:
@Ylianst - yes it´s working as per your very clear instructions. 
thanks a lot for your time.

nedguy, 2025-06-01:
@Ylianst - thanks for all your help.  I updated to the latest code and the polling issue is solved. Thank you!

Ylianst, 2025-06-02:
@arthurfloh - Oh? That is weird. Normally adding `api/services` section should make the "set_hot_tub_time" show up. I create a new automation, add action and type "hot" and see it.

[image]

If you can't see it, can you send over your configuration? Make sure to put XXXX on any encryption keys or anything private. Thanks.

arthurfloh, 2025-06-02:
@Ylianst - Now I see it and configured as shown above. Now I check this function! thanks

cmamatey, 2025-06-03:
@Ylianst just want to say thank you for this awesome repo! We moved into a house at the beginning of the year, and the purchase included a 2021 Hot Springs Sovereign. I unsuccessfully searched a few times earlier in the year for an ESPHome integration and thankfully came across yours last month. I have no clue how I hadn't found it before, but after reviewing your docs, I wasted no time ordering the parts. Upon arrival, I was excited to get an initial setup done in under an hour this past weekend.

I was tinkering with the audio emulation implementation and had one question: **can you clarify what the function of the Audio Power switch is intended to be? **
Turning it on and off in Home Assistant does not appear to have any impact on the controller's audio tab. I assumed it would follow the state of the controller's on screen power button, but turning that button on and off just seems to send play and pause signals.
Is this normal behavior, or do I need to do some data capture to figure out what's going on?

Hot Springs Sovereign 2021
Firmware WR4.04de1cE002DK4.00
Jet 1 setup with 2 speeds
Jet 2 not enabled
Salt system enabled
Audio system emulated

Ylianst, 2025-06-05:
Hi @cmamatey. Thank you so much for your super well written question. Back story, version string, details and all. Thank you so much. So, the Audio Power switch does not do much. On the remote, you should see that the state will be off instead of showing the audio controls. Also, when you turn the audio on/off, play/stop are pressed. I personally use the audio emulation to control lights on my patio and on/off -> play/stop behavior is rather annoying. If you attach light toggle behavior to play/stop buttons, the power on/off will cause them to trigger. There is no way I know of to remove this.

Let me know if this helps. I will add your hot tub to the list of tested ones.

jbunting, 2025-06-06:
Has anyone tried disconnecting the LEDs from the iq2020 board and connecting them directly to the esp32 (and then controlling them with one of the RGB components)?  Looking at the connections I suspect this is possible but I haven't looked at the voltages or signals yet. I was hoping someone had dug into that.

Ylianst, 2025-06-07:
@jbunting - Interesting idea. Not something I tried, but it's certainly technically possible.

arthurfloh, 2025-07-11:
Today I updated the HA-core not the esphome-builder . Minutes later many commas at some entities. especially at chlorine.
The window "Anzeigegenauigkeit" is missing.

[image]

please help!

jay7210, 2025-07-13:
> [@jay7210](https://github.com/jay7210) - You can look [#22](https://github.com/Ylianst/ESP-IQ2020/issues/22) for a previous discussion on the expansion board. Search "Watkins 74150" online. I got mine off eBay a long time ago when you could get them for cheap. It's a passive splitter of all the pins, so, some people have built Y connectors and that works.

Thanks,

I have been probing around my control box and have come to the same conclusion regarding a Y lead. Thanks for the confirmation.

Ylianst, 2025-07-13:
@jay7210 - You can look at the [connectors here](https://github.com/Ylianst/ESP-IQ2020/issues/13#issuecomment-2629576304) if that helps.

arthurfloh, 2025-07-13:
@Ylianst Please can you tell me what´s wrong in my yaml.
Temperature controller 47 seens to be correct but in °C not Fahrenheit.
All temperature correct with °C only this record. - thanks
[

[image]

](url)

Ylianst, 2025-07-14:
Hi @arthurfloh. You will need to post your yaml so I can see what is going on. I will point out that from the default YAML configuration, if you want Celsius, you have to replace `_f_` with `_c_` at 3 places and put `celsius: true`. Your hot tub also has to be set to show celsius.

```
esphome:
  name: hot-tub
  friendly_name: Hot Tub
  comment: "Luxury Spa"

esp32:
  board: m5stack-atom

logger:
  baud_rate: 0
  level: ERROR

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source: github://ylianst/esp-iq2020

# Make sure tx/rx pins are correct for your device.
# GPIO26/32 is ok for M5Stack-ATOM + Tail485, look in GitHub devices link for your device.
uart:
  id: SpaConnection
  tx_pin: GPIO26
  rx_pin: GPIO32
  baud_rate: 38400

iq2020:
   uart_id: SpaConnection
   polling_rate: 65
   port: 1234

# These empty sections are required to compile correctly.
select:
text:
number:

# If using Celsius units on the hot tub remote, replace _f_ with _c_ in the three entries below.
# Feel free to remove any sensor that are not relevant for your hot tub.
sensor:
  - platform: iq2020
    current_f_temperature:
      name: Current Temperature
    target_f_temperature:
      name: Target Temperature
    outlet_f_temperature:
      name: Heater Outlet
    power_l1:
      name: Pumps Power
    power_heater:
      name: Power Heater
    power_l2:
      name: Heater Power
    pcb_f_temperature:
      name: Controller Temperature

switch:
  - platform: iq2020
    name: Lights
    id: lights_switch
    icon: "mdi:lightbulb"
    datapoint: 0
  - platform: iq2020
    name: Spa Lock
    id: spa_lock_switch
    icon: "mdi:lock"
    datapoint: 1
  - platform: iq2020
    name: Temperature Lock
    id: temp_lock_switch
    icon: "mdi:lock"
    datapoint: 2
  - platform: iq2020
    name: Clean Cycle
    id: clean_cycle_switch
    icon: "mdi:vacuum"
    datapoint: 3
  - platform: iq2020
    name: Summer Timer
    id: summer_timer_switch
    icon: "mdi:sun-clock"
    datapoint: 4

fan:
  - platform: iq2020
    name: Jets 1
    id: jets1
    icon: "mdi:turbine"
    datapoint: 0
    speeds: 1
  - platform: iq2020
    name: Jets 2
    id: jets2
    icon: "mdi:turbine"
    datapoint: 1
    speeds: 2

# Set "celsius" to "true" if using celsius units.
climate:
  - platform: iq2020
    name: Temperature
    celsius: false

text_sensor:
  - platform: iq2020
    versionstr:
      name: Version
```

arthurfloh, 2025-07-14:
I have this versions

[image]

arthurfloh, 2025-07-15:
[image]
since uodate 2025.7 only the IQ values have the decimal problem. Are I´m the only one with this problem?
Any idea to solve that?

Ylianst, 2025-07-16:
Something has certainly changed in the recent Home Assistant versions, I am having to deal with more decimals. I need to add rounding for sure.

arthurfloh, 2025-07-16:
@Ylianst Do you have any idea why the marked label is not existing at the IQ-entities
Rounding could be very easy but not possible.

[image]

jay7210, 2025-07-16:
I have a Question regarding the Logo Lights. On my Hotspring Limelight Flair i have 3 Lights, a Blue, a Green and a Yellow. How does that transpose from the two Lights on on the HighLife spas?

Ylianst, 2025-07-16:
Hi @jay7210 - I don't have an answer, but we count on community members to gather such data. You probably [saw here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/extras.md#logo-lights), there seems to be 8 status codes for the logo lights. I only have blue/green logo lights on mine, but I am guessing they may use the 3 lights to express the 8 states differently.

If you see your lights in a given state and can take note of what lights are on/off/blinking in each state, please gather that data and send it over.

Ylianst, 2025-07-16:
@arthurfloh - I have no idea on decimal setting. If anyone else worked on this, please let us know. At some point, I may look into it.

jay7210, 2025-07-17:
@Ylianst I am seeing a 0 for Yellow and Flashing yellow. How can i debug more?

Ylianst, 2025-07-19:
@jay7210 - Ok, if the existing sensor does not give the right data, we need figure out that part of the status command is giving the right data. [The debugging guide is here](https://github.com/Ylianst/ESP-IQ2020/blob/main/documentation/debugging.md). If you don't have it already, add `port: 1234` in your configuration.yaml and update the firmware.

```
iq2020:
   uart_id: SpaConnection
   port: 1234
```

Then, download the [IQ2020 Data Viewer](https://github.com/Ylianst/ESP-IQ2020/raw/main/DataViewer/IQ2020-DataViewer.exe) and run it on a Windows machine and connect to your HotTub. You will see all of the traffic on the RS485 bus of the HotTub in real time.

Go int he "Filter" menu and change to "Connection Kit 0x0256" and in the commands menu, you can click "Ask Status 0x0256".

[image]

The goal is to record this command while your lights are in different modes. For example, my hot tub status right now is:

```
 <-- 1F 01 80 02560008400400000606000A0622F21100201C201C201C8403605400000000313032462C3A97009D300E003C8BDB0751000000000000003B1112000000000000000000D215160002007FE2DA0700000000D03A0F0031303146313031467B00FC00FC000000000000000000000000004F0000000000013C001E00006DA0001A110F1306E90701
```

If we get many of these commands in various states, we can see where the data is located and I will add a new ESP-IQ2020 sensor with this data. Hope that makes sense.

Ylianst, 2025-07-20:
@jay7210 - Wow, you are so fast. Looking at this now.

Ylianst, 2025-07-20:
@jay7210 - Lots of good data here. For the logo light levels, I could not find anything different between Auto/3/2/1 status reports. I lined to the 4 reports and marked any changes between them and there is nothing unexpected. So, it's not encoded in this command. In DataViewer, click "Ask Light Status", you get a response like this:

```
 <-- 1F 01 80 17050000000000010101010202020208080808000000
```

It's a smaller response, you should see quickly any changes. You could also monitor the RS485 bus while you are making the changes to see if there is any commands send. Generally that is not the case since the report is on a different "bus" than the modules.

Here is my decoding of the "Logo Light Level 1" command with any changes from the other light levels annotated. All the values that changes are known.

Logo Light Level 1
```
0256
0008
40
04
0000
06
0600
0A
0802F1
52
06
201C201C201C
8403
6054
00000000
33382E30
4DC5EC00
38E01800
9782A607 <-- Lifetime 1 runtime in seconds (Big-Endian)
3D020000
00000000
4FE20200
00000000
00000000
92F30900
03
00
537BA607 <-- Circulation pump runtime in seconds (Big-Endian)
CD1B0800
00000000
33382E3033382E30 <-- "38.038.0" (Did not change)
F400     <-- Voltage L1 (Big-Endian)
F400
F400
0000
0000
0000
0000
0000
0000
4700     <-- Power Heater wattage (Big-Endian)
0100     <-- Power L2 wattage - Water heater wattage (Big-Endian)
0000
01
3C00
1E00
00
69       <-- PCB temperature
06       <-- Peripheral Current
010E3417 <-- Real-Time-Clock hours/minute/seconds
150BD507 <-- Real-Time-Clock day/month/year
01
```

Ylianst, 2025-07-20:
@jay7210 - For your logo status lights, you present a bunch of new information here I have not seen before. There are 3 flags along with the logo lights value that changed.

Logo status lights:
```
F1 F2 F3 LL 
08 40 04 0A - Green On - Green On - Blue On (Normal)
08 40 04 02 - Green On - Green OFF - Blue On (Not at Temp)
00 00 00 16 - Yellow On - Green On - Blue On
08 00 04 16 - Yellow Flash - Green On - Blue On
```

For F1, I have always seen it set to 0x08. In your case, there is a situation where it's not set.
For F2, 0x40 I think indicates that the salt level is ok. In your case, 2 situations where it's not.
For F3, I have always seen 0x04 set, you have a situation it's not set.
For LL (Logo Lights):
  - 0A Power & ready on --> Green On - Green On - Blue On (Normal)
  - 02 Power on --> Green On - Green OFF - Blue On (Not at Temp)
  - 16 Power & ready salt error --> Yellow On - Green On - Blue On
  - 16 Power & ready salt error --> Yellow Flash - Green On - Blue On

This is my decoding of one of your commands. You only need to focus on the starts of the command, before the `201C201C201C` sequence. There are a lot of unknown values in this region. You can see there F1, F2, and F3 are located.

```
Yellow Flash - Green On - Blue On
0256              - Main Status Data Command.
0008              - [F1] Flags 1: NEW! 0x08 was only set when: [Yellow On - Green On - Blue On]
00                - [F2] Flags 2: 0x01 = Temp Lock, 0x02 = Spa Lock, 0x04 = Jet1 Full, 0x08 = Jet2 Full, 0x10 = Clean Cycle, 0x20 = Summer Timer, 0x40 = Salt Level Confirmed
04                - [F3] Flags 3: 0x01 = Jet1 Medium, 0x02 = Jet2 Medium, 0x04 = AlwaysSet? (NEW!). 4 upper bits are spa lights: 0x0 = Off, 0x1 = Min Bright, 0x5 = Max Bright
0000              - ?
06                - Model type (does not change)
0600              - ?
16                - [LL] Logo status light
0802F1            - ?
52                - Flags: 0x40 is Celsius
06                - ?
```

jay7210, 2025-07-20:
@Ylianst Can you expose both? 

Normally Steady Yellow appears every 10 days to "check Levels and confirm".
Flashing yellow is Salt System Error.

Ylianst, 2025-07-21:
@jay7210  - Ha yes, it's not a surprise that the host tub lights status command would not show the brightness of the logo light. Also, looking at the traffic, I see you have a Freshwater Salt module and the Audio module, but no, nothing stands out sadly.

I will add two new binary sensors for Flag1 and Flag3 above when I get a chance. I think I already have a sensor for Flag2, but I will check and confirm.

milesroper, 2025-07-21:
Hi, ive updated to ESPBuild 2025.7.2, and reinstalled the ESP-IQ2020 component.  Ive noticed a bit of instability since doing this two days ago.  Yesterday it had dropped from home assistant, but will still contactable via the web interface, and i ended up rebuilding and doing an ota update and it came right.

today i was in the spa and the lights turned off, in the control pannel salt setting was set to 255 and was kind of corrupt, and i noticed in the logs again once I got out the integration and dropped off out of HA.  I did a complete power off of the spa and it again came right.

Not sure if this a controller issue on the spa, or since updating.  Its only been an issue though since i updated to ESPBuild.  Any ideas?  Regards Miles

milesroper, 2025-07-21:
Ive installed legacy ESPbuild at 2025.6.3 and reinstalled it, will post how it goes

arthurfloh, 2025-07-22:
After update / flasching I have many records of this.
WARNING Using `select.SELECT_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `select.select_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. 

Should I modify something in the yaml?

milesroper, 2025-07-22:
> [@milesroper](https://github.com/milesroper) - I update to the latest ESPHome and re-flash pretty frequently and have not seen any issues yet. I have no big evidence for this, but if you see instability going on for a while, you may want to consider a different ATOM+RS485. I am going to re-flash now with the latest ESPHome and report back any issues.

Reverting to 2025.6.3 has resolved the issue for me.  No issues in HA logs of connection dropping and everything seems stable so far.

jay7210, 2025-07-30:
@Ylianst i think i have another State for the Logo lights, I currently have a yellow light and my tub is asking to check levels.

The code is;

1F 01 80 02560008000400000606000A0802F15206201C201C201C840360540000000033372E35E5A3ED00BC181900A637B4073D02000000000000AFE4020000000000000000005C0D0A0003006230B4073D2F08000000000033372E3533372E35F600F600F6000000000000000000000000004C000B000000003C001E000068CE002601141E06E90701.

Can you see if thats any different from the previous states i provided?

I set up an alert that triggers only when the state is not 2.0. But it is still at 2.0 even with the yellow light.

Thanks

derhally, 2025-07-30:
> [@Ylianst](https://github.com/Ylianst) i think i have another State for the Logo lights, I currently have a yellow light and my tub is asking to check levels.
> 
> The code is;
> 
> 1F 01 80 02560008000400000606000A0802F15206201C201C201C840360540000000033372E35E5A3ED00BC181900A637B4073D02000000000000AFE4020000000000000000005C0D0A0003006230B4073D2F08000000000033372E3533372E35F600F600F6000000000000000000000000004C000B000000003C001E000068CE002601141E06E90701.
> 
> Can you see if thats any different from the previous states i provided?
> 
> Also the tub send this as i excepted the warning;
> 
> 01 29 80 1E0101003C0A00030000B60C006841 <-- 29 01 40 1E010102FFFFFF00FF01FFFFFFFFFF <-- 01 29 80 1E0101003C0A00030000B60C006841 <-- 37 01 40 23010000FFFFFFFFFFFFFFFFFFFFFF <-- 33 01 40 190100 <-- 01 33 80 190102 <-- 21 01 40 210100 <-- 33 01 40 190100 <-- 01 33 80 190102 <-- 29 01 40 1E010102FFFFFF00FF01FFFFFFFFFF <-- 01 29 80 1E0101003C0A00030000B60C006841
> 
> I set up an alert that triggers only when the state is not 2.0. But it is still at 2.0 even with the yellow light.
> 
> Thanks

This is similar to the stuff I worked with @Ylianst when I integrated with my hot tub.  The logo light will switch to yellow when you need to confirm salt levels every 2 weeks.

[35 more replies]