Ylianst/ESP-IQ2020 #43: Data Viewer data from Freshwater IQ

homeontherocks, 2025-04-28:
IQ A - is related to the sensor.  I set the value to 500 (from zero) and it immediately told me to reset the sensor.  It took resetting it to 0 and rebooting the spa to get back to normal.  Not sure if it’s binary or not (0 or 1).

IQ B - I could not discern any change in the control panel changing values up to 500.  Not sure what this does.

IQ C - is absolutely salt but has a weird relationship to the bar on the control panel (non linear?).  Sometimes I would enter a value that would be reflected immediately and other times I would slowly increase (100 units at a time) and it wouldn’t change until it jumped up or down.  A value of 2250 puts it just slightly below the middle of the green range but according to the manual that should be 1750.
I entered a few values in the Data Viewer:
1850 -> very low (slightly above red in yellow)
1750 very low red
2250 middle green
2300 upper third green
2600 green yellow junction
3200 yellow red junction

If we just subtract 500 from the reading, here are the approximate ranges from high to low:
2700 start of red
2100 start of yellow
1750 middle of green
~1500 start of yellow
~1300 start of red

Interestingly, when the salt is in the red (high) the instructions are drain 50% water and refill

IQ D - I could not discern any change in the control panel entering 1850 and 3000.  Unsure.

IQ E - Chlorine.  Value/10 is ppm

IQ F - pH.  Value/10 is pH

IQ G - Hours left on Salt cartridge.  Interestingly, when you replace it starts at 9999, which would be 416.6 days.

Log files attached for analysis.
Thanks!

[logfile2.txt](https://github.com/user-attachments/files/19932831/logfile2.txt)
[logfile.txt](https://github.com/user-attachments/files/19932830/logfile.txt)

Ylianst, 2025-04-28:
Oh wow. This is the moment I have been waiting for a long time in regards to Freshwater IQ and certainly the correct way to figure things out. Thank you so much @homeontherocks.

For A, one theory would be that it's not a number, but a set of flags. So, 1, 2, 4, 8, 16, 32, 64... would each be a flag. 500 in binary is "000111110100" is, that enabled 6 flags and one of them could have been a faulty sensor flag. If you ever want to try experimenting again, set A to 1 then 2 then 4 then 8... and note each behavior change.

For B, yes, it could be unused and could be for future stuff. It could also be like my theory about A. In that case, B is also a set of flags and the same experiment would need to be run. Set B to 1, 2, 4, 8, 16...

For C. Very interesting!! Looks like something we can graph. Some ChatGPT assisted research on this, it look like this scale could be Parts Per Million (PPM). Having a 2250 midpoint looks ok. I have seen lower mid-points, but it seems some hot tubs have a higher mid-point in the 3000ppm range.

[image]

Quick graph:

[image]

If we had picture of the remote at different values, we could start counting pixel and get a better idea, however this is already a good improvement. There may some odd scaling on the hub's display. I would not be surprised if it was just a set of "If between X and Y put indicator here" in a table.

For G, that interesting it started at 9999. It could be it's not hours but something else. If you have a plot of this value over time in HA, see if it goes now in a linear way and what the rate is. It could be some other unit.

In any case, this is REALLY good data and you just moved the ball forward significantly.

Ylianst, 2025-04-28:
A quick look at your log files. At no point during your testing did the Freshwater IQ module send any "real" data. So, all you fake data was not being thrown off by the Freshwater IQ module.

Your Freshwater IQ module did send some data on a different command, and it's always the same except between 3/29 and 4/26... the last two digits changed from 1 (0x0001) to 674 (0x0242)

```
3/29/2025 11:56:53 AM  <-- 01 37 80 23D5000000003234313130304E30476EFD3815FF7E4430303744000000040000000400000004000031B400002710000027100000000100000001
4/26/2025  1:33:33 PM  <-- 01 37 80 23D5000000003234313130304E30476EFD3815FF7E4430303744000000040000000400000004000031B40000271000002710000002A2000002A2
```

I did not look a lot, but nothing else looked remarkable at first glance in the logs.

homeontherocks, 2025-04-28:
I collected lots of data for C while looking at the control panel.  If you subtract 500 from the current C value, it very closely tracks the accurate salt level.  I tested my Salt with a test strip and it was 1750.  My IQ C value is pretty closely around 2250 so I know that the value represented by C is not the exact Salt measurement, both manually with strip and on the control panel.

The only other interesting thing is the chlorine value.  When I hit Boost (either in your integration or on the control panel), the Chlorine actually FALLS significantly but then equalizes at a higher level (as expected) 24 hours later after the Boost has been completed.

homeontherocks, 2025-04-28:
[image]

I just did a quick and dirty template subtracting 500 from the salt reading for my dashboard.  Pretty happy with this approximation of the Control Panel for Freshwater IQ.

homeontherocks, 2025-04-28:
Also for G, I confirmed that it decreases 1 unit every hour.

RWag-AC, 2025-08-19:
Sure G is the salt cartridge? To me, it looks like the remaining time of the IQ sensor. My sensor is a little bit more than one year old (one year = 8760 hours) and I have now a reading of 1066 remaining hours. This fits to the start reading of 9999 hours...

homeontherocks, 2025-08-19:
> Sure G is the salt cartridge? To me, it looks like the remaining time of the IQ sensor. My sensor is a little bit more than one year old (one year = 8760 hours) and I have now a reading of 1066 remaining hours. This fits to the start reading of 9999 hours...

Who are you responding to?  Please read my posts again.  Of course G is the IQ remaining hours.  I believe C closely represents the salt level but have to subtract 500 from the value.

homeontherocks, 2025-08-19:
> RWag-AC
>  left a comment
> (Ylianst/ESP-IQ2020#43)
>
> Dear homeontherocks,
> Thank you for your friendly reply.
> In your post you have written:
> “IQ G - Hours left on Salt cartridge. Interestingly, when you replace it starts at 9999, which would be 416.6 days.“
> In my opinion, G are the remaining hours of the IQ sensor (the measuring sensor) and not the remaining hours of the salt cartridge.
> C is corresponding to the salt level. I can confirm that the numbers are higher, maybe 500, than the actual level I measured with a separate measuring sensor.

I misread your post and thought initially that you were referring to Salt levels, which is obviously wrong.  I immediately deleted my post though, as you can see if you visit the thread.  I agree that you are probably correct as I replaced both the Salt cartridge as well as the Sensor cartridge when I took the reading of 9999.  I do not yet see a sensor which indicates time left for the Salt cartridge.

Ylianst, 2025-12-18:
Closing. Please re-open if needed.

closed 2025-12-18 by Ylianst