Ylianst/ESP-IQ2020 #46: ESP Home cannot find fan.iq2020

alindner19, 2025-05-22:
First off, thanks much for doing this.  Was exactly what I was looking for.  I have a 2025 Grandee with the Freshwater IQ which I'm sure you would be shocked to know that their connecting device an App doesn't support the Freshwater IQ.  Just salt level and jets.
This worked about a day or two ago and today just stopped.  I run HA and ESP Home in docker containers.

INFO ESPHome 2025.4.2
INFO Reading configuration /config/hot-tub.yaml...
ERROR Unable to load component iq2020.fan:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 181, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/.esphome/external_components/24b0febb/components/iq2020/fan.py", line 16, in <module>
    CONFIG_SCHEMA = fan.fan_schema(IQ2020Fan).extend({
                    ^^^^^^^^^^^^^^
AttributeError: module 'esphome.components.fan' has no attribute 'fan_schema'
ERROR Unable to load component iq2020.fan:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 181, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/.esphome/external_components/24b0febb/components/iq2020/fan.py", line 16, in <module>
    CONFIG_SCHEMA = fan.fan_schema(IQ2020Fan).extend({
                    ^^^^^^^^^^^^^^
AttributeError: module 'esphome.components.fan' has no attribute 'fan_schema'
Failed config

fan.iq2020: [source /config/hot-tub.yaml:182]
  
  Platform not found: 'fan.iq2020'.
  platform: iq2020
  name: Jets 1
  id: jets1
  icon: mdi:turbine
  datapoint: 0
  speeds: 1
fan.iq2020: [source /config/hot-tub.yaml:188]
  
  Platform not found: 'fan.iq2020'.
  platform: iq2020
  name: Jets 2
  id: jets2
  icon: mdi:turbine
  datapoint: 1
  speeds: 2

Any help would be appreciated.

Ylianst, 2025-05-22:
Oh. I made this change yesterday since the old schema syntax is getting deprecated. I will revert for now.

Ylianst, 2025-05-23:
@alindner19 - Fixed. Let me know if it works.

alindner19, 2025-05-23:
I still get the same thing.  Do I have to do something to get it to reload?

Ylianst, 2025-05-23:
Add `refresh: 10s` under the `external_components:`. It will not really refresh every 10 seconds, but basically, it will reload reach time you compile which is useful for me when I make quick changes.

```
external_components:
  - source: github://ylianst/esp-iq2020
    refresh: 10s
```

alindner19, 2025-05-23:
That got it to work.  Thanks much.

closed 2025-05-23 by alindner19