Ylianst/ESP-IQ2020 #51: Compile Failure

jreagor, 2025-06-29:
Thank you very much for this guide and application. 
IQ2020 in a 2024 Hot Springs Envoy
Compiling via browser through the ESPHome GUI with the M5stack on a USB cable connected to Home Assist Green.

I got to the point of installing the code on the M5 stack and received the error shown below using the .yml below the error. Thank you in advance for the help! I'm very new to this and wouldn't be able to have even attempted without your video.

Compiling .pioenvs/hot-tub/src/esphome/components/iq2020/iq2020_fan.cpp.o
src/esphome/components/iq2020/iq2020.cpp: In member function 'int IQ2020Component::processIQ2020Command()':
src/esphome/components/iq2020/iq2020.cpp:657:8: error: 'select_pending' was not declared in this scope
    if (select_pending[SELECT_LIGHTS_CYCLE_SPEED] != NOT_SET) {
        ^~~~~~~~~~~~~~
src/esphome/components/iq2020/iq2020.cpp:657:8: note: suggested alternative: 'switch_pending'
    if (select_pending[SELECT_LIGHTS_CYCLE_SPEED] != NOT_SET) {
        ^~~~~~~~~~~~~~
        switch_pending
src/esphome/components/iq2020/iq2020.cpp:660:10: error: 'g_iq2020_select' was not declared in this scope
      if (g_iq2020_select[i + 1] == NULL) continue;
          ^~~~~~~~~~~~~~~
src/esphome/components/iq2020/iq2020.cpp:660:10: note: suggested alternative: 'g_iq2020_switch'
      if (g_iq2020_select[i + 1] == NULL) continue;
          ^~~~~~~~~~~~~~~
          g_iq2020_switch
src/esphome/components/iq2020/iq2020.cpp:661:24: error: 'g_iq2020_select' was not declared in this scope
      int cycleColor = (g_iq2020_select[i + 1]->traits.get_options().size() == 8); // SELECT_LIGHTS1_COLOR
                        ^~~~~~~~~~~~~~~
src/esphome/components/iq2020/iq2020.cpp:661:24: note: suggested alternative: 'g_iq2020_switch'
      int cycleColor = (g_iq2020_select[i + 1]->traits.get_options().size() == 8); // SELECT_LIGHTS1_COLOR
                        ^~~~~~~~~~~~~~~
                        g_iq2020_switch
src/esphome/components/iq2020/iq2020.cpp:689:8: error: 'select_pending' was not declared in this scope
    if (select_pending[SELECT_LIGHTS_CYCLE_SPEED] == NOT_SET) {
        ^~~~~~~~~~~~~~
src/esphome/components/iq2020/iq2020.cpp:689:8: note: suggested alternative: 'switch_pending'
    if (select_pending[SELECT_LIGHTS_CYCLE_SPEED] == NOT_SET) {
        ^~~~~~~~~~~~~~
        switch_pending
Compiling .pioenvs/hot-tub/src/esphome/components/iq2020/iq2020_number.cpp.o
In file included from src/esphome/components/iq2020/iq2020_number.cpp:2:
src/esphome/components/iq2020/iq2020_number.h:4:10: fatal error: esphome/components/number/number.h: No such file or directory
 #include "esphome/components/number/number.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/hot-tub/src/esphome/components/iq2020/iq2020_number.cpp.o] Error 1
*** [.pioenvs/hot-tub/src/esphome/components/iq2020/iq2020.cpp.o] Error 1
========================= [FAILED] Took 184.97 seconds =========================

.yml file.....

esphome:
  name: hot-tub
  friendly_name: Hot Tub

esp32:
  board: m5stack-atom

# Enable logging
logger:
  baud_rate: 0
  level: ERROR

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

ota:
  - platform: esphome
    password: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source: github://ylianst/esp-iq2020

# Make sure tx/rx pins are correct for your device.
# GPIO26/32 is ok for M5Stack-ATOM + Tail485, look in GitHub devices link for your device.
uart:
  id: SpaConnection
  tx_pin: GPIO26
  rx_pin: GPIO32
  baud_rate: 38400

iq2020:
   uart_id: SpaConnection
   polling_rate: 65
   port: 1234

# If using Celsius units on the hot tub remote, replace _f_ with _c_ in the three entries below.
# Feel free to remove any sensor that are not relevant for your hot tub.
sensor:
  - platform: iq2020
    current_f_temperature:
      name: Current Temperature
    target_f_temperature:
      name: Target Temperature
    outlet_f_temperature:
      name: Heater Outlet
    power_l1:
      name: Pumps Power
    power_heater:
      name: Power Heater
    power_l2:
      name: Heater Power
    pcb_f_temperature:
      name: Controller Temperature

switch:
  - platform: iq2020
    name: Lights
    id: lights_switch
    icon: "mdi:lightbulb"
    datapoint: 0
  - platform: iq2020
    name: Spa Lock
    id: spa_lock_switch
    icon: "mdi:lock"
    datapoint: 1
  - platform: iq2020
    name: Temperature Lock
    id: temp_lock_switch
    icon: "mdi:lock"
    datapoint: 2
  - platform: iq2020
    name: Clean Cycle
    id: clean_cycle_switch
    icon: "mdi:vacuum"
    datapoint: 3
  - platform: iq2020
    name: Summer Timer
    id: summer_timer_switch
    icon: "mdi:sun-clock"
    datapoint: 4

fan:
  - platform: iq2020
    name: Jets 1
    id: jets1
    icon: "mdi:turbine"
    datapoint: 0
    speeds: 1
  - platform: iq2020
    name: Jets 2
    id: jets2
    icon: "mdi:turbine"
    datapoint: 1
    speeds: 2

# Set "celsius" to "true" if using celsius units.
climate:
  - platform: iq2020
    name: Temperature
    celsius: false

text_sensor:
  - platform: iq2020
    versionstr:
      name: Version

Ylianst, 2025-06-30:
Oh. Something is very much not right with your ESPHome. Try to clear the build files and try again...

[image]

jreagor, 2025-06-30:
Thank you for the suggestion. I tried that but got the same error upon reinstalling. Any other suggestions? Running on HA Core 2025.6.1 with ESPHome 2025.6.2

closed 2025-06-30 by Ylianst