Ylianst/ESP-IQ2020 #52: Button Sensor stopped working

john159753, 2025-07-03:
I'm not sure if this is an esphome issue (or a breaking change in esphome), or just me. but since sometime in June, cant totally pinpoint, my buttons sensor stopped sending updates to home assistant.

The logs show the button presses, and updates to the sensor, but i dont see the number changing in the esphome dashboard or home assistant 

[21:53:53][D][iq2020:369]: AUDIO - PAUSE
[21:53:54][D][sensor:099]: 'Buttons': Sending state 1.00000  with 0 decimals of accuracy
[21:53:54][D][sensor:099]: 'Buttons': Sending state 0.00000  with 0 decimals of accuracy
[21:53:54][D][iq2020:368]: AUDIO - PLAY
[21:53:54][D][sensor:099]: 'Buttons': Sending state 2.00000  with 0 decimals of accuracy
[21:53:54][D][sensor:099]: 'Buttons': Sending state 0.00000  with 0 decimals of accuracy
[21:53:54][D][iq2020:369]: AUDIO - PAUSE
[21:53:56][D][sensor:099]: 'Buttons': Sending state 3.00000  with 0 decimals of accuracy
[21:53:56][D][sensor:099]: 'Buttons': Sending state 0.00000  with 0 decimals of accuracy
[21:53:56][D][iq2020:370]: AUDIO - NEXT
[21:53:57][D][sensor:099]: 'Buttons': Sending state 3.00000  with 0 decimals of accuracy
[21:53:57][D][sensor:099]: 'Buttons': Sending state 0.00000  with 0 decimals of accuracy
[21:53:57][D][iq2020:370]: AUDIO - NEXT
[21:53:57][D][sensor:099]: 'Buttons': Sending state 4.00000  with 0 decimals of accuracy
[21:53:57][D][sensor:099]: 'Buttons': Sending state 0.00000  with 0 decimals of accuracy
[21:53:57][D][iq2020:371]: AUDIO - BACK
[21:53:57][D][sensor:099]: 'Buttons': Sending state 4.00000  with 0 decimals of accuracy
[21:53:57][D][sensor:099]: 'Buttons': Sending state 0.00000  with 0 decimals of accuracy
[21:53:57][D][iq2020:371]: AUDIO - BACK
[21:53:58][D][sensor:099]: 'Buttons': Sending state 4.00000  with 0 decimals of accuracy
[21:53:58][D][sensor:099]: 'Buttons': Sending state 0.00000  with 0 decimals of accuracy
[21:53:58][D][iq2020:371]: AUDIO - BACK
[21:54:00][D][sensor:099]: 'Buttons': Sending state 2.00000  with 0 decimals of accuracy
[21:54:00][D][sensor:099]: 'Buttons': Sending state 0.00000  with 0 decimals of accuracy

This is what HA shows
[image]

Ylianst, 2025-07-03:
So, this is by design. The integration will switch "Buttons" state value to a number and switch it right back to 0 afterwards. This is so that if you push the same button 3 times, it will go to the value and back to zero 3 times. It's made so you can write an automation that is the value change to "2" you can toggle a light. Each time you press "Play", the value change to "2" and back to zero and your automation will be triggered twice.

If I don't do that, you hit play, it change to value 2 and then pressing play again would do nothing since it's already set to value 2.

Ok, I hope that made sense.

Ylianst, 2025-07-03:
I see it's not in the log... but try attaching an automation and see if it works?

john159753, 2025-07-03:
Update -- I think I have found the cause, and I did a quick fix in my fork and now the buttons are working as expected.

I think this is related a breaking change in esphome https://github.com/esphome/esphome/pull/9012

It looks like they implemented a batching system for sensor data, and the default batch period/delay is 100ms, which i think meant that if the change happened in less than 100ms it got lost and no actual state change would be sent.

I added a delay in the set audio button function, for 150ms and then it started working. Is this the right way forward? Or should it be more robust?

```cs
void IQ2020Component::setAudioButton(int button) {
#ifdef USE_SENSOR
	if (this->buttons_sensor_) this->buttons_sensor_->publish_state(button);
	if (this->buttons_sensor_) {
		this->set_timeout(150, [this]() {
			this->buttons_sensor_->publish_state(0);
		});
	}
#endif
}
```

Ylianst, 2025-07-03:
@john159753 Oh my! Thank you so much for looking into this. I was going to try my hot tub in the next few days but you clearly got a solution here. I will add it in right away.

Ylianst, 2025-07-03:
@john159753 - I put in your fix. Let me know if that works.

closed 2025-07-03 by Ylianst