Ylianst/ESP-IQ2020 #55: Platform not found: 'fan.iq2020'.

fclauson, 2025-08-16:
same as https://github.com/Ylianst/ESP-IQ2020/issues/46 

have made the YAML updates (and even deleted the .esphome dir so I am sure we have a clean load)_ 

:\Users\francis\Documents\M5 - Tub>esphome run tubv9.yaml
INFO ESPHome 2025.2.1
INFO Reading configuration tubv9.yaml...
INFO Cloning https://github.com/ylianst/esp-iq2020.git@None
ERROR Unable to load component iq2020.fan:
Traceback (most recent call last):
  File "C:\Users\francis\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\esphome\loader.py", line 181, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "C:\Users\francis\Documents\M5 - Tub\.esphome\external_components\24b0febb\components\iq2020\fan.py", line 16, in <module>
    CONFIG_SCHEMA = fan.fan_schema(IQ2020Fan).extend({
AttributeError: module 'esphome.components.fan' has no attribute 'fan_schema'
ERROR Unable to load component iq2020.fan:
Traceback (most recent call last):
  File "C:\Users\francis\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\esphome\loader.py", line 181, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "C:\Users\francis\Documents\M5 - Tub\.esphome\external_components\24b0febb\components\iq2020\fan.py", line 16, in <module>
    CONFIG_SCHEMA = fan.fan_schema(IQ2020Fan).extend({
AttributeError: module 'esphome.components.fan' has no attribute 'fan_schema'
Failed config

fan.iq2020: [source tubv9.yaml:170]

  Platform not found: 'fan.iq2020'.
  platform: iq2020
  name: Jets 1
  id: jets1
  icon: mdi:turbine

Ylianst, 2025-08-16:
@fclauson - Try adding the following line to your yaml, even if it's empty:

```
fan:
```

If that is not the problem, post your yaml so I can check. Make sure to remove any private information before posting it.

Ylianst, 2025-08-16:
@fclauson - (I made some formatting edits to your posts above for clarity). I compiled your problem yaml and it worked for me. I now realize the problem is likely your not using the latest ESPHome which now has breaking changes. However, I have a trick. I kept an older version of ESP-IQ2020 available for just this case. In your yaml, change all `platform: iq2020` to `platform: iq2020-legacy` and try again or update to latest ESPHome.

Let me know if that works.

fclauson, 2025-08-16:
FIXED

upgraded esphome 

thanks for the help

closed 2025-08-16 by fclauson

Ylianst, 2025-08-16:
Nice. Thank you @fclauson for reporting back!