Ylianst/ESP-IQ2020 #59: Add ESP‑IDF Framework Support for ESPHome Builds

dhellerstedt, 2025-09-03:
Hi @Ylianst,

First off, thank you for all the work you’ve put into ESP‑IQ2020 — it’s been rock‑solid for my hot tub integration under ESPHome + Arduino.

With ESPHome announcing that the default ESP32 framework will switch from Arduino to ESP‑IDF in 2026.1.0 ([ESPHome migration guide](https://esphome.io/guides/esp32_arduino_to_idf/)), I wanted to ask if there are plans to make ESP‑IQ2020 compatible with ESP‑IDF builds.

---

## Why this matters
- ESPHome will eventually default to ESP‑IDF for ESP32 boards.
- Without IDF support, existing Arduino‑based builds may break or require pinning the framework.
- IDF offers smaller binaries, better performance, and more active upstream testing.

---

## Current Arduino‑specific dependencies
From reviewing the source, the main Arduino‑only calls are:

| Arduino API | Location | IDF Equivalent |
|-------------|----------|----------------|
| `HardwareSerial Serial2` / `Serial2.begin(...)` | UART init | `uart_param_config()`, `uart_set_pin()`, `uart_driver_install()` |
| `Serial2.write(buf, len)` | Packet send | `uart_write_bytes(uart_num, (const char *)buf, len)` |
| `Serial2.available()` | Check RX buffer | `uart_get_buffered_data_len(uart_num, &len)` |
| `Serial2.readBytes(buf, len)` | Packet receive | `uart_read_bytes(uart_num, buf, len, pdMS_TO_TICKS(timeout_ms))` |
| `millis()` | Timing | `(esp_timer_get_time() / 1000ULL)` |
| `delay(ms)` | Delay | `vTaskDelay(pdMS_TO_TICKS(ms))` |
| `Serial.printf(...)` | Logging | `ESP_LOGI(TAG, "...", args...)` |
| `String` | String handling | `std::string` or C strings |
| `pinMode()` / `digitalWrite()` | RS‑485 DE/RE control | `gpio_set_direction()`, `gpio_set_level()` |

---

## Suggested migration approach
To keep both Arduino and IDF builds working during transition, wrap the changes in preprocessor guards:

```cpp
#if defined(USE_ARDUINO)
  Serial2.begin(38400, SERIAL_8N1, RX_PIN, TX_PIN);
#else
  uart_config_t uart_config = {
    .baud_rate = 38400,
    .data_bits = UART_DATA_8_BITS,
    .parity    = UART_PARITY_DISABLE,
    .stop_bits = UART_STOP_BITS_1,
    .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
  };
  uart_param_config(UART_NUM_1, &uart_config);
  uart_set_pin(UART_NUM_1, TX_PIN, RX_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
  uart_driver_install(UART_NUM_1, 1024, 0, 0, NULL, 0);
#endif

---

## **2️⃣ Recommended Source Code Changes**

Here’s a **drop‑in replacement** for the UART transport layer that will work under both Arduino and ESP‑IDF.  
Save this as `iq2020_uart.cpp` (or similar) in the component’s `src` folder.

```cpp
#include "esphome.h"

#if defined(USE_ARDUINO)
  #include <HardwareSerial.h>
  static HardwareSerial iq2020_serial(2);
#else
  #include "driver/uart.h"
  #include "driver/gpio.h"
  #include "esp_log.h"
  static const char *TAG = "iq2020";
  static const uart_port_t uart_num = UART_NUM_1;
#endif

#define IQ2020_BAUD 38400
#define IQ2020_TX   GPIO_NUM_26
#define IQ2020_RX   GPIO_NUM_32
#define IQ2020_DE   GPIO_NUM_21  // Example DE/RE pin

class IQ2020UART : public Component {
 public:
  void setup() override {
  #if defined(USE_ARDUINO)
    iq2020_serial.begin(IQ2020_BAUD, SERIAL_8N1, IQ2020_RX, IQ2020_TX);
    pinMode(IQ2020_DE, OUTPUT);
    digitalWrite(IQ2020_DE, LOW);
  #else
    uart_config_t uart_config = {
      .baud_rate = IQ2020_BAUD,
      .data_bits = UART_DATA_8_BITS,
      .parity    = UART_PARITY_DISABLE,
      .stop_bits = UART_STOP_BITS_1,
      .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
    };
    uart_param_config(uart_num, &uart_config);
    uart_set_pin(uart_num, IQ2020_TX, IQ2020_RX, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
    uart_driver_install(uart_num, 1024, 0, 0, NULL, 0);

    gpio_set_direction(IQ2020_DE, GPIO_MODE_OUTPUT);
    gpio_set_level(IQ2020_DE, 0);
  #endif
  }

  void send_data(const uint8_t *buf, size_t len) {
  #if defined(USE_ARDUINO)
    digitalWrite(IQ2020_DE, HIGH);
    iq2020_serial.write(buf, len);
    iq2020_serial.flush();
    digitalWrite(IQ2020_DE, LOW);
  #else
    gpio_set_level(IQ2020_DE, 1);
    uart_write_bytes(uart_num, (const char *)buf, len);
    uart_wait_tx_done(uart_num, pdMS_TO_TICKS(100));
    gpio_set_level(IQ2020_DE, 0);
  #endif
  }

  size_t read_data(uint8_t *buf, size_t len, uint32_t timeout_ms) {
  #if defined(USE_ARDUINO)
    return iq2020_serial.readBytes(buf, len);
  #else
    return uart_read_bytes(uart_num, buf, len, pdMS_TO_TICKS(timeout_ms));
  #endif
  }
};

iq2020_uart.h

`#pragma once

#include "esphome.h"

class IQ2020UART : public Component {
 public:
  void setup() override;
  void send_data(const uint8_t *buf, size_t len);
  size_t read_data(uint8_t *buf, size_t len, uint32_t timeout_ms);
};
`

iq2020_uart.cpp

`#include "esphome.h"
#include "iq2020_uart.h"

#if defined(USE_ARDUINO)
  #include <HardwareSerial.h>
  static HardwareSerial iq2020_serial(2);
#else
  #include "driver/uart.h"
  #include "driver/gpio.h"
  #include "esp_log.h"
  static const char *TAG = "iq2020";
  static const uart_port_t uart_num = UART_NUM_1;
#endif

#define IQ2020_BAUD 38400
#define IQ2020_TX   GPIO_NUM_26
#define IQ2020_RX   GPIO_NUM_32
#define IQ2020_DE   GPIO_NUM_21

void IQ2020UART::setup() {
#if defined(USE_ARDUINO)
  iq2020_serial.begin(IQ2020_BAUD, SERIAL_8N1, IQ2020_RX, IQ2020_TX);
  pinMode(IQ2020_DE, OUTPUT);
  digitalWrite(IQ2020_DE, LOW);
#else
  uart_config_t uart_config = {
    .baud_rate = IQ2020_BAUD,
    .data_bits = UART_DATA_8_BITS,
    .parity    = UART_PARITY_DISABLE,
    .stop_bits = UART_STOP_BITS_1,
    .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
  };
  uart_param_config(uart_num, &uart_config);
  uart_set_pin(uart_num, IQ2020_TX, IQ2020_RX, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
  uart_driver_install(uart_num, 1024, 0, 0, NULL, 0);
  gpio_set_direction(IQ2020_DE, GPIO_MODE_OUTPUT);
  gpio_set_level(IQ2020_DE, 0);
#endif
}

void IQ2020UART::send_data(const uint8_t *buf, size_t len) {
#if defined(USE_ARDUINO)
  digitalWrite(IQ2020_DE, HIGH);
  iq2020_serial.write(buf, len);
  iq2020_serial.flush();
  digitalWrite(IQ2020_DE, LOW);
#else
  gpio_set_level(IQ2020_DE, 1);
  uart_write_bytes(uart_num, (const char *)buf, len);
  uart_wait_tx_done(uart_num, pdMS_TO_TICKS(100));
  gpio_set_level(IQ2020_DE, 0);
#endif
}

size_t IQ2020UART::read_data(uint8_t *buf, size_t len, uint32_t timeout_ms) {
#if defined(USE_ARDUINO)
  return iq2020_serial.readBytes(buf, len);
#else
  return uart_read_bytes(uart_num, buf, len, pdMS_TO_TICKS(timeout_ms));
#endif
}
`

hot-tub-idf.yaml

`esphome:
  name: hot-tub-idf
  platform: ESP32
  board: m5stack-atom

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

logger:
ota:
api:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

custom_component:
  - lambda: |-
      static auto iq_uart = new IQ2020UART();
      return {iq_uart};

external_components:
  - source: github://Ylianst/esp-iq2020
    components: [ iq2020 ]

uart:
  id: spa_uart
  tx_pin: GPIO26
  rx_pin: GPIO32
  baud_rate: 38400

fan:
  - platform: iq2020
    name: Jets 1
    id: jets1
    datapoint: 0
    speeds: 1

  - platform: iq2020
    name: Jets 2
    id: jets2
    datapoint: 1
    speeds: 2

climate:
  - platform: iq2020
    name: Spa Temperature
    id: spa_temp
    datapoint: 2
`
I’m happy to:

- Test IDF builds on M5Stack Atom hardware
- Provide compile logs and runtime validation
- Submit a PR with these changes if you’d prefer

Thanks again for maintaining this project!

Ylianst, 2025-09-03:
Hi @dhellerstedt and thanks for raising this issue. In issue #57, I took a look at ESP-IDF for the first time and indeed, the resulting binary is much smaller. I compared both binaries and my results are below. Obviously, this will change a little based on what configuration you have.

As far as I know, the current code works for both Arduino and ESP-IDF, but you seem to know more about this than I do. Is the serial code change needed and/or what are the benefits? Now that I see your proposed changes, I think I need to get more information on this change. Any feedback appreciated. I am personally running ESP-IDF right now on my Hot tub, but maybe I am not doing it correctly or fully?

```
type: arduino

Linking .pioenvs/hot-tub/firmware.elf
                            Memory Type Usage Summary                             
┏━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━┳━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━┓
┃ Memory Type/Section ┃ Used [bytes] ┃ Used [%] ┃ Remain [bytes] ┃ Total [bytes] ┃
┡━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━╇━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━┩
│ Flash Code          │       845468 │          │                │               │
│    .text            │       845468 │          │                │               │
│ Flash Data          │       177684 │          │                │               │
│    .rodata          │       177428 │          │                │               │
│    .appdesc         │          256 │          │                │               │
│ IRAM                │        86994 │    66.37 │          44078 │        131072 │
│    .text            │        85967 │    65.59 │                │               │
│    .vectors         │         1027 │     0.78 │                │               │
│ DRAM                │        47512 │    38.14 │          77068 │        124580 │
│    .bss             │        24264 │    19.48 │                │               │
│    .data            │        23248 │    18.66 │                │               │
└─────────────────────┴──────────────┴──────────┴────────────────┴───────────────┘
Total image size: 1133394 bytes (.bin may be padded larger)
Note: The reported total sizes may be smaller than those in the technical reference manual due to reserved memory and application configuration. The total flash size available for the application is not included by default, as it cannot be reliably determined due to the presence of other data like the bootloader, partition table, and application partition size.
RAM:   [=         ]  14.5% (used 47512 bytes from 327680 bytes)
Flash: [======    ]  61.8% (used 1133138 bytes from 1835008 bytes)
Building .pioenvs/hot-tub/firmware.bin
Warning: DEPRECATED: 'esptool.py' is deprecated. Please use 'esptool' instead. The '.py' suffix will be 

type: esp-idf

Linking .pioenvs/hot-tub/firmware.elf
                            Memory Type Usage Summary                             
┏━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━┳━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━┓
┃ Memory Type/Section ┃ Used [bytes] ┃ Used [%] ┃ Remain [bytes] ┃ Total [bytes] ┃
┡━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━╇━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━┩
│ Flash Code          │       684448 │          │                │               │
│    .text            │       684448 │          │                │               │
│ Flash Data          │       133052 │          │                │               │
│    .rodata          │       132796 │          │                │               │
│    .appdesc         │          256 │          │                │               │
│ IRAM                │        92722 │    70.74 │          38350 │        131072 │
│    .text            │        91695 │    69.96 │                │               │
│    .vectors         │         1027 │     0.78 │                │               │
│ DRAM                │        34908 │    19.31 │         145828 │        180736 │
│    .bss             │        18808 │    10.41 │                │               │
│    .data            │        16100 │     8.91 │                │               │
└─────────────────────┴──────────────┴──────────┴────────────────┴───────────────┘
Total image size: 926322 bytes (.bin may be padded larger)
Note: The reported total sizes may be smaller than those in the technical reference manual due to reserved memory and application configuration. The total flash size available for the application is not included by default, as it cannot be reliably determined due to the presence of other data like the bootloader, partition table, and application partition size.
RAM:   [=         ]  10.7% (used 34908 bytes from 327680 bytes)
Flash: [=====     ]  50.5% (used 926066 bytes from 1835008 bytes)
Building .pioenvs/hot-tub/firmware.bin
```

Ylianst, 2025-09-13:
Closing since this seems to be a solved issue. Please re-open if needed.