Ylianst/ESP-IQ2020 #62: fan Includes not found

RobertTurner2, 2025-09-16:
Thanks for a great project !

The last time the project compiled was 2025.7.5 after that this error happens.

INFO ESPHome 2025.8.4
INFO Reading configuration /config/esphome/hot-tub.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing hot-tub (board: m5stack-atom; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Package configuration completed successfully
INFO Package configuration completed successfully
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50402.0 (5.4.2) 
 - tool-cmake @ 3.30.2 
 - tool-esp-rom-elfs @ 2024.10.11 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.13.1 
 - tool-scons @ 4.40801.0 (4.8.1) 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
Reading CMake configuration...
No dependencies
Compiling .pioenvs/hot-tub/src/esphome/components/iq2020/iq2020.cpp.o
Compiling .pioenvs/hot-tub/src/esphome/components/uart/uart.cpp.o
Compiling .pioenvs/hot-tub/src/esphome/components/uart/uart_component.cpp.o
Compiling .pioenvs/hot-tub/src/esphome/components/uart/uart_component_esp_idf.cpp.o
Compiling .pioenvs/hot-tub/src/esphome/components/uart/uart_debugger.cpp.o
Compiling .pioenvs/hot-tub/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/hot-tub/src/esphome/components/wifi/wifi_component_esp_idf.cpp.o
Compiling .pioenvs/hot-tub/src/esphome/core/application.cpp.o
src/esphome/components/iq2020/iq2020.cpp:2:10: fatal error: fan/iq2020_fan.h: No such file or directory
    2 | #include "fan/iq2020_fan.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pioenvs/hot-tub/src/esphome/core/color.cpp.o
*** [.pioenvs/hot-tub/src/esphome/components/iq2020/iq2020.cpp.o] Error 1
========================== [FAILED] Took 2.74 seconds ==========================

Yaml configuration 

esphome:
  name: hot-tub
  friendly_name: Hot Tub
  comment: "Hot Springs Sovereign Spa"

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

logger:
  baud_rate: 0

api:
 
ota:
  - platform: esphome
    
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source: github://Ylianst/ESP-IQ2020
    components: [iq2020]
    refresh: 10s
  - source:
      type: local
      path: local_components
    components: []

uart:
  id: SpaConnection
  tx_pin: GPIO26
  rx_pin: GPIO32
  baud_rate: 38400

iq2020:
  uart_id: SpaConnection
  port: 1234

i2c:
  sda: GPIO25
  scl: GPIO21
  scan: False

time:
  - platform: homeassistant
    id: homeassistant_time
    on_time_sync:
      then:
      - logger.log: "Synchronized system clock"

number:
  - platform: iq2020
    id: salt_power
    name: Salt Power
    datapoint: 5

binary_sensor:    
  - platform: iq2020
    salt_level_confirmed:
      name: Salt Level Confirmed

sensor:
  - platform: iq2020
    current_f_temperature:
      name: Current Temperature
    target_f_temperature:
      name: Target Temperature
    power_l1:
      name: Pumps Power
    current_l1:
      name: Current Pumps
    power_l2:
      name: Heater Power
    current_l2:
      name: Current Heater
    salt_content:
      name: Salt Content
    lifetime_runtime:
      name: Lifetime Runtime
      id: Lifetime
  
switch:
  - platform: iq2020
    name: Clean Cycle
    id: clean_cycle_switch
    icon: "mdi:vacuum"
    datapoint: 3  

climate:
  - platform: iq2020
    name: Temperature
    celsius: false

text_sensor:
  - platform: iq2020
    versionstr:
      name: Version

select:
text:
fan:

Thanks

RobertTurner2, 2025-09-17:
Thanks i did not realise empty fan: section caused the problem, and did not try to add it because i only used the clean cycle from the  homeassistant front end.
Added your sugestion and it works just fine i just wont add the switch to the spa card.

Thanks again.

closed 2025-09-17 by RobertTurner2

Ylianst, 2025-09-18:
@RobertTurner2 - Nice. Thanks for reporting back.